realsenseai/librealsense #1083: [D415] multiple view reconstruction and measurement

MorsLiang, 2018-01-26:
| Required Info  |   |
|---|---|
| Camera Model | D415 | 
| Firmware Version |  | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |
| SDK Version |2.9.1 |

How do I set up the post-processing algorithm so that the point cloud that is scanned is closest to the surface of the scanned object?  I am using "High Accuracy" mode.

Because I found that the edge of the cloud was deformed, I used ICP to match point cloud to the high-precision point cloud which as ground truth and found that the edge of point cloud was tilted up,This problem is not obvious in Kinect V1 and Kinect V2.

The red is ground truth ,show as following:
[image]

[image]

RealSense-Customer-Engineering, 2018-05-16:
[Realsense Customer Engineering Team Comment]
Hello @MorsLiang, there is a white paper on best known methods for tuning that may be helpful. You can download it here, https://realsense.intel.com/intel-realsense-downloads/#whitepaper. Please let us know if you find this helpful or if you still have questions.

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Any further assistance needed?  If not, this ticket will be closed in 5 days.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, and please feel free to open new ticket if any question.

closed 2018-07-02 by RealSense-Customer-Engineering