realsenseai/librealsense #1084: Inconsistency in Projection Wiki page?

jb455, 2018-01-26:
First off, I want to say thanks to @zivsha: the [Projection Wiki page](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0) does a great job of explaining everything. I've spent the last few days trying to get my head around alignment & (de)projection so this was a very welcome find!

However, there's one point which has me confused. 
In the [Depth Image Formats](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#depth-image-formats) section it quotes the following code to retrieve pixel depth:
```
const float scale = rs2_get_depth_scale(sensor, NULL);
const uint16_t * image = (const uint16_t *)rs2_get_frame_data(frame, NULL);
float depth_in_meters = scale * image[pixel_index];
```
But then at the bottom in [Appendix: Model Specific Details](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#appendix-model-specific-details), it quotes similar but different code to also get the pixel depth:
```
const float scale = rs2_get_depth_scale(sensor, NULL);
const uint16_t * image = (const uint16_t *)rs2_get_frame_data(frame, NULL);
float depth_in_meters = scale / image[pixel_index];
```
Is the latter intended when the frame is a different format to the standard depth image? Or is that how we calculate depth using a D400 and the former is just for SR300 so we have to determine which camera is in use before calculating the depth? (The preceding paragraph talks about an inverse relationship so I don't think it's just a typo).

Also, a suggestion/request. The [measure example readme](https://github.com/IntelRealSense/librealsense/blob/master/examples/measure/readme.md#converting-between-pixels-and-points-in-3d) states that repeated calling of get_distance can result in poor performance, so would it be possible to get examples for obtaining an array with the depth of whole frame at once please? Some C# examples would be fantastic too if possible.

Thanks!

zivsha, 2018-01-26:
Hi @jb455 ,
The second point in the appendix refers to disparity and the code is confusing as it is indeed referring to an image with a format of `DISPARITY16`. Unfortunately, the D400 cameras do not provide this format at the moment ( not sure if they will in the future), so I think I will remove the code snippet from there to avoid confusion.

Regarding the comment in measure example, the way to avoid calling cross dll boundaries each pixel, you can save the depth scale from the depth sensor, and multiply that per pixel. I think we do that in the align example.

A C# version for that would look something like this (this is a bit pseudo code):
```csharp
Sensor s = getDepthSensor();
var depthSensor = s as DepthSensor;
var scale = depthSensor.getDepthScale();
foreach( var pixel in image)
{
   var distMeters = scale * pixel.z;
}
```

jb455, 2018-01-30:
Thanks for the clarification.
I guess that C# snippet will be iterating through the point cloud (as it's pixel.z)? Though PointCloud isn't exposed through the wrapper yet... I do have it working by 'unsafely' traversing through the DepthImage.Data pointer array but thought there may have been a better way of doing it. No worries, I'll leave it as it is for now!

closed 2018-01-30 by jb455