realsenseai/librealsense #1085: Pointer type incorrect in rs2_supports_option() and friends

SirDifferential, 2018-01-26:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | N/A   | 
| Operating System & Version | Debian 9   |
| Kernel Version (Linux Only)  |  4.9.65 with patches applied |
| SDK Version | 2.9.1.0, built with `-march=native` |

In the C API there are several functions for getting and setting the assorted sensor options. The functions [take a parameter](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_option.h#L146) of type `rs2_options`, but there seemingly isn't a way to get a list of such  types in the C API. This type would seem to map to a C++ base class in [rs.cpp:45](https://github.com/IntelRealSense/librealsense/blob/master/src/rs.cpp#L44).

Are we supposed to perform a pointer cast or is this an oversight in the C API? The documentation on those functions suggests the intent was to use `rs2_sensor*`

```
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_option.h>

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>

uint8_t check_error(rs2_error* e)
{
    if (e)
    {
        fprintf(stderr, "rs_error was raised when calling %s(%s): \n",
               rs2_get_failed_function(e), rs2_get_failed_args(e));
        fprintf(stderr, "%s\n", rs2_get_error_message(e));
        return 1;
    }
    return 0;
}

int main()
{
    rs2_error* e = 0;

    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    if (check_error(e) != 0)
        return 1;

    rs2_device_list* device_list = rs2_query_devices(ctx, &e);
    if (check_error(e) != 0)
        return 1;

    int dev_count = rs2_get_device_count(device_list, &e);
    if (check_error(e) != 0)
        return 1;

    printf("There are %d connected RealSense devices.\n", dev_count);
    if (0 == dev_count)
        return 1;

    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
    rs2_device* dev = rs2_create_device(device_list, 0, &e);
    if (check_error(e) != 0)
        return 1;

    rs2_sensor_list* slist = rs2_query_sensors(dev, &e);
    if (check_error(e) != 0)
        return 1;

    int sensor_list_count = rs2_get_sensors_count(slist, &e);
    if (check_error(e) != 0)
        return 1;

    for (int sensor_index = 0; sensor_index <sensor_list_count; sensor_index++)
    {
        rs2_sensor* sen = rs2_create_sensor(slist, sensor_index, &e);

        if (check_error(e) != 0) {
            fprintf(stderr, "Failed creating temp sensor object %d / %d\n", sensor_index, sensor_list_count);
            return 1;
        }

        // Scary cast here
        int supports = rs2_supports_option((const rs2_options*)sen, RS2_OPTION_LASER_POWER, &e);
        if (check_error(e) != 0)
            return 1;

        if (supports) {

            float min, max, step, def;
            // Scary cast here
            rs2_get_option_range((const rs2_options*)sen, RS2_OPTION_LASER_POWER, &min, &max, &step, &def, &e);
            if (check_error(e) != 0)
                return 1;

            // Scary cast here
            rs2_set_option((const rs2_options*)sen, RS2_OPTION_LASER_POWER, max, &e);
            if (check_error(e) != 0)
                return 1;
            fprintf(stderr, "Set laser power to: %f\n", max);
        }

        rs2_delete_sensor(sen);
    }

    rs2_delete_sensor_list(slist);
    rs2_delete_device(dev);
    rs2_delete_device_list(device_list);
    rs2_delete_context(ctx);

    return EXIT_SUCCESS;
}
```

zivsha, 2018-01-28:
> Are we supposed to perform a cast or is this an oversight in the C API? The documentation on those functions suggests the intent was to use rs2_sensor*
Yes. e.g:
```cpp
rs2_sensor* sensor = rs2_create_sensor(sensors, 0, &e);
int is_supported = rs2_supports_option((rs2_options*)sensor, RS2_OPTION_EXPOSURE, &e); 
```

It might be missing in the documentation but the `rs2_*option*` family of functions that takes a `rs2_options*`, can be used with either a sensor, or a `rs2_processing_block*`, e.g.:
```cpp
rs2_processing_block* colorizer = rs2_create_colorizer(&e);
int is_supported = rs2_supports_option((rs2_options*)colorizer, RS2_OPTION_COLOR_SCHEME, &e);
```
This was done from a C++ developer's point of view in order to support inheritance inside the library.
I will add the follow up tag to this, so we will remember to update our documentation (if it doesn't exist)

SirDifferential, 2018-01-28:
Thank you for the clarification.

closed 2018-04-18 by RealSense-Customer-Engineering