realsenseai/librealsense #1087: Python wrapper causes `free(): invalid pointer` exception

zhutq, 2018-01-27:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0 | 
| Operating System & Version | Ubuntu 16.04.3 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-112-generic |
| SDK Version | 2.9.1 |

Python version: 3.5.2

The python wrapper is installed following https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python , built from tag v2.9.1.

The python file only has a `import` line:
```python
import pyrealsense2 as rs
```

Output:
```
*** Error in `python3': free(): invalid pointer: 0x00007f3f13489cac ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f3f148207e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f3f1482937a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f3f1482d53c]
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x167707)[0x7f3f1346d707]
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x17ec62)[0x7f3f13484c62]
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x17ec81)[0x7f3f13484c81]
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x176146)[0x7f3f1347c146]
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x176165)[0x7f3f1347c165]
python3[0x4ab8e9]
python3[0x4e9898]
python3[0x4e8365]
python3(PyDict_Clear+0xd6)[0x59b3b6]
python3[0x580796]
python3[0x51860d]
python3(_PyGC_CollectNoFail+0x27)[0x60a9b7]
python3(PyImport_Cleanup+0x24f)[0x523b3f]
python3(Py_Finalize+0x5e)[0x60da0e]
python3(Py_Main+0x646)[0x63fd16]
python3(main+0xe1)[0x4cfeb1]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f3f147c9830]
python3(_start+0x29)[0x5d6049]
======= Memory map: ========
00400000-007a9000 r-xp 00000000 103:01 9699723                           /usr/bin/python3.5
009a9000-009ab000 r--p 003a9000 103:01 9699723                           /usr/bin/python3.5
009ab000-00a42000 rw-p 003ab000 103:01 9699723                           /usr/bin/python3.5
00a42000-00a73000 rw-p 00000000 00:00 0
00fbe000-011e2000 rw-p 00000000 00:00 0                                  [heap]
7f3f0c000000-7f3f0c021000 rw-p 00000000 00:00 0
7f3f0c021000-7f3f10000000 ---p 00000000 00:00 0
7f3f11432000-7f3f11472000 rw-p 00000000 00:00 0
7f3f11472000-7f3f11479000 r-xp 00000000 103:01 6821984                   /lib/x86_64-linux-gnu/librt-2.23.so
7f3f11479000-7f3f11678000 ---p 00007000 103:01 6821984                   /lib/x86_64-linux-gnu/librt-2.23.so
7f3f11678000-7f3f11679000 r--p 00006000 103:01 6821984                   /lib/x86_64-linux-gnu/librt-2.23.so
7f3f11679000-7f3f1167a000 rw-p 00007000 103:01 6821984                   /lib/x86_64-linux-gnu/librt-2.23.so
7f3f1167a000-7f3f1169b000 r-xp 00000000 103:01 9700222                   /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7f3f1169b000-7f3f1189a000 ---p 00021000 103:01 9700222                   /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7f3f1189a000-7f3f1189b000 r--p 00020000 103:01 9700222                   /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7f3f1189b000-7f3f1189c000 rw-p 00021000 103:01 9700222                   /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0
7f3f1189c000-7f3f118b2000 r-xp 00000000 103:01 6820309                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f3f118b2000-7f3f11ab1000 ---p 00016000 103:01 6820309                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f3f11ab1000-7f3f11ab2000 rw-p 00015000 103:01 6820309                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f3f11ab2000-7f3f11c24000 r-xp 00000000 103:01 9699844                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f3f11c24000-7f3f11e24000 ---p 00172000 103:01 9699844                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f3f11e24000-7f3f11e2e000 r--p 00172000 103:01 9699844                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f3f11e2e000-7f3f11e30000 rw-p 0017c000 103:01 9699844                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f3f11e30000-7f3f11e34000 rw-p 00000000 00:00 0
7f3f11e34000-7f3f13009000 r-xp 00000000 103:01 10109330                  /usr/local/lib/librealsense2.so.2.9.1
7f3f13009000-7f3f13209000 ---p 011d5000 103:01 10109330                  /usr/local/lib/librealsense2.so.2.9.1
7f3f13209000-7f3f1323f000 r--p 011d5000 103:01 10109330                  /usr/local/lib/librealsense2.so.2.9.1
[... 30 lines cut ...]
7f3f14b69000-7f3f14b6d000 r--p 001c0000 103:01 6821966                   /lib/x86_64-linux-gnu/libc-2.23.so
7f3f14b6d000-7f3f14b6f000 rw-p 001c4000 103:01 6821966                   /lib/x86_64-linux-gnu/libc-2.23.so
7f3f14b6f000-7f3f14b73000 rw-p 00000000 00:00 0
7f3f14b73000-7f3f14b8b000 r-xp 00000000 103:01 6821965                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f3f14b8b000-7f3f14d8a000 ---p 00018000 103:01 6821965                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f3f14d8a000-7f3f14d8b000 r--p 00017000 103:01 6821965                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f3f14d8b000-7f3f14d8c000 rw-p 00018000 103:01 6821965                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f3f14d8c000-7f3f14d90000 rw-p 00000000 00:00 0
7f3f14d90000-7f3f14db6000 r-xp 00000000 103:01 6821964                   /lib/x86_64-linux-gnu/ld-2.23.so
7f3f14dd4000-7f3f14f8b000 rw-p 00000000 00:00 0
7f3f14f8d000-7f3f14f8e000 rw-p 00000000 00:00 0
7f3f14f8e000-7f3f14fac000 r-xp 00000000 103:01 6820646                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f3f14fac000-7f3f14fad000 r--p 0001d000 103:01 6820646                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f3f14fad000-7f3f14fae000 rw-p 0001e000 103:01 6820646                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f3f14fae000-7f3f14fb5000 r--s 00000000 103:01 9962951                   /usr/lib/x86_64-linux-gnu/gconv/gconv-modules.cache
7f3f14fb5000-7f3f14fb6000 r--p 00025000 103:01 6821964                   /lib/x86_64-linux-gnu/ld-2.23.so
7f3f14fb6000-7f3f14fb7000 rw-p 00026000 103:01 6821964                   /lib/x86_64-linux-gnu/ld-2.23.so
7f3f14fb7000-7f3f14fb8000 rw-p 00000000 00:00 0
7fff62eb5000-7fff62ed7000 rw-p 00000000 00:00 0                          [stack]
7fff62fbc000-7fff62fbf000 r--p 00000000 00:00 0                          [vvar]
7fff62fbf000-7fff62fc1000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)
```
realsense-viewer works well.

zivsha, 2018-01-28:
Hi @zhutq ,
First, Please ignore the references to `rs2_set_devices_changed_callback` issue, I got the number wrong in my commit, (PR #1088 is referring to Issue #1082).

Second, I fail to reproduce this issue on my machine. Is this a regression for you (did it used to work?), did you build `pyrealsense2` using the same version you are running it (can specify the version using the cmake flag `-DPYTHON_EXECUTABLE=<path to python>`)

zhutq, 2018-01-29:
Hi @zivsha ,
Thanks for the response!
It is not a regression for me. It's the first time I use the Python wrapper.
I tried to uninstall and clean and rebuild with `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3` and it's the same version building and running, the exception still exists.

bitzombie1, 2018-01-31:
So I had opencv_viewer_example.py running in Llibrealsense version 2.8.1 and would exit without a stack trace even if exited using control-c. Now in version 2.9.1 I get the same output that @zhutq  has shown here. I also got a similar backtrace when I run the align-depth2color.py example shown at the bottom. @Zivsha reminded me that the depth stream cannot be configured as shown here:
`
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
`
The depth stream should run at width: 640, depth 480 and I found that out in version 2.8.1 when I got the error:
`
RuntimeError: Failed to resolve request. No device found that satisfies all requirements
`

If I was to guess, I would say the above RuntimeError is returned from the the C/C++ librealsense code and when that error is handed to the Python wrapper it fails to capture or handle the error creating the free(): invalid pointer  stack dump. 

This was my backtrace running align-depth2color.py with the wrong depth stream dimensions: 

(cv) charles@UP-16-fd:~/librealsense/wrappers/python/examples$ python align-depth2color.py
Traceback (most recent call last):
File "align-depth2color.py", line 25, in 
profile = pipeline.start(config)
RuntimeError: Failed to resolve request. No device found that satisfies all requirements
*** Error in python': free(): invalid pointer: 0x00007f412d988c8c ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f412eaec7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f412eaf537a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f412eaf953c]
/home/charles/.virtualenvs/cv/lib/python3.5/site-packages/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x1676e7)[0x7f412d96c6e7]
/home/charles/.virtualenvs/cv/lib/python3.5/site-packages/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x17ec42)[0x7f412d983c42]

zivsha, 2018-02-01:
Thanks guys, I am able to reproduce this as well.
We'll look into it.

closed 2018-02-08 by zivsha

likestudy, 2019-11-23:
Hello @zivsha . Has this problem been solved? I have encountered a similar problem as [#5307](https://github.com/IntelRealSense/librealsense/issues/5307). Would you like to provide some relevant advice?