realsenseai/librealsense #1089: Multiple cameras in pyrealsense2

holstenholsten, 2018-01-28:
| Required Info  |   |
|---|---|
| Camera Model | SR300  | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | macOs 10.12.6 |
| SDK Version | 2.8.2 |

Hi! Is there any way to capture depth and rgb from multiple sensors, using the python wrapper?

zivsha, 2018-01-28:
Hi @holstenholsten ,
What do you mean by multiple sensors?
In general, anything you can do with the C/ C++ API, you can do with the python wrapper. Since all of our tools and example are using the C/C++ API , whatever you can do with them you can do with pyrealsense2

holstenholsten, 2018-01-28:
Thank you for your reply! I meant cameras. Do you know of any example code that is equivalent to the C++ rs-multicam example, but in python? https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

zivsha, 2018-01-29:
Unfortunately no. But the logic of rs-multicam could be easily converted to python.
I think that someone posted some code or link to code that does multicam in Python, so I suggest looking it up. Feel free to ask for help if you have any issue implementing such a script, and if you feel like what you've done, and feel like sharing,we can add it to our python examples.

holstenholsten, 2018-01-31:
Thank you, I have looked into it and it seems doable. 
We have experienced some trouble with connecting more than one camera. Streaming of depth and colour works well for one camera. However, when we have two cameras connected, then only one shows up in device list in python. The same thing happens in C++.
In realsense-viewer, the connected camera disconnects when we plug in a new camera. 
The new camera is then connected, and we can't see the first one anymore. 
Is this a known problem?

zivsha, 2018-01-31:
This usually ends up being a USB port/cable issue, You can try to look up other issues with similar issues and try the suggested solutions. Some major points: are you using USB3? is your cable long (>2m)? is this a USB hub with no external power? does the same thing happen on a different machine?

jlgarcia75, 2018-02-14:
Hello @holstenholsten,

Do you still need help with this issue?

Regards,
Jesus
Intel Customer Support

holstenholsten, 2018-02-14:
Hi @jlgarcia75,
Thank you for checking in. I tried with a powered usb 3.0 hub. The result turned out be be very unexpected and weird:
It does not work when we connect one camera to the hub with external power. But if I then pull out the external power cable, it works! I do not know why this is happening. Maybe there are some capacitors in the hub that makes the power supply even? 
When we connect a second camera, none of the them work (with or with out external power). 
To answer your previous questions: 
we use usb 3.0,
the cable is about one meter,
we have tried on two machines: MacBook Pro (Early 2015) and a Mac Pro.

We think we have the code in python for multiple cameras, but due to the hardware issues, we have not gotten to test it out yet.

As far as I understand, there is only one usb bus in the macs. I also think the usb hub only has one chip-set. I am no expert in the usb-protocol, but could this be an issue? Do you have any ideas about what is happening? 

Kind regards,
Fredrik

RealSense-Customer-Engineering, 2018-03-06:
[Realsense Customer Engineering Team Comment] 
Hi @holstenholstenyo

You can run below command to show the USB port capability on your MAC NB to us.

    $ ioreg -p IOUSB -w0 -l

I suspect it could be the USB bandwidth issue once two D415 cameras both stream. If only one XHCI controller on MAC, it could be the case.
However, recently, just tried one D410 and one D415 both on two separate USB3 ports of one MacBook Air (Early 2014) and used the 2.10.1 realsense-viewer (Add Source option) with the default settings to enable all streams. D410 (IR and Depth) and D415 (IR, Depth, Color) all streams can work well.
Note: Both D410 and D415 firmware version is 5.8.15.

holstenholsten, 2018-03-07:
I have two sr300 connected to an active usb 3.0 hub. It's a lot, but here it is: 

  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x100000352, registered, matched, active, busy 0 (6 ms), retain 11>
    | {
    |   "iManufacturer" = 0
    |   "bNumConfigurations" = 1
    |   "idProduct" = 32775
    |   "bMaxPacketSize0" = 8
    |   "Built-In" = Yes
    |   "iProduct" = 0
    |   "USB Product Name" = "AppleUSBXHCI Root Hub Simulation"
    |   "iSerialNumber" = 0
    |   "bDeviceClass" = 9
    |   "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=4,"CapabilityFlags"=32768,"MaxPowerState"=4,"DriverPowerState"=4}
    |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
    |   "locationID" = 335544320
    |   "bDeviceSubClass" = 255
    |   "bcdUSB" = 256
    |   "non-removable" = "yes"
    |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |   "bDeviceProtocol" = 3
    |   "USB Vendor Name" = "Apple Inc."
    |   "Device Speed" = 3
    |   "idVendor" = 1452
    |   "IOGeneralInterest" = "IOCommand is not serializable"
    |   "IOClassNameOverride" = "IOUSBRootHubDevice"
    | }
    | 
    +-o Bluetooth USB Host Controller@14300000  <class AppleUSBDevice, id 0x1000005b6, registered, matched, active, busy 0 (10 ms), retain 21>
    |   {
    |     "sessionID" = 10227604755
    |     "iManufacturer" = 1
    |     "bNumConfigurations" = 1
    |     "idProduct" = 33424
    |     "bcdDevice" = 339
    |     "Bus Power Available" = 250
    |     "USB Address" = 4
    |     "bMaxPacketSize0" = 64
    |     "iProduct" = 2
    |     "iSerialNumber" = 0
    |     "bDeviceClass" = 239
    |     "Built-In" = Yes
    |     "locationID" = 338690048
    |     "bDeviceSubClass" = 2
    |     "bcdUSB" = 513
    |     "USB Product Name" = "Bluetooth USB Host Controller"
    |     "PortNum" = 3
    |     "non-removable" = "yes"
    |     "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |     "bDeviceProtocol" = 1
    |     "IOUserClientClass" = "IOUSBDeviceUserClientV2"
    |     "IOPowerManagement" = {"DevicePowerState"=2,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
    |     "kUSBCurrentConfiguration" = 1
    |     "Device Speed" = 1
    |     "USB Vendor Name" = "Broadcom Corp."
    |     "idVendor" = 1452
    |     "IOGeneralInterest" = "IOCommand is not serializable"
    |     "IOClassNameOverride" = "IOUSBDevice"
    |   }
    |   
    +-o USB2.0 Hub@14200000  <class AppleUSBDevice, id 0x10000151e, registered, matched, active, busy 0 (3 ms), retain 13>
    | | {
    | |   "sessionID" = 18652721430821
    | |   "iManufacturer" = 1
    | |   "bNumConfigurations" = 1
    | |   "idProduct" = 1552
    | |   "bcdDevice" = 36914
    | |   "Bus Power Available" = 250
    | |   "USB Address" = 17
    | |   "bMaxPacketSize0" = 64
[... 286 lines cut ...]
              "bcdDevice" = 12528
              "Bus Power Available" = 450
              "USB Address" = 26
              "bMaxPacketSize0" = 9
              "iProduct" = 2
              "iSerialNumber" = 0
              "bDeviceClass" = 239
              "Built-In" = No
              "locationID" = 341053440
              "bDeviceSubClass" = 2
              "bcdUSB" = 768
              "USB Product Name" = "Intel(R) RealSense(TM) Camera SR300"
              "PortNum" = 1
              "non-removable" = "yes"
              "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
              "bDeviceProtocol" = 1
              "IOUserClientClass" = "IOUSBDeviceUserClientV2"
              "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=4,"CapabilityFlags"=32768,"MaxPowerState"=4,"DriverPowerState"=4}
              "kUSBCurrentConfiguration" = 1
              "Device Speed" = 3
              "USB Vendor Name" = "Intel(R) RealSense(TM) Camera SR300"
              "idVendor" = 32902
              "IOGeneralInterest" = "IOCommand is not serializable"
              "IOClassNameOverride" = "IOUSBDevice"
            }

sharpriy, 2018-05-14:
Hi...
Is this issue resolved? Could you pls share the python script to capture frames parallely from multiple D415 using pyrealsense2 
I have tried with one and it's fine with that

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
Hi @sharpriy,

You can refer to below to see if this code resolves your issue.
https://github.com/IntelRealSense/librealsense/issues/1735

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @sharpriy,

do you need more support on this topic?

closed 2018-07-26 by dorodnic

ivomarvan, 2021-05-19:
I wrote an example of connecting multiple Realsense cameras in Python:
https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras
Enjoy it if you find it useful.