realsenseai/librealsense #1090: Application of patch 10-realsense-camera-formats-4.4.patch failed.

arunavanag591, 2018-01-29:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04.03 |
| Kernel Version (Linux Only)  |  4.4.0-109-generic |
| SDK Version | |

Hi I am following the installation instruction as given [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). When I do 

`sudo apt-get install realsense-uvcvideo`
I am thrown with the following failed message:

```
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/realsense-uvcvideo.0.crash'
Error! Application of patch 10-realsense-camera-formats-4.4.patch failed.
Check /var/lib/dkms/realsense-uvcvideo/4.10.0/build/ for more information.
Processing triggers for shim-signed (1.32~16.04.1+0.9+1474479173.6c180c6-1ubuntu1) .
```

ev-mp, 2018-01-29:
Hello @arunavanag591 ,
The warning you received is due to the fact that the DKMS module tried to installed both for 4.4 and 4.10 even if it not present. Note that while we need improve the DKMS flow the warning does not affect the installation of hte driver for kernel 4.4 as apparently it succeeded
> USB Video Class driver (**1.1.1-3-realsense**)

As for the device not being recognized - R200 is supported on the [legacy](https://github.com/IntelRealSense/librealsense/tree/legacy) branch with librealsense version1.  Please switch source base and proceed accordingly

ev-mp, 2018-02-04:
Hi @arunavanag591 , I'm closing the ticket now. 
Feel free to reopen in case you need more assistance.

closed 2018-02-04 by ev-mp

arunavanag591, 2018-02-05:
Thanks @ev-mp , I think its resolved for now.