realsenseai/librealsense #1091: The connected device doesn't support depth streaming!

anson2004110, 2018-01-29:
| Required Info  |   |
|---|---|
| Camera Model | D435 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  Linux firefly 4.4.77  |
| SDK Version | 2.0 |

development platform:  RK3399  http://wiki.t-firefly.com/index.php/Firefly-RK3399

we recomplied the librealsense from SDK 2.0 successfully.
But when we run the depth sample, we got the error like:

firefly@firefly:~/Downloads/librealsense-master/librealsense-master/examples/C/depth$ sudo ./rs-depth
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense USB2
    Serial number: 747112070266
    Firmware version: 05.08.15.00

The connected device doesn't support depth streaming!

so what is the problem and how to resolve it?

anson2004110, 2018-01-29:
and we also run the capture sample, the log would be useful.

root@firefly:/home/firefly/Downloads/librealsense-master/librealsense-master/examples/capture# ./rs-capture 
rs-capture: /home/firefly/Downloads/librealsense-master/librealsense-master/src/pipeline.cpp:469: void librealsense::pipeline::unsafe_start(std::shared_ptr<librealsense::pipeline_config>): Assertion `profile->_multistream.get_profiles().size() > 0' failed.
Aborted (core dumped)

ev-mp, 2018-01-29:
Hello @anson2004110 , you have the device recognized as USB2 by host.
> Using device 0, an Intel RealSense USB2

D400 are designed to run with USB3 connection. In USB2 the device provides a limited subset of functionalities/streaming profiles due to bandwidth/power constrains.

Please verify the HW setup - in most cases the improper enumeration is due to HW issues such as cables/insufficient power/USB port in host. Connection the device via self-powered USB3 hub can also help to mitigate power consumption issues.

anson2004110, 2018-01-30:
hi
We restart the system, and the problem was resolved.
Please close this issue.

closed 2018-01-30 by anson2004110

kingsj0405, 2018-04-30:
For more information, you can check connected as USB2 or USB3 by popup message on RealSense Viewer when connected.

fadonoso, 2018-06-15:
Hi @anson2004110,

could you give us guidelines of how to compile librealsense2 on RK3399, I'm having many issues with kernel version and openGl.

Best regards,
Felipe D.

SunQiChang, 2018-10-16:
hi @fadonoso 
do you get guidelines of how to compile librealsense2 on RK3399?

fadonoso, 2018-10-16:
Hi,

Not at all. Sorry mate.

holinfier, 2020-02-20:
Hi @anson2004110,
I compiled libsense-2.25.0 on Firefly RK3399,but it seems like the D435i doesn't connect to RK3399.
Did I miss some import step?