realsenseai/librealsense #1094: Can i get two rgb image, left and right stereo image. not a rgb image and a depth image.

xiaopenghuang, 2018-01-30:
The device is D410 .

dorodnic, 2018-01-30:
Hello @xiaopenghuang 
The D410 offers the following stream combinations: Depth+Left IR (Y8), Depth+Left RGB, Depth + Left IR + Right IR (you can also configure Depth + Right IR, but it is emulated in software if I recall correctly)

xtrawurst, 2018-01-30:
Can you provide some code on how to obtain the right IR frame from a frameset if the configuration is  Depth + Left IR + Right IR?

Methods rs2::frameset::first(RS2_STREAM_INFRARED) and rs2::frameset::get_color_frame() would probably not work here. I think those would return the left IR image.

dorodnic, 2018-01-30:
Yes,
Since the use of the Right stream is rather limited there is no helper function to retrieve it. You can, however, iterate over the frameset and distinguish "Right" IR using the `stream_index`. Something like:
```cpp
auto frameset = pipe.wait_for_frames();
rs2::frame right;
for (auto&& frame : frameset)
{
    if (frame.get_profile().stream_index() == RS2_STREAM_INFRARED &&
        frame.get_profile().stream_index() == 2)
    {
         right = frame;
    }
}
return right.as<rs2::video_frame>();
```

xtrawurst, 2018-02-01:
Thanks a lot. I didn't realize you can just iterate over frames in a for loop.
Is the stream_index for the right stream actually == 2? If yes, what's stream index 0 and 1?

zivsha, 2018-02-01:
@xiaopenghuang ,
When dealing with multiple streams of the same type they are indexed `[1-N]`.
When dealing with a single stream, it is indexed as `0`.

xtrawurst, 2018-02-01:
@zivsha 
Ok. Not super intuitive but thanks much for the clarification.

zivsha, 2018-02-01:
Maybe, but when you hold a `rs2::frame` you can call:
```cpp
frame.get_profile().stream_name()
```
for a human readable string.

xtrawurst, 2018-02-01:
Ok, so just to clarify:

If my combination is Depth + Left IR, my stream index for the Left IR image is 0.
If my combination is Depth + Right IR, my stream index for the Right IR image is 0.
If my combination is Depth + Left IR + Right IR, my stream index for the Left IR image is 1 and my stream index to the right IR image is 2.

Correct?

xtrawurst, 2018-02-01:
@dorodnic 
Just a minor thing: In your code

`frame.get_profile().stream_index() == RS2_STREAM_INFRARED`

should probably be

`frame.get_profile().stream_type() == RS2_STREAM_INFRARED`

zivsha, 2018-02-01:
I update @dorodnic 's code.

The streams index as you said is not very intuitive, there is not correlation between index of the stream and the other streams that are streamed from the same sensor.
The way to start streaming are:
1. Using `sensor.open(stream_profile)`
    If you use the first option, you actually only choose the index of the stream from the collection you'll get from `sensor.get_stream_profiles()` - there is no way to create a stream profile yourself. So if you choose: `Infrared #2 (Video Stream: Y8 640x480@ 30Hz)` and `Depth #0 (Video Stream: Z16 640x480@ 30Hz)` you know which profiles you chose.

2. Using `config.enalbe_stream(...+5 overloads)` and `pipeline.start(config)`.
    Using this option you need to specify which index of the stream you want, or pass a wild card.
    If pass a wildcard then we assume you only want 1 steam of that type, and if you want two of the same type, you need to pass `1` and `2`.

Using `rs-sensor-control` I've printed out the list of available profiles for the D410:

```
Sensor provides the following stream profiles:
0  : Infrared #2 (Video Stream: Y16 1920x1080@ 25Hz)
1  : Infrared #1 (Video Stream: Y16 1920x1080@ 25Hz)
2  : Infrared #1 (Video Stream: Y8 1920x1080@ 25Hz)
3  : Infrared #2 (Video Stream: Y8 1920x1080@ 25Hz)
4  : Infrared #1 (Video Stream: Y16 1920x1080@ 15Hz)
5  : Infrared #2 (Video Stream: Y16 1920x1080@ 15Hz)
6  : Infrared #2 (Video Stream: Y8 1920x1080@ 15Hz)
7  : Infrared #1 (Video Stream: Y8 1920x1080@ 15Hz)
8  : Infrared #0 (Video Stream: UYVY 1280x720@ 30Hz)
9  : Infrared #0 (Video Stream: Y16 1280x720@ 30Hz)
10 : Infrared #1 (Video Stream: Y8 1280x720@ 30Hz)
11 : Infrared #2 (Video Stream: Y8 1280x720@ 30Hz)
12 : Infrared #0 (Video Stream: BGRA8 1280x720@ 30Hz)
13 : Infrared #0 (Video Stream: RGBA8 1280x720@ 30Hz)
14 : Infrared #0 (Video Stream: BGR8 1280x720@ 30Hz)
15 : Infrared #0 (Video Stream: RGB8 1280x720@ 30Hz)
16 : Infrared #0 (Video Stream: UYVY 1280x720@ 15Hz)
17 : Infrared #0 (Video Stream: Y16 1280x720@ 15Hz)
18 : Infrared #1 (Video Stream: Y8 1280x720@ 15Hz)
19 : Infrared #2 (Video Stream: Y8 1280x720@ 15Hz)
20 : Infrared #0 (Video Stream: BGRA8 1280x720@ 15Hz)
21 : Infrared #0 (Video Stream: RGBA8 1280x720@ 15Hz)
22 : Infrared #0 (Video Stream: BGR8 1280x720@ 15Hz)
23 : Infrared #0 (Video Stream: RGB8 1280x720@ 15Hz)
24 : Infrared #0 (Video Stream: UYVY 1280x720@ 6Hz)
25 : Infrared #0 (Video Stream: Y16 1280x720@ 6Hz)
26 : Infrared #1 (Video Stream: Y8 1280x720@ 6Hz)
27 : Infrared #2 (Video Stream: Y8 1280x720@ 6Hz)
28 : Infrared #0 (Video Stream: BGRA8 1280x720@ 6Hz)
29 : Infrared #0 (Video Stream: RGBA8 1280x720@ 6Hz)
30 : Infrared #0 (Video Stream: BGR8 1280x720@ 6Hz)
31 : Infrared #0 (Video Stream: RGB8 1280x720@ 6Hz)
32 : Infrared #2 (Video Stream: Y16 960x540@ 25Hz)
33 : Infrared #1 (Video Stream: Y16 960x540@ 25Hz)
34 : Infrared #1 (Video Stream: Y16 960x540@ 15Hz)
35 : Infrared #2 (Video Stream: Y16 960x540@ 15Hz)
36 : Infrared #0 (Video Stream: UYVY 848x480@ 90Hz)
37 : Infrared #0 (Video Stream: Y16 848x480@ 90Hz)
38 : Infrared #1 (Video Stream: Y8 848x480@ 90Hz)
39 : Infrared #2 (Video Stream: Y8 848x480@ 90Hz)
40 : Infrared #0 (Video Stream: BGRA8 848x480@ 90Hz)
41 : Infrared #0 (Video Stream: RGBA8 848x480@ 90Hz)
42 : Infrared #0 (Video Stream: BGR8 848x480@ 90Hz)
43 : Infrared #0 (Video Stream: RGB8 848x480@ 90Hz)
44 : Infrared #0 (Video Stream: UYVY 848x480@ 60Hz)
45 : Infrared #0 (Video Stream: Y16 848x480@ 60Hz)
46 : Infrared #2 (Video Stream: Y8 848x480@ 60Hz)
47 : Infrared #1 (Video Stream: Y8 848x480@ 60Hz)
48 : Infrared #0 (Video Stream: BGRA8 848x480@ 60Hz)
49 : Infrared #0 (Video Stream: RGBA8 848x480@ 60Hz)
50 : Infrared #0 (Video Stream: BGR8 848x480@ 60Hz)
51 : Infrared #0 (Video Stream: RGB8 848x480@ 60Hz)
52 : Infrared #0 (Video Stream: UYVY 848x480@ 30Hz)
53 : Infrared #0 (Video Stream: Y16 848x480@ 30Hz)
54 : Infrared #1 (Video Stream: Y8 848x480@ 30Hz)
[... 187 lines cut ...]
242: Depth #0 (Video Stream: Z16 848x480@ 15Hz)
243: Depth #0 (Video Stream: Z16 848x480@ 6Hz)
244: Depth #0 (Video Stream: Z16 640x480@ 90Hz)
245: Depth #0 (Video Stream: Z16 640x480@ 60Hz)
246: Depth #0 (Video Stream: Z16 640x480@ 30Hz)
247: Depth #0 (Video Stream: Z16 640x480@ 15Hz)
248: Depth #0 (Video Stream: Z16 640x480@ 6Hz)
249: Depth #0 (Video Stream: Z16 640x360@ 90Hz)
250: Depth #0 (Video Stream: Z16 640x360@ 60Hz)
251: Depth #0 (Video Stream: Z16 640x360@ 30Hz)
252: Depth #0 (Video Stream: Z16 640x360@ 15Hz)
253: Depth #0 (Video Stream: Z16 640x360@ 6Hz)
254: Depth #0 (Video Stream: Z16 480x270@ 90Hz)
255: Depth #0 (Video Stream: Z16 480x270@ 60Hz)
256: Depth #0 (Video Stream: Z16 480x270@ 30Hz)
257: Depth #0 (Video Stream: Z16 480x270@ 15Hz)
258: Depth #0 (Video Stream: Z16 480x270@ 6Hz)
259: Depth #0 (Video Stream: Z16 424x240@ 90Hz)
260: Depth #0 (Video Stream: Z16 424x240@ 60Hz)
261: Depth #0 (Video Stream: Z16 424x240@ 30Hz)
262: Depth #0 (Video Stream: Z16 424x240@ 15Hz)
263: Depth #0 (Video Stream: Z16 424x240@ 6Hz)
```

Hope this answers your question...

xtrawurst, 2018-02-01:
@zivsha 

Almost. Is there any correlation between the Stream Index and whether the stream is real or just computed from another stream?

Also it would be good to have a little bit of documentation on the relationships of
- rs2::sensor
- rs2::stream_profile
- rs2::pipeline
- rs2::config

I am able to find my way around in librealsense and typically can get stuff to stream but I don't really have a good mental model on how it all fits together. Perhaps this documentation exists and I'm just not aware of it (please point me to it if so).

zivsha, 2018-02-01:
> Almost. Is there any correlation between the Stream Index and whether the stream is real or just computed from another stream?

Not that I'm aware of, @dorodnic please correct me if I'm wrong.

Regarding documentation, here's what I remember and copy paste quickly:
* [Docs](https://github.com/IntelRealSense/librealsense/tree/master/doc) - Collection of the documentations, I suggest reading [API Architecture](https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md) and [API How To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To).
* [librealsense2 API](https://github.com/IntelRealSense/librealsense/blob/c8ee8fa1912b9297df13bfe097d527667fe0afba/include/readme.md)    
* [Sensor Control Example](https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control) - should clarify on context/device/sensor/stream

xtrawurst, 2018-02-01:
@zivsha 
If I use the first method using `sensor.open(stream_profile)`, how do I get frames? Can I still use an `rs2::pipeline`?

xiaopenghuang, 2018-02-02:
Thanks, i will try it.

closed 2018-02-13 by xiaopenghuang