realsenseai/librealsense #1095: librealsense / ros-wrapper issue - Failed to read busnum/devnum.

compiaffe, 2018-01-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.14.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2 on Auvidea J120 **headless** |
| SDK Version                            |  2.9.1 / development branch                       | 

### Issue Description
I am not 100% sure this is a librealsense or a ros-wrapper issue:
When starting the the basic `roslaunch realsense_ros_camera rs_aligned_depth.launch` the node starts, but _does not publish any images and warns about_:
```
Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
...
 (sensor.cpp:313) Unregistered Media formats : [ UYVY ]; Supported: [ ]
```
or in full beauty:
```
 Initializing nodelet with 4 worker threads.
[ INFO] [1517279953.613911935]: RealSense ROS v2.0.1
[ INFO] [1517279953.613994846]: Running with LibRealSense v2.9.1
 30/01 02:39:13,619 WARNING [547856953344] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 30/01 02:39:13,620 WARNING [547856953344] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 30/01 02:39:13,844 WARNING [547856953344] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 30/01 02:39:13,844 WARNING [547856953344] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
[ INFO] [1517279954.009781809]: getParameters...
[ INFO] [1517279954.216205233]: setupDevice...
[ INFO] [1517279954.216384815]: JSON file is not provided
[ INFO] [1517279954.216587437]: ROS Node Namespace: camera
[ INFO] [1517279954.216731916]: Device Name: Intel RealSense 435
[ INFO] [1517279954.217056936]: Device Serial No: 728312070140
[ INFO] [1517279954.217155015]: Device FW version: 05.08.14.00
[ INFO] [1517279954.217299430]: Device Product ID: 0B07
[ INFO] [1517279954.217383781]: Enable PointCloud: Off
[ INFO] [1517279954.217641058]: Align Depth: On
[ INFO] [1517279954.217719777]: Sync Mode: On
[ INFO] [1517279954.218952500]: Device Sensors: 
[ INFO] [1517279954.220034281]: Stereo Module was found.
[ INFO] [1517279954.220422789]: RGB Camera was found.
[ INFO] [1517279954.221715575]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1517279954.221895605]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1517279954.222189714]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1517279954.222598990]: setupPublishers...
[ INFO] [1517279954.312323900]: setupStreams...
 30/01 02:39:15,103 WARNING [547856953344] (sensor.cpp:313) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1517279955.111905038]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1517279955.112951331]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1517279955.113698331]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1517279955.131979322]: color stream is enabled - width: 640, height: 480, fps: 30
 30/01 02:39:17,124 WARNING [547331502480] (backend-v4l2.cpp:809) Empty frame has arrived.
 30/01 02:39:17,124 WARNING [547331502480] (backend-v4l2.cpp:809) Empty frame has arrived.
 30/01 02:39:17,124 WARNING [547331502480] (backend-v4l2.cpp:809) Empty frame has arrived.
 30/01 02:39:17,124 WARNING [547331502480] (backend-v4l2.cpp:809) Empty frame has arrived.
[ INFO] [1517279957.126727924]: publishStaticTransforms...
[ INFO] [1517279957.127601099]: RealSense Node Is Up!
 30/01 02:39:18,849 WARNING [547531403664] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 30/01 02:39:18,849 WARNING [547531403664] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 30/01 02:39:22,128 WARNING [547331502480] (backend-v4l2.cpp:827) Frames didn't arrived within 5 seconds
 30/01 02:39:23,055 WARNING [547856953344] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1517279963.056012633]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error: 
 30/01 02:39:23,866 WARNING [547531403664] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 30/01 02:39:23,866 WARNING [547531403664] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 30/01 02:39:27,132 WARNING [547331502480] (backend-v4l2.cpp:827) Frames didn't arrived within 5 seconds
^C 30/01 02:39:28,870 WARNING [547531403664] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
```
----

The `librealsense` has been build from source and enumerates,
```
$ rs-enumerate-devices 
 Device info: 
    Name                          : 	Intel RealSense 435
    Serial Number                 : 	728312070140
    Firmware Version              : 	05.08.14.00
    Physical Port                 : 	/sys/devices/3530000.xhci/usb2/2-1/2-1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
```

and measure distance. However, it sometimes hangs/works on second execution
```
$ rs-distance 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense 435
    Serial number: 728312070140
    Firmware version: 05.08.14.00

rs_error was raised when calling rs2_pipeline_wait_for_frames(pipe:0x431830):
    Frame didn't arrived within 5000

$ rs-distance 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense 435
    Serial number: 728312070140
    Firmware version: 05.08.14.00

The camera is facing an object 0.962 meters away.
The camera is facing an object 0.953 meters away.
The camera is facing an object 0.955 meters away.
The camera is facing an object 0.958 meters away.

```

----

The udev rules are installed:
```
$ ls -alh /dev/video*
crw-rw-rw-+ 1 root plugdev 81, 0 Jan 30 02:22 /dev/video0
crw-rw-rw-+ 1 root plugdev 81, 1 Jan 30 02:22 /dev/video1
crw-rw-rw-+ 1 root plugdev 81, 2 Jan 30 02:22 /dev/video2
```
The user is in the right groups:
```
$ groups
nvidia adm dialout cdrom floppy sudo audio dip video plugdev
```
----
The kernel patches were installed using a modified `patch-realsense-ubuntu-xenial.sh` script as per the instructions in `docs/install_jetson.md`:
https://github.com/aivero/librealsense/blob/jetson-tx2-install/scripts/patch-realsense-ubuntu-xenial.sh#L26

It is based on a slightly modified version of the install script by JetsonHacks (now also a submodule in above mentioned repo): https://github.com/aivero/buildJetsonTX2Kernel/blob/fix/jetson_tx2/scripts/getKernelSources_noGUI.sh
The 2 differences are that I am now pulling v2.0 of the r28 L4T and that I am skipping the (GUI) kernel configuration. We only want the kernel sources I understand.

----
Any ideas what this could be?
Thanks a lot :)

freemanlo, 2018-01-30:
Suggest to use L4T 28.2 BSP instead of the L4T 28.1 BSP which has the stability issue. Probably, you see sometimes the librealsense app can run at the 2nd try. You can also apply the patch for 28.2.

icarpis, 2018-01-30:
Hello @compiaffe, thank you for the detailed information.
It seems that the ROS-Wrapper haven't published the topics because the color, depth and IR frames haven't provided by the camera (according to this print: "Frame didn't arrived within 5000"). I managed to reproduce this issue and found that it's a firmware bug which going to be fixed at the next versions. As a w/a please try to unplug and plug back the camera prior you running the Launch file.

compiaffe, 2018-01-31:
@freemanlo I am using 28.2 already: https://github.com/aivero/buildJetsonTX2Kernel/blob/fix/jetson_tx2/scripts/getKernelSources_noGUI.sh#L6

@icarpis great that you could already identify a firmware bug.
Regarding the intermittent connection from `rs-distance` replugging it works.

However, it does not work around the issues when starting a ROS node:
`roslaunch realsense_ros_camera rs_aligned_depth.launch`

Did you reproduce the ROS behaviour or the `rs-distance`? Which one will be solved with the firmware fix?

icarpis, 2018-01-31:
I reproduced the ROS behavior and I have w/o it by replugging the camera every time before using the Launch file.

compiaffe, 2018-01-31:
OK, I have not been able to work around it on the Jetson TX2.

However, thinking about it I was already using that workaround on the laptop.

Still the issues on the Jetson is a bit of a mystery. Could it be connected to the 
`Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0` warning?

icarpis, 2018-02-04:
The `busnum/devnum` prints are not connected to this issue.
Is there a chance that you can setup a USB controller card on your Jetson?
I installed a Renesas USB3 controller card and now it looks quite stable.
[image]

compiaffe, 2018-02-05:
I am running on a J120 carrier board by Auvidea: https://auvidea.com/j120/ so trying a PCI USB controller card is not a quick option.

I re-installed the kernel patches by running the following three commands from the new branch `realsense_patching` of https://github.com/aivero/buildJetsonTX2Kernel/tree/realsense_patching

```
$ cd buildJetsonTX2Kernel
$ ./getKernelSources.sh
$ ./ makeKernel.sh
$ ./copyImage.sh
```

And installed the newest wrapper

----

As mentioned in your workaround - replugging the camera or simply restarting the rosnode works now.
Messages on  `/camera/depth/image_rect_raw` get published on every second run of `roslaunch realsense_ros_camera rs_aligned_depth.launch`

So that is fixed.
----

However, I still get the same 
- `Frame didn't arrive` warnings
- `Unregistered Media formats : [ UYVY ]`
- `Fisheye, 0) sensor isn't supported by current device! -- Skipping...`

ykim104, 2018-02-06:
I also get frame didn't arrived in 5seconds wanrings. And there is no streaming GUI for me to see if I get images. Does this means it is publishing empty image? My project requires me to subscribe to publisher and record stream of images. Please let me know how this can be done. Thank you!

ykim104, 2018-02-06:
If I'm not getthing those warnings, does that mean it is publishing image topics? There isn't supposed to be streaming GUI, correct? Thank you!

compiaffe, 2018-02-07:
@ykim104 What do you mean by `no streaming GUI`?
You should be able to see the all the video streams (infra1, infra2, depth, RGB) in the viewer that is installed with the librealsense:
```
$ realsense-viewer
```

In **ROS**, you can check if data arrives by first starting the node with i.e. 
```
roslaunch realsense_ros_camera rs_aligned_depth.launch
```
and then checking the rate of messages on one of the topics:
```
rostopic hz /camera/depth/image_raw
```
If you are looking for a pointcloud keep in mind that those are turned off in most of the launch files.

compiaffe, 2018-02-07:
@icarpis Coming back to the original issue here, the 
```
Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
```
Do you have suggestions on how this can be fixed?

icarpis, 2018-02-08:
@compiaffe, as I already mentioned, the message of `Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0` is not relating to any of the issues you are facing with. 
Regarding the `Frame didn't arrive` warning, right now as a w/o I can only suggest you to replugging the camera *each time* before running the Launch file.
Are there any cases that replugging the camera is not helping?

compiaffe, 2018-02-08:
@icarpis Replugging or restarting works. However, I cannot not open the fisheye and UYVY is missing. I will open a new issue for this though. Thank you for the help :)

closed 2018-02-08 by compiaffe

jetsonhacks, 2018-03-30:
The device path: /sys/bus/iio/devices/iio:device0 (actually iio:device0 through ii:device3) is the ina3221x power monitor on board the Jetson module. The library code here makes the assumption that there is an associated bus number and device number in the directory structure, and issues the warning when it can not find it. I believe that the power monitor is on an internal i2c bus which is not advertised. The warning can be ignored, or guard code can be placed for that particular instance.