realsenseai/librealsense #1098: SR300 Camera could not be found and driver installation fails

mirhec, 2018-01-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | I can't click info because no camera was detected | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.9.1                          | 

### Issue Description
No camera is displayed in the Realsense Viewer and in the device manager there is no driver installed. But the cameras are named correctly. If I start Skype for example, the cameras are found and I can see an image. But in the Realsense viewer I see nothing. I tried to update the device drivers, but it gives me an error. Here is the event:

```xml
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
  <Provider Name="Microsoft-Windows-UserPnp" Guid="{96F4A050-7E31-453C-88BE-9634F4E02139}" /> 
  <EventID>20001</EventID> 
  <Version>0</Version> 
  <Level>4</Level> 
  <Task>7005</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8000000000000000</Keywords> 
  <TimeCreated SystemTime="2018-01-30T10:00:57.420874000Z" /> 
  <EventRecordID>1291</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="16664" ThreadID="27600" /> 
  <Channel>System</Channel> 
  <Computer>DESKTOP-PSOS7V2</Computer> 
  <Security UserID="S-1-5-18" /> 
  </System>
<UserData>
<InstallDeviceID xmlns="http://manifests.microsoft.com/win/2004/08/windows/userpnp">
  <DriverName>realsensesr300depth.inf_amd64_c0b85d3599a4f443\realsensesr300depth.inf</DriverName> 
  <DriverVersion>3.3.27.5718</DriverVersion> 
  <DriverProvider>Intel(R) Corporation</DriverProvider> 
  <DeviceInstanceID>USB\VID_8086&PID_0AA5&MI_02\6&2BE25EA&0&0002</DeviceInstanceID> 
  <SetupClass>{6BDD1FC6-810F-11D0-BEC7-08002BE2092F}</SetupClass> 
  <RebootOption>false</RebootOption> 
  <UpgradeDevice>true</UpgradeDevice> 
  <IsDriverOEM>true</IsDriverOEM> 
  <InstallStatus>0x643</InstallStatus> 
  <DriverDescription>Intel(R) RealSense(TM) Camera SR300 Depth</DriverDescription> 
  </InstallDeviceID>
  </UserData>
</Event>
```

Franziska-Mueller, 2018-01-30:
I am having a similar issue.
The device manager (Windows 10) displays both RGB and Depth camera and I can see the RGB video in Skype. But the Depth Camera Manager as well as the Realsense Viewer do not detect any camera.
I tried running the rs_enumerate_devices.exe and it gives me the following error:

ERROR [220] (types.h:178) CoCreateInstance(CLSID_SensorManager, NULL, CLSCTX_INPROC_SERVER, IID_PPV_ARGS(&pSensorManager)) returned: HResult 0x800704ec: "This program is blocked by group policy. For more information, contact your system administrator."

I could not find anything in the group policies but I am also not sure what to look for exactly. Could you check if you get the same error when running this exe?

mirhec, 2018-01-31:
No if I run this rs-enumerate-devices.exe I get "No device detected. Is it plugged in?". I have an other laptop that works with this camera (also Win10). It has three devices in the device manager:

- Intel RealSense Camera SR300 Depth
- Intel RealSense Camera SR300 RGB
- Intel RealSense Camera SR300 Virtual Driver

On my PC I only see the Depth device ...

Franziska-Mueller, 2018-01-31:
I found some information about this on the Intel website:
- https://www.intel.com/content/www/us/en/support/articles/000023694/emerging-technologies/intel-realsense-technology.html?_ga=2.46771751.1469966166.1517393916-879225375.1517393885
- https://communities.intel.com/community/tech/realsense/blog/2017/04/07/windows-10-creators-update-build-15063-is-not-compatible-with-intel-realsense-sr300-camera-driver
- https://www.intel.com/content/www/us/en/support/articles/000023716/emerging-technologies/intel-realsense-technology.html?_ga=2.145663606.332834236.1517393925-879225375.1517393885

It sounds like starting from Windows 10 Creators Update build 15063 you need to be lucky that the driver installation works.

Franziska-Mueller, 2018-01-31:
I managed to install the driver by:
- deinstalling the driver
- shutting down Windows (important: not restarting)
- turning on Windows
- installing the driver

Maybe that helps you. But ideally, this should get fixed by Intel.

mirhec, 2018-01-31:
Thank you for this infos ... I'll try it later, hopefully it'll work for me as well.

mirhec, 2018-02-08:
Unfortunately it did not work for me ...

I have a Realsense 400 camera here, they is working without problems, but my SR300 doesnt show up correctly in the device manager.

Please help me, because it is urgent for us to make it working!

zivsha, 2018-02-08:
Hi @mirhec ,
The issues you are describing sounds like driver related issues. Can you please tell us:
1. The exact windows version you are using (`WinKey + R` (run) --> `winver` --> Alt+PrtScn (takes a screenshot of that window - or use `snippingtool`) --> Ctrl+V on Github)
2. Is DCM installed? Open `Programs and Features ( `WinKey + R`--> `appwiz.cpl`) and check if you see `Intel RealSense Depth Camera Manager SR300`, if so- which versions? (Since you have the `SR300 Virtual Driver` I assume you have this installed)
3. What is the version of the Depth, and RGB drivers (in `Device Manager`)

mirhec, 2018-02-08:
Here are the required information:

**1. The exact windows version you are using (WinKey + R (run) --> winver --> Alt+PrtScn (takes a screenshot of that window - or use snippingtool) --> Ctrl+V on Github)**

[image]

---

**2. Is DCM installed? Open Programs and Features (WinKey + R--> appwiz.cpl) and check if you see Intel RealSense Depth Camera Manager SR300, if so- which versions? (Since you have the SR300 Virtual Driver I assume you have this installed)**

DCM is not installed, I only have installed the Intel RealSense SDK 2.0 version 2.9.1. The problem is that I can't install DCM ... it tries to install:

[image]

But after a while the following error occurs:

[image]

---

**3. What is the version of the Depth, and RGB drivers (in Device Manager)**
The only device I see when I connect a SR300 is the USB device and the depth camera:
[image]

The drivers are not really installed, if I click on details it states me I need to install the drivers:
[image]
[image]
[image]
[image]

---

**4. In Skype - when you choose the Depth camera, do you see a green screen?**
No in Skype I don't see the Depth camera, but only the RGB camera:
[image]

zivsha, 2018-02-08:
I just noticed that your comment regarding seeing 3 devices in the device manager refers to another laptop. So this is indeed a driver issue, There is a post online referring to how to uninstall the driver and reinstalling it (which should be performed automatically via Windows Update), I will look it up for you and post again

zivsha, 2018-02-08:
Before trying the suggestions in that post, please note that you do not need DCM (Depth Camera Service) installed, RealSense SDK 2.0 works without it, and in fact it is preferred if DCM is absent.
In general what you should do is disconnect the camera, uninstall and completely remove the SR300 driver, and then connect the camera and reinstall the driver (hopefully Windows Update should take care of that by itself).

I've found this guide which might help the issue (https://communities.intel.com/thread/113973), please take extra precaution as some of the actions there are advanced.

mirhec, 2018-02-08:
I can't say why and how, but I managed it to work again. I tried to follow this description you posted, but at first it doesn't seem to work. At some time the camera was listed under "System devices" and I uninstalled it from this location, reconnected it and voila - it worked!

Thank you for your help! I'll close this issue.

closed 2018-02-08 by mirhec

eebanos, 2019-01-25:
Still no solution for this issue???, same problem here.

Yhaws, 2019-09-02:
> I managed to install the driver by:
> 
> * deinstalling the driver
> * shutting down Windows (important: not restarting)
> * turning on Windows
> * installing the driver
> 
> Maybe that helps you. But ideally, this should get fixed by Intel.

Made a account just to say "thank you", this worked for me. Sooo... thank you (y)