realsenseai/librealsense #1103: Aligning D410 with Platform Camera

jb455, 2018-01-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 05.08.14.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC (C#.NET)  |
| SDK Version                            |  2.9.1 | 

Hi,

I have a device which has a D410 depth module with no RealSense colour camera, but it does also have a rear-facing webcam. I want to be able to map from the colour photo from the platform camera (webcam) to the depth data so I can tap on the colour image to get the corresponding depth value. The physical distance between the right lens of the depth camera and the colour camera is approx. (93mm, 17mm, 10mm), and they have different focal lengths, fields of view etc. I am using C# to further complicate the matter.

I've gotten the extrinsic calibration between the two cameras, as well as their individual intrinsics, using OpenCv; and I've gotten them both streaming at the same time (using `Sensor.Start` for the platform camera and `Pipeline.Start` for depth) but I'm struggling with doing the actual mapping.

I have tried several approaches: reverse engineering the `Align` class & implementing` align_images` myself in C# (wasn't mapping correctly & was extremely slow); exposing a new entry point into the existing `Align` class allowing me to inject my own extrinsics (`align.Process` always times out); and finally, using the new `SoftwareDevice` interface, which I've had a go at wrapping myself and seems fine up until I pass my `Syncer` object to `Sensor.Start` where I get a compile error that `Syncer` can't convert to `FrameQueue` or `FrameCallback`.

So anyway, my real question is: do any of these approaches sound reasonable? In your opinion, do you think what I want to do is possible with the current SDK (or with relatively small modifications to it)? 
It's entirely likely that I've made mistakes while modifying the SDK as I'm not very well versed in C++ or interop stuff, but I don't know if it's worth me spending the time going into it if it turns out that either I'll need to do loads of extra stuff beyond just getting the wrapper working, or you guys are working on something that'll make it really easy, or the differences between the cameras makes it a really hard problem to solve without lots of expertise (that I lack!)

Thanks for any help or advice you may be able to offer.
James

dorodnic, 2018-01-31:
Hello James,

Your approach is absolutely reasonable. We will do our best to assist. While this is not a flow we currently support out of the box, we would like to add documentation and reference code on this subject in the future.  
At the moment the C# wrapper does not expose 100% of the functionality, but it should be easy enough to fill the gaps. 
The way I see it - you want to create a software device with relevant intrinsics and register extrinsics (we will add the relevant methods to the wrapper), and pass depth frame + last frame from the platform camera to a `syncer`. The output from the `syncer` can then be passed to `align` or `pointloud` processing blocks.

jb455, 2018-01-31:
Thanks for the reassurance @dorodnic! 

I've tried exposing most of it already (I have a fork here: https://github.com/jb455/librealsense/ but unfortunately my git-fu failed me and the merge went a bit screwy so it's hard to see all I've changed: it's mostly in [wrappers/csharp/src](https://github.com/jb455/librealsense/tree/master/wrappers/csharp/src)).

The problems I had were: 

- passing the `deleter` delegate in the [`SoftwareVideoFrame`](https://github.com/jb455/librealsense/blob/master/wrappers/csharp/src/Types.cs#L22) struct used in SoftwareSensor.OnVideoFrame (I fudged it so if I pass `IntPtr.Zero` the [C++ code](https://github.com/jb455/librealsense/blob/master/include/librealsense2/hpp/rs_internal.hpp#L97) will replace that with `[](void*) {}` which is what's used in the C++ sample) which stops it crashing when it tries to access the deleter but probably isn't ideal!

- the [Syncer class](https://github.com/jb455/librealsense/blob/master/wrappers/csharp/src/Processing.cs#L115). The `Syncer` object I use I get from `rs2_create_sync_processing_block` similar to `Align` & `Colorizer` - should it be something other than a `ProcessingBlock` so it can be used as a `FrameCallback` in `Stream.Start`?

Also, will I need to rectify my images before mapping? I couldn't find any references to rectification in the source, but thought maybe it normally does that on the device?

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
hi @jb455,

I didn't see anything relative to this in v2.11.0 release note, but would like to confirm with you if the issue still the same for your usage in latest release?
thanks.

jb455, 2018-05-11:
Hi,
It looks as though the new FrameSource class will be useful to 'bundle' the two frames together (negating the need for a separate syncer I think), but there doesn't appear to be a way to register extrinsics between them. Getting there though!

Thanks
James

PS: I get a couple of errors when trying to build the wrapper from a freshly-cloned copy of source at the moment: "Your project.json doesn't have a runtimes section. You should add '"runtimes": { "win-x64": { } }' to your project.json and then re-run NuGet restore." and "The command "CMake --version" exited with code 9009." (opening the vs15 solution in vs17 because I don't want .net core)

dorodnic, 2018-05-30:
Hi @jb455 
Sorry for not getting there faster. We still have to wrap the software device and offer some sample code. 
We are still interested in supporting this use-case, just a question of time and resources. 
I wasn't able to reproduce the build problem, please check that you have cmake and it is reachable from command line.

jb455, 2018-06-15:
Hi, 
I did have a look at the dynamic calibration tool and whilst it looked promising, it doesn't appear as though you can enable the platform camera in the tool to obtain the extrinsic calibration between the Realsense depth camera and the platform colour. I have gotten calibration code working through OpenCV though. WIth the API which allows you to write custom calibrations to a camera, would that allow me to write my extrinsics to the D410 module and have it use that when mapping to the platform camera?

I've now been trying to get this done in C++ as a 'proof of concept' to check it does work whilst waiting for the relevant parts to be included in the wrapper, but I'm having problems.

You can see what I've tried here: https://gist.github.com/jb455/5ed20c1d77dd8695799bdf8a58d8a545, starting from the rs-capture app and incorporating parts from a few other samples including rs-software-device. Note I'm a C++ novice so there's bound to be better ways of doing what I've done!

My problem is that I can only ever get one image from the syncer at line 186. With the code as it is in the gist, only the depth image shows up. If I swap the `dev_color_sensor.on_video_frame` section (#171-175) with `dev_depth_sensor.on_video_frame` (#177-182) so depth is processed before colour, only the colour frame shows up.  I thought it may be a problem with the syncer matcher but I've tried a couple of different matching methods with the same result. I've also tried using a pipeline instead of the sensor to run the depth camera but again, no difference.
Can you see anything I've done wrong?

Thanks!

jb455, 2018-06-18:
I've now managed to get both camera streams out of the software device, by passing the timestamps of the actual frames rather than the synthetic timestamp generated from the frame number. My gist has been updated with this change: https://gist.github.com/jb455/5ed20c1d77dd8695799bdf8a58d8a545/revisions.

However, I'm now getting an exception when trying to align the images. When calling `align.process` on the frameset obtained from the syncer, I get `0x000001a43fcad300 "Error occured during execution of the processing block! See the log for more info"`. I can't find the log file to diagnose further (I did set the variable as suggested in https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-how-do-i-enable-librealsense-logs but can't find where the log file is saved).
Do you know what might be wrong now please?

Thanks

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
hi @jb455,

for log file, i would suggest you to refer to rs-data-collect, having rs2::log_to_file(<log level>); in the first line, then when you run the binary, the log file will be generated at the same folder, then we can check what we have in the log.

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
hi @jb455,

thanks for confirm so i think we can close this ticket now?

jb455, 2018-07-02:
Hi, we still need the software device stuff implemented in the C# wrapper ideally. Me doing the C++ app was just to let us know if it would be worth the wait!
Thanks

jb455, 2018-07-18:
Ok, I'm really struggling here. I hit a wall trying to create the `syncer` object in the wrapper again so now I'm trying to add new functions to the `software_device` api which will create an internal `syncer` and start the `software_sensor`s using that syncer so I can create another function like `software_device::wait_for_frames` to return frames from the `syncer` so I don't need the `syncer` at all in the wrapper. Does that make sense? sound reasonable?

So, in `software-device.h` I want to add a private 'syncer' object. Can I use `rs2::syncer`, or will that cause problems with opening it up to the C api? The  `rs2::syncer` class is defined in `rs_processing.hpp` whereas all the other types used are defined in the header files. The only syncer-type object I can find in the C headers is `syncer_process_unit`, but `software_sensor.start` won't accept that as an argument.

If you could help by just filling in the following please hopefully I'll be alright from there, I can use the template followed by all the other functions to open it out to the API assuming the solution to this won't be completely different to what's come before. I'm on a really tight deadline now and tearing my hair out trying to get this working so any quick help would be hugely appreciated!

```
[software-device.h]
public:
    void start_streaming();

    rs_frame* wait_for_frames();
private:
    //define syncer object. Will one of these work?
    //std::shared_ptr<syncer_process_unit> _syncer;
    //std::shared_ptr<rs2::syncer> _syncer;
```

```
[software_device.cpp]
void software_device::start_streaming()
{
    for (auto&& s : _software_sensors)
        s->start(_syncer);
        //I assume I should convert back from the shared pointer to the actual syncer object but don't know how to.
}

rs2_frame* software_device::wait_for_frames()
{
    return _syncer.WaitForFrames();
}
```

closed 2018-07-24 by jb455