realsenseai/librealsense #1105: High CPU Utilization

jpapon, 2018-02-01:
This is listed in #744 and #707, but since both of those are closed I am opening a new issue.
OpenMP causes librealsense to consume a full CPU core - this is likely due to the use of `#pragma omp parallel for schedule(dynamic)` .

Disabling OpenMP in CMake with `-DBUILD_WITH_OPENMP=false` is a workaround, but this could increase latency at high-framerates/resolutions.

zivsha, 2018-02-01:
OpenMP provides some [environment variables](https://computing.llnl.gov/tutorials/openMP/#EnvironmentVariables) for controlling the number of threads used by the program, I wonder if limiting the number to half the maximum available will lower the CPU load but still provide low latency. Could you have a go at it?

jpapon, 2018-02-01:
Okay, I did some profiling using OpenMP environment variables.
This was done on an  i7-7700, which has 4 physical (8 HT) cores.
The numbers below jump around of course, these are just eyeballed averages.

640x480 @ 30hz, depth & rgb streams enabled
CPU usage (shown in htop) with OMP_NUM_THREADS =
1 - 35% 
2 - 65%
4 - 200%
6 - 345%
8 - 440%

OMP_DYNAMIC=FALSE made no difference
If I build without openmp, CPU usage is the same as OMP_NUM_THREADS=1

1280x720 @ 30hz, rgb and depth streams, 
CPU usage (shown in htop) with OMP+NUM_THREADS=
1 - 65%
2 - 125% 
4 - 220%
6 - 330%
8 - 460%

dorodnic, 2018-02-02:
Hi @jpapon 
The main problem we are trying to solve is RGB streaming in Full-HD. Our current SSE-optimized YUY to RGB conversion is still too taxing to keep 60 (or even 30) FPS and we want to avoid frame-drops. 
We checked jumping to AVX, but not everyone has AVX support at this point, so it would require dynamic CPUID identification and it is hard to do cross-platform. Obviously, on ARM the situation is even more dire (we are looking into potential NEON optimization). That's why OpenMP was chosen as a robust, cross-platform alternative. 
In addition, we know that both the colorizer and the projection code needs to be better optimized.

skohlbr, 2018-02-02:
As mentioned in #707 I saw the same issue with `#pragma omp parallel for schedule(dynamic)` in a different application (here specifically: https://github.com/ros-planning/moveit_ros/issues/563). While it makes sense to try to reduce latency using OpenMP, I'd expect that the enormous additional CPU consumption we see eats up potential improvements in latency (especially if there are other processes doing CPU-heavy computation).

I didn't dig deeper into the problem (instead just remove the OpenMP parallelization), but it seems playing with different OpenMP parallelization options (static vs. dynamic, different chunkSize settings) would be a low hanging fruit to try. See https://software.intel.com/en-us/articles/performance-obstacles-for-threading-how-do-they-affect-openmp-code

jpapon, 2018-02-02:
@dorodnic I see the need for using OpenMP to reduce latency @ HD resolution, but I think there's simply a bug in the usage of OpenMP. Looking at the CPU usage numbers, it looks to me like each thread is replicating the computation of the full frame by itself. It just seems too coincidental to me that each time the thread count is doubled, so is the CPU usage. That or something funky is happening, like using one thread per pixel, so the overhead of adding more threads is outweighing the benefit.

I don't have time to investigate further at the moment, but I agree with @skohlbr that this could probably be fixed with some simple changes to how OpenMP is used.

cynosure4sure, 2018-02-06:
Compiling with -DBUILD_WITH_OPENMP=false, results in not being able to run D435 on my machine with it throwing out the following errors:
1. (Color, 0) sensor isn't supported by current device!
2. Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Format: Z16, Width: 640, Height: 480, FPS: 30
3. Given depth profile is not supported by current device
Re-compiling with OPENMP allows me to see the run the camera but with CPU utilization around 50% across all threads on my machine with i7-6700HQ.

dorodnic, 2018-02-06:
Hi @cynosure4sure 
Compiling without OpenMP should in no way affect the functionality, we test it both on and off all the time. Could it be that during the no-OpenMP run the camera got identified as USB2? (the symptoms you describe are very much indicative of that)

cynosure4sure, 2018-02-06:
Hi @dorodnic 
It was another mistake on my end and OpenMP was not responsible. Sorry my bad.
I am also posting my CPU consumption on ubuntu 14.04, kernel 4.4.0.111 for reference purposes.
[image]

cynosure4sure, 2018-02-07:
Just for reference. Turning off OpenMP when compiling librealsense and ROS realsense camera gives me a CPU usage of 50-60% for grabbing color and depth images only without visualization on an i7-6700HQ.
[image]

shaun-edwards, 2018-02-10:
I can verify the results @jpapon reported [here](https://github.com/IntelRealSense/librealsense/issues/1105#issuecomment-362340811).  I verified that changing environment variables as suggested by @zivsha [here](https://github.com/IntelRealSense/librealsense/issues/1105#issuecomment-362171326) can improve things.  Unfortunately, this won't work for me because I use OpenMP in other programs and their performance is affected.  I plan to try changing the OpenMP line `#pragma omp parallel for schedule(dynamic)` to `#pragma omp parallel for schedule(static)`.  I will report back if it improves things.

Otherwise, +1 for fixing this as it makes doing anything other than running a single camera very slow.

jpapon, 2018-02-10:
I can confirm static will help. From the Intel docs, about ‘schedule (dynamic):

> By default, the chunk size is 1. Be careful when using this scheduling type because of the extra overhead involved.

The loop needs to be structured properly to help avoid cache misses as well. I’ll take a look at it and propose a fix on Monday.

shaun-edwards, 2018-02-10:
I created a PR (to my own fork), [here](https://github.com/plusone-robotics/librealsense/pull/1).  Changing to `static` did not make much of a difference.  I also limited the number of threads used in color conversion, [here](https://github.com/plusone-robotics/librealsense/pull/1/files#diff-158cb5134a6554549a160a6a0995d771R236).  As stated above (or maybe in the other issues), this is the main cause of high usage.  I've changed my camera configuration to publish depths images at a high fps (60) and color images at a lower fps (15).  This reduces the CPU usage to a reasonable level, which works for my application, but is a bit disappointing.

cynosure4sure, 2018-02-10:
@shaun-edwards Can you point me to the changes you made to lower color images fps to 15 because as per the datasheet the minimum publish rate for color and depth cameras is 30fps.

shaun-edwards, 2018-02-10:
I'm using an intel 400 series camera.  I'm using the ROS wrapper and [this](https://github.com/intel-ros/realsense/blob/development/realsense_ros_camera/launch/demo_pointcloud.launch#L14) launch file.  As it turns out, I only see 30fps on the depth image, but I do get 15fps on the color image.

cynosure4sure, 2018-02-10:
@shaun-edwards thank you, I am using the same file. I can lower the fps to 15 for both color and depth camera. I earlier tried to lower it to 10fps and 20fps but I got a warning that the stream configuration was not supported.

jpapon, 2018-02-12:
From a little investigation, it seems to me that the alignment code is just unusable at this point - it eats up all cores, and even then, is still unable to maintain 30FPS.

What's more, it looks like it corrupts the depth image. Top is original, bottom is aligned. This is using the alignment code from [the ROS node](https://github.com/intel-ros/realsense/blob/ad05c9b44f84d65ef6c03e4353ccc7ff722efa32/realsense_ros_camera/src/base_realsense_node.cpp#L380), which appears to be copied from the code in librealsense.
[image]

jpapon, 2018-02-12:
Top is a point cloud generated using un-aligned images, bottom is aligned.
Appears to me that the aligned version has ugly artifacts from the reprojection, and does nothing to improve the alignment between RGB and depth (the blue from my hoodie still appears on the wall behind me).

[image]

[image]

jpapon, 2018-02-28:
So the biggest issue for me is the alignment code - it's extremely CPU intensive. 
I'm going to try making an approximate version that's faster, since the full reprojection probably isn't necessary for many applications.

marvin-hao, 2018-03-02:
Hi @dorodnic,

As a reference, we keep OpenMP turned on, then only fetch the frames in YUY2 format and use OpenCV to convert the color space. The CPU usage during live streaming has dropped down.

The following images show the change. The CPU usage is generated as a time series of running 0, 1, 2 SR300 cameras.

[image]

[image]

jaroslavstehlik, 2018-03-20:
Hello,
I have still 100% CPU consumption on Core i7-6700K 4Ghz 4 cores 8ht
I just tested the default Unity project and it always takes 100% in editor also in build.
I tried even the Environment.SetEnvironmentVariable("OMP_NUM_THREADS", "1");
which did not help.

dorodnic, 2018-04-23:
@elektrikmoonkey and @jaroslavstehlik 
Just to make sure, you see this problem with either Unity or C#, not C++? For C++ I believe this should no longer be an issue, since we disabled OpenMP by default and invested in better optimised colour-conversion in recent releases.

jpapon, 2018-04-24:
Disabling OpenMP is not an option if depth --> rgb alignment is used (as in the align example).
There's no way to keep 30 fps without using multiple cores... I need about 250% CPU if I use alignment.

jpapon, 2018-07-19:
This was fixed in #2040 in commit  029b0d9a3f75b8d6d721b5cd6716dd927eb55f22 

Thanks for the hard work, it is much much faster now, and easily runs on a single core now.

I would remove all of the OpenMP stuff now as well, since performance is actually better without OpenMP - the threading seems to add some overhead (maybe due to cache misses?) which increases CPU usage.

closed 2018-07-19 by jpapon

thetravelingtechdude, 2022-04-05:
> Hi @cynosure4sure Compiling without OpenMP should in no way affect the functionality, we test it both on and off all the time. Could it be that during the no-OpenMP run the camera got identified as USB2? (the symptoms you describe are very much indicative of that)

Hi @dorodnic, I've built the Intel RealSense SDK using the Windows Installer in the past but I'm trying to build it from source (on Windows) using CMake to overcome the high CPU utilization issue. I've got it successfully configuring and generating on CMake and managed to get librealsense2.sln to build the solution successfully. What are the steps to link the realsense2.dll and realsense2.lib file to my Visual Studio project application?