realsenseai/librealsense #1106: Python API request: stream_profile::register_extrinsics_to and projection util functions

zhutq, 2018-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Ubuntu 16.04.3 LTS  | 
| Kernel Version (Linux Only)    |  4.4.0-112-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.9.1 | 

I'm working on calibrating my SR300's extrinsics from color to depth.
I can now successfully register my custom rotation matrix and translation vector in C++ with `stream_profile::register_extrinsics_to` and get a fairly good result.
But this API is not exposed to Python, and extrinsics in Python is not editable.
It's appreciated if it can be exposed to Python.
Also, `rs2_deproject_pixel_to_point`, `rs2_transform_point_to_point` and `rs2_project_point_to_pixel` can make calibrating and debugging in Python even easier.
Thank you very much!

zivsha, 2018-02-01:
Hi @zhutq,

I think we can safely add these functions.
It shouldn't be very difficult to add it by yourself for the meantime so that we don't block you. If you feel like it, take a look at `python.cpp`.

zivsha, 2018-02-04:
Hi @zhutq , we pushed  PR #1118 , adding these functions

zhutq, 2018-02-05:
Thank you @zivsha !
The only problem is that `extrinsics` cannot be created/modified from Python.
There's no constructor defined. And `rotation` and `translation` are readonly properties.

zivsha, 2018-02-05:
Oops :) I tried registering an existing one so missed that...
I'll change it too, thanks

closed 2018-02-08 by zivsha

Zeraphil, 2018-05-31:
Didn't find an answer on the Wiki, but since this is related didn't want to open a new issue. Are these APIs exposed in the C#/Unity wrapper? I can't find documentation on it. Thanks.