realsenseai/librealsense #1110: Exception: "Could not set Media Type. Device may be locked" when using Left IR + Right IR

xtrawurst, 2018-02-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400
| Firmware Version                   | 5.8.15 | 
| Operating System & Version |  Win 10 | 
| SDK Version                            |  2.8.3   | 

### Issue Description
I'm trying to access a real stream of the Right IR image.
According to [this thread](https://github.com/IntelRealSense/librealsense/issues/1094) if I want to get the real Right IR stream and not a computed stream I need to get both the Left and Right IR streams and then select the right IR stream when I need to.

However if I do that I get an exception when calling rs2::pipeline::start():
"Could not set Media Type. Device may be locked"

Interestingly the same happens when I try the same in the RS Viewer (selecting both IR streams). In that case I get a little dialog box with the same Exception message. I'm using version 2.8.3.

Is this a bug that has been fixed in one of the more recent versions of librealsense?

At the end of the day I simply want to access the Right IR stream as it comes from the camera, just as I can access the Left IR stream.

zivsha, 2018-02-01:
This looks like an error that you'd get when trying to stream profiles that do not work together.
Which profiles are you trying to stream?

xtrawurst, 2018-02-01:
IR Left 1280x720 RGB8
+ IR Right 1280x720 Y8

I just found out that it works if I use Y8 for both streams.

dorodnic, 2018-02-02:
Hi @xtrawurst,

The D415 offers the following stream combinations:
1. Depth
2. Left IR
3. Left RGB
4. Left IR + Right IR
5. Depth + Right IR
6. Depth + Left IR
7. Depth + Left RGB
8. Depth + Left IR + Right IR
9. Depth + Right IR

The hardware is incapable of producing both RGB and an IR stream from the stereo module.

xtrawurst, 2018-02-02:
@dorodnic 
Ok, thanks for the clarification.

closed 2018-02-08 by zivsha