realsenseai/librealsense #1112: RealSense Repository Key Unavailable, apt-get update fails

jallard-sr, 2018-02-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04.03  | 
| Kernel Version (Linux Only)    |   4.13.0-31-generic #34~16.04.1-Ubuntu SMP Fri Jan 19 17:11:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux   | 
| Platform                                 | Intel NUC  |
| SDK Version                            |  2.0  | 

### Issue Description
The repository key is not available right now, causing ```sudo apt-get update`` to fail with the following error.

```
Err:18 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial Release.gpg
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
```
The attached file has the full output.

Cheers,

Jim
[apt-get-failure.txt](https://github.com/IntelRealSense/librealsense/files/1686403/apt-get-failure.txt)

akashjinandra, 2018-02-01:
You can always download from source while waiting for Intel to fix this issue.

jallard-sr, 2018-02-01:
Yes, I'm rewriting our deployment scripts to work from a tarball snapshot I'll get from GitHub. This is not the preferred option, but it will have to do until the distribution repository proves itself to be stable.

dorodnic, 2018-02-02:
Hi @jallard-sr 
Unfortunately, the signing keys we are using are about to expire (both for the PPA and the Windows installer). We are working on the transition to the new ones, I believe it will help.

dorodnic, 2018-02-11:
We have updated the package with a new signature. In order to use it, please run:
```shell
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
```

jallard-sr, 2018-02-12:
Thank you, works for me now.

jallard-sr, 2018-02-12:
It seems ```doc/distribution_linux.md``` still references the old recv-key argument.

closed 2018-02-12 by jallard-sr

baptiste-mnh, 2018-04-28:
Hi,
It doesn't work with the key 6F3EFCDE

`W: The repository 'http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial Release' does not have a Release file.
`

What's the fix please?

DaleBaker, 2018-06-03:
This is really frustrating, intel still sells r200 cameras that rely on this for linux - but the repositories are so out of date

AnGPU, 2018-06-07:
At least they could justify that r200 model is "old" but there is no excuse with exactly the same problem with the zr300 (that would fix r200 problem too).

KastanDay, 2018-06-27:
Has anyone been able to update their d400-gen cameras (like the 415 or 435)? 
I am still unable to pull the proper keys.

I also tried to `ping keys.gnupg.net` and ran the same command using their IP address, but same 'network is unreachable' result. 

Any ideas? I'd also love a link to the source files I could build myself. 

``` Bash
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
Executing: /tmp/tmp.upcFMTPATd/gpg.1.sh --keyserver
keys.gnupg.net
--recv-key
6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
?: keys.gnupg.net: Network is unreachable
gpgkeys: HTTP fetch error 7: couldn't connect: Network is unreachable
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: keyserver unreachable
gpg: keyserver communications error: public key not found
gpg: keyserver receive failed: public key not found
```

mjscosta, 2018-07-16:
Hi, any new on this ? Is there a new key available already ? 
Thanks

bertrand-holveck, 2019-01-18:
Hi,

same issue here on January 2018.

`$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
Executing: /tmp/tmp.17LyQzIJSF/gpg.1.sh --keyserver
keys.gnupg.net
--recv-key
6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpgkeys: key 6F3EFCDE can't be retrieved
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: keyserver helper general error
gpg: keyserver communications error: unknown pubkey algorithm
gpg: keyserver receive failed: unknown pubkey algorithm
`

syh4661, 2019-01-24:
If anyone in intel realsense team please fix this key and package lost problem in linux 16.04 for d400 customer. there is no release file by following in that dfu guide pdf.

grafoteka, 2019-02-15:
Hi @bertrand-holveck did you get the key? I'm in the same error:

```
jorge@silth:~/ws/realsense_ws$ sudo apt-key adv --keyserver hkp://keys.gnupg.net:80 --recv-keys 6F3EFCDE
Executing: /tmp/tmp.fe7CH55mwJ/gpg.1.sh --keyserver hkp://keys.gnupg.net:80 --recv-keys 6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpgkeys: key 6F3EFCDE can't be retrieved
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: keyserver helper general error
gpg: keyserver communications error: unknown pubkey algorithm
gpg: keyserver receive failed: unknown pubkey algorithm
```
Kinds regards, 
Jorge

samxchandler, 2019-02-15:
@bertrand-holveck @grafoteka @dorodnic I'm having the same issue. Please let me know if there is a fix.

ARRevolution, 2019-03-30:
Even at home on standard ADSL router this didn't work.
I used my phones connection as recommended above and it worked.

grafoteka, 2019-04-02:
@samxchandler, you can do it copying the key to a file in your own computer, then install the key with the route to the file.
It work for me

YeeCY, 2019-07-26:
Try:

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys C8B3A55A6F3EFCDE`
`sudo apt-get update`

It works on my Ubuntu 18.04.

tonioteran, 2019-12-06:
@YeeCY's solution works for me on Ubuntu 18.04 as well.