realsenseai/librealsense #1113: Kernel patches on ARM (ODROID-XU4)

p0wdrdotcom, 2018-02-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.14.5-92                                        | 
| Platform                                 | ODROID-XU4  |
| SDK Version                            |  2 - master                          | 

### Issue Description
Hi, attempting to get a d435 running on an ARM device. I'm using an ODROID-XU4 as it has USB3 and 8 cores.  
I've built librealsense from source and installed it.  the realsense-viewer runs and I can access the depth stream.  The stream drops frames and I cannot get access to the IR streams like I can on x86.

I had issues running the v4l backend patches which is where I think the issue.  The script appears to run but exits silently.  I don't see the patches getting applied or the kernel modules getting loaded.

Output from dmesg when the device is plugged in.
```
[60577.341641] usb 4-1.2: new SuperSpeed USB device number 6 using xhci-hcd
[60577.362484] usb 4-1.2: New USB device found, idVendor=8086, idProduct=0b07
[60577.362491] usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[60577.362496] usb 4-1.2: Product: Intel(R) RealSense(TM) 430
[60577.362501] usb 4-1.2: Manufacturer: Intel(R) RealSense(TM) 430
[60577.362506] usb 4-1.2: SerialNumber: -------------
[60577.363833] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[60577.363845] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[60577.364061] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[60577.364915] uvcvideo: Unable to create debugfs 4-6 directory.
[60577.365044] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[60577.365052] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[60577.365060] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[60577.365067] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[60577.365314] input: Intel(R) RealSense(TM) 430: Int as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input7
[60577.366599] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0b07)
[60577.366952] uvcvideo: Unable to create debugfs 4-6 directory.
[60577.367074] uvcvideo 4-1.2:1.3: Entity type for entity Processing 7 was not initialized!
[60577.367080] uvcvideo 4-1.2:1.3: Entity type for entity Extension 8 was not initialized!
[60577.367086] uvcvideo 4-1.2:1.3: Entity type for entity Camera 6 was not initialized!

```
Warnings appearing in stdout when realsense-viewer is running

```
 01/02 19:38:50,020 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
 01/02 19:38:50,020 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:0/device/../../../../../../../../../../
 01/02 19:38:50,021 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:6/device/../../../../../../../../../../
 01/02 19:38:50,022 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:4/device/../../../../../../../../../../
 01/02 19:38:50,023 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:3/device/../../../../../../../../../../
 01/02 19:38:50,023 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:1/device/../../../../../../../../../../
 01/02 19:38:50,024 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:7/device/../../../../../../../../../../
 01/02 19:38:50,025 WARNING [2908111344] (backend-v4l2.cpp:459) Failed to read busnum/devnum. Device Path: /sys/dev/char/81:5/device/../../
```

ev-mp, 2018-02-04:
Hello @p0wdrdotcom ,
We publish kernel patches and DKMS for Ubuntu Linux LTS builds.

You may find useful info by looking into setting up librealsense on NVidia/Jetson (ARM-based) board with patches provided by community [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)

closed 2018-02-07 by ev-mp

mzahana, 2018-08-27:
@p0wdrdotcom have you managed to run the sdk on ODROID XU4?