realsenseai/librealsense #1115: Install Intel.RealSense.SDK error!

zhengtiesheng, 2018-02-02:
Error message:

Creating shortcuts..
C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Intel?RealSense?Viewer.lnk

**IPersistFile::Save failed;code 0x8007007B.
文件名、目录名或卷标语法不正确。**

halton, 2018-02-02:
@zhengtiesheng Where and what version of SDK you're installing? 

If you have https://software.intel.com/en-us/realsense-sdk-windows-eol, that is discontinued. The [SDK 2.0](https://software.intel.com/en-us/realsense/sdk) is what now under active developing.

zhengtiesheng, 2018-02-02:
@halton    I'm using sdk2.0.
This is the error of installing the sdk2.0.

dorodnic, 2018-02-02:
Hi @zhengtiesheng 
I assume there is a problem with localization of the trademark and the copyright signs. We will look into it.
In the meanwhile, you can try [building the SDK from source](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md) - it is relatively easy and offers everything the installer offers and some extra functionality. 
Please let us know if you encounter any problems

Loriskar, 2018-02-14:
I've encountered the same problem. Maybe for that reason, I can't even find the solution file.

RealSense-Customer-Engineering, 2018-02-26:
[Realsense Customer Engineering Team Comment] 
hi @Loriskar,@zhengtiesheng
I can reproduce this issue by changing windows display language to Chinese, so workaround for this is you can build the SDK from source as @halton mentioned or you can change windows display language to English, install the SDK and back to Chinese.

ericabd888, 2018-03-11:
Sorry I have the same problem too, even though I change my display language to English, it still have the same error message.
How to build the SDK from source in windows ? I have click the link but i don't know how to use powershell to build it. Thank you.

RealSense-Customer-Engineering, 2018-03-13:
[Realsense Customer Engineering Team Comment] 
hi @ericabd888,

if you need help for building SDK from source (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md), please open another ticket for this and tell us at which step you have question, so we can help.
thanks.

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
hi @zhengtiesheng, @ericabd888,

would like to know if building from source solve your issue? if so I would like to close this ticket for now.
thanks

zhengtiesheng, 2018-04-17:
@RealSense-Customer-Engineering      The new version of the SDK can be installed correctly.thank you.

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
thank you, let me close this ticket and feel free to create new ticket if any question.

closed 2018-04-17 by RealSense-Customer-Engineering