realsenseai/librealsense #1119: Location of Origin for D435 Camera

ndaley7, 2018-02-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (2.9.1) | 
| Operating System & Version |    {Ubuntu 14/16.04 LTS }| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2}                          | 

Where is the origin (x=0,y=0,z=0) located on a D435 Camera?  I was not able to find this information in any of the documentation.

zivsha, 2018-02-04:
Since you are asking for (x,y,**z**) I'm assuming you don't mean the camera's principal point, so which origin are you asking about?

ndaley7, 2018-02-04:
The location in 3D space where the point cloud data will have a coordinate value of (0,0,0). For example,  from MSDN regarding the 3D coordinate system for Kinect V 2.0 :+1: 
[image]

Any and all help is appreciated.

oordan, 2018-02-05:
The camera 3D coordinate system origin is located at the center of the left imager of the camera.

jallard-sr, 2018-02-05:
I presume you mean "left imager" when looking at the physcial camera from the back. :-)

How many millimeters deep into the camera from the glass surface is the origin located?

ndaley7, 2018-02-05:
Thank you, that was what I was asking.  This can be marked closed unless the 2nd question needs answering as well.

closed 2018-02-06 by ndaley7