realsenseai/librealsense #1126: Unix builds not using distro provided dependencies

srware, 2018-02-05:
See: [commit](https://github.com/IntelRealSense/librealsense/commit/d77cfb41354b72ee30aa2fe4148355f3462dee83#diff-47fb227e42855c6a2e12cdd32e7f126e)

In the above commit Unix builds have moved to using third party libraries (such as libusb) provided in the 'third-party' directory of the repository. This may be required for Android building but when compiling a shared library for Unix the preferred method is to compile against distro provided libraries using pkgconfig to find these (as was the case prior to the Android changes).

Forcing people to compile librealsense like this goes completely against Unix norms and also breaks the build for Yocto!!!

dorodnic, 2018-02-05:
Hi @srware 
Could you elaborate why did it break Yocto? The intent of this change, as you mentioned, was to enable AOS compilation and simplify compilation for users (making the repo more self-contained), so at least the intentions are good. 
If you think we are making a mistake, we will do our best to rectify it.

scooter1556, 2018-02-05:
I have fixed the initial issue I was having with Yocto which was due to libudev.h not being available in the build system.

I don't deny that having a self-contained build environment can be useful, however, for Linux this means you can only safely build a static library. To build a shared library (which is generally preferred) it has to be built against the libs installed by the distro's package manager or there is a high chance that the resulting librealsense library won't work when installed due to missing libs.

In my opinion there should be an option in the CMake script for building a shared library which ensures that pkgconfig is used to find all required dependencies (glfw, libusb etc...) and the library is built against these and fails if suitable versions are not found.

dorodnic, 2018-02-06:
We will keep the current solution under CMake flag and revert back to finding `libusb` pkg-config.

closed 2018-02-07 by srware