realsenseai/librealsense #1130: D435 High CPU Usage

cynosure4sure, 2018-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.08.15.00 | 
| Operating System & Version |  Linux (Ubuntu 14.04  | 
| Kernel Version (Linux Only)    | 4.4.0.111 | 
| Platform                                 | PC|
| SDK Version                            | LibRealSense v2.9.1  | 

### Issue Description
I am using D435 camera in Ubuntu 14.04 with ROS.
For only Color Camera and Depth Camera enabled I get a high CPU usage on my i7-6700HQ laptop. Please see the image below for only Color and Depth Camera running with pointcloud disabled.
It seems very unusual as compared to using ZR300 module.

[image]

fgzhu, 2018-02-07:
turn of OpenMP default will solve this problem

cynosure4sure, 2018-02-07:
Can you share your CPU usage after turning off OpenMP?

fgzhu, 2018-02-07:
less than 5% if just grab frame without showing

cynosure4sure, 2018-02-07:
@fgzhu Thank you for sharing your CPU usage.
After compiling with OpenMP=OFF and only grabbing color and depth frames without visualization, I am still seeing CPU usage of 50-60% on one thread as shown below. Are there any other options that you turned off and can you please let me know the specs of your system. Thank you.
[image]

zivsha, 2018-02-08:
@cynosure4sure do you use a pre-compiled version of librealsense? or are you building both the wrapper and the library?

cynosure4sure, 2018-02-09:
@zivsha I am building both librealsense and ros wrapper from source.
Now my CPU usage is below 10% on single thread.
I compiled librealsense with the following flags:
cmake ../ -DBUILD_EXAMPLES=OFF -DBUILD_GRAPHICAL_EXAMPLES=OFF -DBUILD_WITH_OPENMP=OFF -DBUILD_UNIT_TESTS=OFF -DCMAKE_BUILD_TYPE=Release -DENABLE_ZERO_COPY=ON -DFORCE_LIBUVC=OFF

and the ROS wrapper -DBUILD_WITH_OPENMP=OFF

I also have turned off the depth/color alignment.

dheera, 2018-02-17:
I can confirm the same issue. Can this be fixed in the pre-compiled version as well?
[image]

mfe7, 2018-03-08:
I also followed this suggestion to disable OpenMP, and I see the CPU usage is very low when just grabbing frames without visualizing (<5% of a core). When I also enable pointcloud computation, the CPU usage hardly changes. As soon as I start visualizing the pointcloud, the CPU usage rises to >100%. That is not surprising, because I assume visualization is intensive. 

The strange thing is that when I do `rostopic hz <pointcloud_topic>`, the CPU usage reaches >100% as well. So, it seems like the realsense node doesn't do much until something subscribes to one of its topics, and then uses a lot of CPU.

Is that expected behavior? Or is there something I am doing incorrectly?

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @mfe7,

is it works for you?

mfe7, 2019-04-18:
@RealSense-Customer-Engineering no, my issue is not resolved.

I just installed the latest firmware on the D435 (5.11.1), got the latest Debian `librealsense` pkgs, and compiled the latest `realsense_ros` version.

Here's my htop when running `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`, and nothing is subscribing to the topics (10% of a CPU):
[image]

When I do `rostopic echo` on the `/camera/color/image_raw` and `/camera/depth/color/points` topics (41% of a CPU):
[image]

When I view those two topics in RViz (113% of a CPU):
[image]

mfe7, 2019-04-18:
I just tested the same camera on a different computer (the above comment refers to a laptop Gigabyte Aero 15", this comment uses a desktop), with the same installation setup (debian librealsense, built latest realsense ros pkg).

The CPU usage is much more reasonable on the desktop, with the ROS nodelet manager capping out at <50%CPU when viewing the pointcloud in RViz. When I use a 424x240 depth and color stream, it is consistently <30%CPU.

It might be an issue specific to the laptop I have been using, but it is a reasonably powerful gaming laptop so that is surprising.