realsenseai/librealsense #1131: Frames not appearing and Camera unresponsive on realsense viewer and ros wrapper

akashjinandra, 2018-02-06:
-------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Linux (Ubuntu 16 
| Kernel Version (Linux Only)   4.13.0-32-generic                                     | 
| Platform                                 | PC
| SDK Version                            | 2.9.1 | 

### Issue Description
Hello all,

I've been having this issue ever since upgrading to Librealsense 2.9.1 
05/02 14:26:10,117 WARNING [139653277546240] (backend-v4l2.cpp:827) Frames didn't arrived within 5 seconds
 05/02 14:26:14,509 WARNING [139653634246400] (backend-v4l2.cpp:827) Frames didn't arrived within 5 seconds

I looked at #1086 but this seems to be an issue after an improper shutdown. My issue happens on first boot up. 

This makes my frames not appear and the camera becomes unresponsive and I have to unplug and replug in my camera to get it to work again. I am also using the ros wrapper on version 2.02. I tried rolling back to 2.8.1 where I didn't have the issue, but then I don't have the aligned depth to color stream that I need. 

Please advice on what I should do to fix this issue. I've tried on two different computers with two different realsense d415's.

akashjinandra, 2018-02-07:
Has there been any comments on this?

icarpis, 2018-02-08:
Hello @akashjinandra, this issue is a firmware bug that will be solved in the next FW releases.
If resetting the device is an option that can help you to workaround this issue, please try to work with [this branch](https://github.com/icarpis/realsense/tree/reset_dev).
In this branch I added a device `hardware_reset` operation each time the ROS node is loading.

RealSense-Customer-Engineering, 2018-04-05:
[Realsense Customer Engineering Team Comment] 
@akashjinandra please try upgrading to f/w 5.9.2.0.

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@akashjinandra if the latest f/w did not resolve you issue please open / reopen an issue.

closed 2018-04-18 by RealSense-Customer-Engineering

arya001, 2018-08-09:
@icarpis Can you tell me how to do this `hardware_reset` operation each time the ROS node is loading?