realsenseai/librealsense #1132: Linux support

dordure, 2018-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300  | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux (Centos - Red Hat ) | 
| Kernel Version (Linux Only)    |  (7.4)                                         | 
| Platform                                 |   |
| SDK Version                            |                         | 

### Issue Description

Currently, Intel RealSense SDK 2.0 provides installation packages for Debian OS and its derivatives.
But, will it be a future support for the installation of Centos?

Thanks!

dorodnic, 2018-02-06:
Hi @dordure 
At the moment we don't plan to expand our CI and validation infrastructures to additional Linux distributions. We are still working on supporting all the different flavors of Ubuntu 16 and adding Ubuntu 17.  
In practice, this should not prevent you from using RealSense on your system, and we will do our best to help. 

There are two main ways to enable RealSense on any Linux-based system:
1. Follow the steps in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) to build the SDK from source and apply necessary kernel patches. Check what is your kernel version (using `uname -r`) - for the SR300 you should be able to avoid kernel patches starting from kernel 4.12. 
1. Build the SDK from source with CMake flag `-DFORCE_LIBUVC=true`. This will bypass Video4Linux subsystem eliminating the need to patch the kernel. This comes at some expense of stability since everything that is usually done in the driver will now be performed in user-space.

dordure, 2018-02-08:
Thanks your feedback @dorodnic

dorodnic, 2018-02-12:
Feel free to open a new issue if you encounter any problem

closed 2018-02-12 by dorodnic