realsenseai/librealsense #1133: librealsense2 / rs-capture issues with different RealSense cameras

hinmanj, 2018-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 / SR300 / ZR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Windows 10 | 
| Platform                                 | PC  |
| SDK Version                            |  latest                        | 

### Issue Description
I just updated librealsense for the first time in a long time. Upon trying to use rs-capture with my R200 I got the following error:
> RealSense error calling rs2_pipeline_start(pipe:000001FF9FD461D0):
>     No device connected

I then learned R200 and ZR300 aren't supported anymore... but then here's the error for my ZR300 anyway:
> RealSense error calling rs2_pipeline_start(pipe:000001FD2DA80E40):
>     No device connected

I did notice that the SR300 is supposedly supported, so when I plugged that in I got this error:
> RealSense error calling rs2_pipeline_start(pipe:0000013E3120E0A0):
>     No device connected

Then, as I recalled whenever using my SR300 in the past, no cable ever seemed to work with the SR300 except the shortest tiny cable that came with the R200, so I tried using that, and it worked for a couple seconds until my computer blew up with a BUGCODE_USB3_DRIVER.

[image]

I'm afraid to try using the SR300 with the tiny cable again now for fear of my computer exploding again, so I guess my question is... for the R200/ZR300, is the only option now to use that last legacy version of librealsense that supports it? I updated because I have some R435's coming in today and I'd like to use those, but not being able to use R200's and R435's and SR300's with the same version of librealsense seems a little strange to me.

dorodnic, 2018-02-06:
Hi @hinmanj 

I am sorry you are getting such bad experience so far. 

The SR300 should definitely work, what you are describing is not a "known-issue". 
I don't think this can have anything to do with the version of librealsense. Perhaps the problem is specific to the Windows 10 build you are using? 

Regarding our legacy devices - 
Ideally, we would want to support all devices on all platforms/architectures/distributions and with all programming languages. Unfortunately, we can't do that at the moment. The SR300 is our best structured-light solution and the D400 is our best stereo-based solution. This is where we choose to concentrate our support and R&D focus right now.

jlgarcia75, 2018-02-15:
Hello @hinmanj,

Do you still need help with this issue?

Best regards,
Jesus G.
Intel Customer Support

closed 2018-02-16 by hinmanj

bf826989019, 2019-01-19:
I have this issue too.
When I run rs-capture, it shows 
##################################################
RealSense error calling rs2_pipeline_start(pipe:0xaf16c0):
    No device connected
###############################################
but, I have used sr300. Why ? Can you help me ? Thank you!