realsenseai/librealsense #1134: Fail to build with OpenMP flag set to false

yichenpan, 2018-02-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0                                         | 
| SDK Version                            |  v2.9.1                          | 

### Issue Description
I tried to build the v2.9.1 SDK with -DBUILD_WITH_OPENMP=false and it failed on sensor_control.cpp with error message
```
undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```
Then I tried to skip examples by set -DBUILD_EXAMPLES flag to false, but I still encountered the same error for live-test
```
/usr/bin/ld: CMakeFiles/live-test.dir/unit-tests-live.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```
Finally I checkout to v2.9.0 and it works with -DBUILD_EXAMPLES set to false, but still have the same error message for sensor control... I was wondering if anyone has the same problem here?

RealSense-Customer-Engineering, 2018-02-07:
[Realsense Customer Engineering Team Comment] 
Hi,
Please modify the CMakeLists.txt,  and include  -pthread  to CXX compiler  flag.
Thank you.

yichenpan, 2018-02-07:
@RealSense-Customer-Engineering  Thanks. It works now

dorodnic, 2018-02-25:
This was fixed in #1209

closed 2018-02-25 by dorodnic