realsenseai/librealsense #1135: Installation issue: Hunk #1 FAILED at 119.

waspinator, 2018-02-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  ZR300  | 
| Firmware Version                   | 2.0.71.28 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0-32-generic                                      | 
| Platform                                 | PC (Intel NUC)  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
I'm trying to install using this guide

    https://githacub.com/IntelRealSense/librealsense/blob/master/doc/installation.md

but I'm getting a `Hunk #1 FAILED at 119.` error

    sudo apt-get --reinstall install 'ros-*-librealsense'
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Note, selecting 'ros-kinetic-librealsense' for glob 'ros-*-librealsense'
    0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 34 not upgraded.
    Need to get 0 B/325 kB of archives.
    After this operation, 0 B of additional disk space will be used.
    (Reading database ... 358429 files and directories currently installed.)
    Preparing to unpack .../ros-kinetic-librealsense_1.12.1-0xenial-20171104-172651-0800_amd64.deb ...
    Unpacking ros-kinetic-librealsense (1.12.1-0xenial-20171104-172651-0800) over (1.12.1-0xenial-20171104-172651-0800) ...
    Processing triggers for libc-bin (2.23-0ubuntu10) ...
    Setting up ros-kinetic-librealsense (1.12.1-0xenial-20171104-172651-0800) ...

    Creating symlink /var/lib/dkms/uvcvideo/1.1.1-3-realsense/source ->
                     /usr/src/uvcvideo-1.1.1-3-realsense

    DKMS: add completed.

    Kernel preparation unnecessary for this kernel.  Skipping...

    Running the pre_build script:

    dkms: Pre-build script started...

    Downloading kernel sources...
    Reading package lists...
    Picking 'linux-hwe' as source package instead of 'linux-image-4.13.0-32-generic'
    NOTICE: 'linux-hwe' packaging is maintained in the 'Git' version control system at:
    git://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial -b hwe
    Please use:
    git clone git://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial -b hwe
    to retrieve the latest (possibly unreleased) updates to the package.
    Need to get 164 MB of source archives.
    Get:1 http://us.archive.ubuntu.com/ubuntu xenial-updates/main linux-hwe 4.13.0-32.35~16.04.1 (dsc) [5,886 B]
    Get:2 http://us.archive.ubuntu.com/ubuntu xenial-updates/main linux-hwe 4.13.0-32.35~16.04.1 (tar) [155 MB]
    Get:3 http://us.archive.ubuntu.com/ubuntu xenial-updates/main linux-hwe 4.13.0-32.35~16.04.1 (diff) [8,712 kB]
    gpgv: Signature made Thu 25 Jan 2018 04:56:44 AM EST using RSA key ID CBEECEA3
    gpgv: Can't check signature: public key not found
    dpkg-source: warning: failed to verify signature on ./linux-hwe_4.13.0-32.35~16.04.1.dsc
    dpkg-source: info: extracting linux-hwe in linux-hwe-4.13.0
    dpkg-source: info: unpacking linux-hwe_4.13.0.orig.tar.gz
    dpkg-source: info: applying linux-hwe_4.13.0-32.35~16.04.1.diff.gz
    dpkg-source: info: upstream files that have been modified: 
    W: Can't drop privileges for downloading as file 'linux-hwe_4.13.0-32.35~16.04.1.dsc' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
    Fetched 164 MB in 55s (2,980 kB/s)
    INFO: No Intel RealSense(TM) cameras are currently supported.
    Patching uvcvideo sources...
    patching file drivers/media/usb/uvc/Makefile
    patching file drivers/media/usb/uvc/uvc_driver.c
    Hunk #1 succeeded at 203 with fuzz 2 (offset 55 lines).
    patching file drivers/media/usb/uvc/uvcvideo.h
    Hunk #1 FAILED at 119.
    1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej
    patching file include/uapi/linux/videodev2.h
    Hunk #1 FAILED at 621.
    1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej

    Building module:
    cleaning build area....
    make KERNELRELEASE=4.13.0-32-generic -C /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/linux-src M=drivers/media/usb/uvc/....(bad exit status: 2)
    ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/ros-kinetic-librealsense.0.crash'
    Error! Bad return status for module build on kernel: 4.13.0-32-generic (x86_64)
    Consult /var/lib/dkms/uvcvideo/1.1.1-3-realsense/build/make.log for more information.
    WARNING: DKMS module failed to installed; removing...

    ------------------------------
    Deleting module version: 1.1.1-3-realsense
    completely from the DKMS tree.
    ------------------------------
    Done.
    WARNING: Unknown configuration for Intel RealSense(TM) cameras!
           To resolve, please follow the installation directions at:
               http://wiki.ros.org/librealsense#Installation
    Processing triggers for libc-bin (2.23-0ubuntu10) ...

ev-mp, 2018-02-07:
Hello @waspinator ,
The dkms driver for ZR300 and Librealsense v1.12 is targeted for kernel version 4.4.

Switching to  kernel 4.4 would the fastest way to make this work.
Alternatively, if you can switch to kernel 4.10 then you can use librealsense dkms driver for SDK v2.0 instead which is backward-compatible with ZR300 in terms of kernel patches.

waspinator, 2018-02-08:
thanks @ev-mp. downgrading to 4.4.0-112-generic worked.

closed 2018-02-08 by waspinator