realsenseai/librealsense #1137: 4.14.15 - no formats full support

rcny, 2018-02-07:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   CentOS 7.4  | 
| Kernel Version (Linux Only)    |  4.14.15                                         | 
| Platform                                 | PC |
| SDK Version                            |  none | 

### Issue Description

Hello!

I'm using heavily modified CentOS 7.4 with vanilla 4.14.15 kernel. It is somewhat patched, basically for additional specific hardware support. I've thought that everything needed for SR300 support was already upstreamed, but it seems not. I've looked at the patches available here and manually performed necessary changes to provide video backend and driver with tweaks and missing video formats. 

This is what I've done:

- Added missing additional formats to `uvc_fmts[]` into `uvc_driver.c`:

```
	{
		.name		= "FlickerIR 8-bit (RELI)",
		.guid		= UVC_GUID_FORMAT_RELI,
		.fcc		= V4L2_PIX_FMT_RELI,
	},
	{
		.name		= "Depth:IR 16:8 24-bit (INRI)",
		.guid		= UVC_GUID_FORMAT_INRI,
		.fcc		= V4L2_PIX_FMT_INRI,
	},
	{
		.name		= "Depth 16-bit (INVR)",
		.guid		= UVC_GUID_FORMAT_INVR,
		.fcc		= V4L2_PIX_FMT_INVR,
	},
	{
		.name		= "Raw data 8-bit (RAW8)",
		.guid		= UVC_GUID_FORMAT_RAW8,
		.fcc		= V4L2_PIX_FMT_GREY,
	},
	{
		.name		= "Raw data 16-bit (RW16)",
		.guid		= UVC_GUID_FORMAT_RW16,
		.fcc		= V4L2_PIX_FMT_RW16,
	},
	{
		.name		= "Luminosity data 8-bit (L8)",
		.guid		= UVC_GUID_FORMAT_L8,
		.fcc		= V4L2_PIX_FMT_GREY,
	},
	{
		.name		= "Luminosity data 16-bit (L16)",
		.guid		= UVC_GUID_FORMAT_L16,
		.fcc		= V4L2_PIX_FMT_Y16,
	},
	{
		.name		= "Depth data 16-bit (D16)",
		.guid		= UVC_GUID_FORMAT_D16,
		.fcc		= V4L2_PIX_FMT_Z16,
	},
	{
		.name		= "Packed raw data 10-bit",
		.guid		= UVC_GUID_FORMAT_W10,
		.fcc		= V4L2_PIX_FMT_W10,
	},
```

- Added missing additional formats into `uvcvideo.h` header file:

```
#define UVC_GUID_FORMAT_RELI \
	{ 'R',  'E',  'L',  'I', 0x14, 0x13, 0x43, 0xf9, \
	 0xa7, 0x5a, 0xee, 0x6b, 0xbf, 0x01, 0x2e, 0x23}
#define UVC_GUID_FORMAT_INRI \
	{ 'I',  'N',  'R',  'I', 0x90, 0x2d, 0x58, 0x4a, \
	 0x92, 0x0b, 0x77, 0x3f, 0x1f, 0x2c, 0x55, 0x6b}
#define UVC_GUID_FORMAT_INVR \
	{ 'I',  'N',  'V',  'R', 0x90, 0x2d, 0x58, 0x4a, \
	 0x92, 0x0b, 0x77, 0x3f, 0x1f, 0x2c, 0x55, 0x6b}
#define UVC_GUID_FORMAT_RAW8 \
    { 'R',  'A',  'W',  '8', 0x66, 0x1a, 0x42, 0xa2, \
     0x90, 0x65, 0xd0, 0x18, 0x14, 0xa8, 0xef, 0x8a}
#define UVC_GUID_FORMAT_RW16 \
    { 'R',  'W',  '1',  '6', 0x00, 0x00, 0x10, 0x00, \
     0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}
#define UVC_GUID_FORMAT_L8 \
	{ '2', 0x00,  0x00,  0x00, 0x00, 0x00, 0x10, 0x00, \
	 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}
#define UVC_GUID_FORMAT_L16 \
	{ 'Q', 0x00,  0x00,  0x00, 0x00, 0x00, 0x10, 0x00, \
	 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}
#define UVC_GUID_FORMAT_D16 \
	{ 'P', 0x00,  0x00,  0x00, 0x00, 0x00, 0x10, 0x00, \
	 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}
#define UVC_GUID_FORMAT_W10 \
    { 'W',  '1',  '0',  ' ', 0x00, 0x00, 0x10, 0x00, \
	 0x80, 0x00, 0x00, 0xaa, 0x00, 0x38, 0x9b, 0x71}
```

- Added missing additional formats into `videodev2.h`:

```
#define V4L2_PIX_FMT_RELI     v4l2_fourcc('R', 'E', 'L', 'I') /* 8 IR alternating on off illumination */
#define V4L2_PIX_FMT_INRI     v4l2_fourcc('I', 'N', 'R', 'I') /* 24 Depth/IR 16:8 */
#define V4L2_PIX_FMT_INVR     v4l2_fourcc('I', 'N', 'V', 'R') /* 16 Depth */
#define V4L2_PIX_FMT_RW16 v4l2_fourcc('R', 'W', '1', '6') /* Raw data 16-bit */
#define V4L2_PIX_FMT_W10      v4l2_fourcc('W', '1', '0', ' ') /* Packed raw data 10-bit */
```

- Added missing additional formats into `v4l2-ioctl.c`:

```
	case V4L2_PIX_FMT_RW16:		descr = "16-bit Raw data"; break;
	case V4L2_PIX_FMT_W10:		descr = "10-bit packed 8888[2222]"; break;
```

- Applied `realsense-metadata-ubuntu-xenial.patch` to USB Video Class driver

- Applied `realsense-powerlinefrequency-control-fix.patch`

Looks like that wasn't enough because `v4l2` still reports with unrecognized formats:

```
[root@XXXXXXX ~]# v4l2-ctl --list-formats-ext -d /dev/video1
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'Z16 '
	Name        : 16-bit Depth
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'INZI'
	Name        : Planar 10:16 Greyscale Depth
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 3
	Type        : Video Capture
	Pixel Format: ''
	Name        : 435a4e49-b602-480f-978c-e4e88ae
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 4
	Type        : Video Capture
	Pixel Format: 'Y10 '
	Name        : 10-bit Greyscale
		Size: Discrete 640x480
			Interval: Discrete 0.005s (200.000 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 5
	Type        : Video Capture
	Pixel Format: ''
	Name        : 52494150-8536-4841-b6bf-8fc6ffb
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
```

What are `435a4e49-b602-480f-978c-e4e88ae` and `52494150-8536-4841-b6bf-8fc6ffb`? I understand that my setup may be unsupported, but it looks like patches basically add support for Intel-specific video formats. I've added them all and kernel still isn't able to recognize everything that camera supports. What should I do? 

Thanks in advance.

rcny, 2018-02-07:
Something more.

`uvcvideo` also reports that it wasn't able to recognize these two formats:

```
[    8.117123] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[    8.117125] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
```

As I understand, these two are Intel's `INZC` and `PAIR`. I haven't seen anywhere that there are patches supporting them. Is it possible to fully run SR300 on Linux systems?

ev-mp, 2018-02-07:
Hello @rcny ,
The formats that SR300 supports is Depth (INVZ), Infrared (INVI), Depth+Infrared (INZI) and Color (YUYV).
These formats are fully upstreamed since kernel 4.12.

There are additional proprietary formats, such as 'RELI', 'PAIR' 'INZC', that are used in production line for calibration and other activities. These are not upstreamed for obvious reasons. We also do not use it in Librealsense.
Additionally, some of the formats you're trying to add are for D400 series (UVC_GUID_FORMAT_D16 , UVC_GUID_FORMAT_L8) and are not applicable for SR300.

With kernel 4.14 there should be no need for additional patches to stream SR300.
Let us know if you cannot receive Depth/IR streams on CentOS.

rcny, 2018-02-07:
@ev-mp, thanks for the clarification!
I am able to get all needed streams, but the kernel log is flooded with dumps from `v4l2`. Basically, I've thought that the problem persists because of the missing formats and that was the thing that initially led me to the patches published here. How should I deal with these warnings generated by `v4l2` due to unknown formats? 

```
*   Unknown pixelformat 0x00000000
*   ------------[ cut here ]------------
*   WARNING: CPU: 3 PID: 1283 at drivers/media/v4l2-core/v4l2-ioctl.c:1294 v4l_enum_fmt+0xd7b/0x1450 [videodev]
*   Modules linked in: intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp vfat fat kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc iTCO_wdt iTCO_vendor_support arc4 snd_hda_codec_realtek iwldvm snd_hda_codec_hdmi snd_hda_codec_generic aesni_intel mac80211 crypto_simd glue_helper cryptd snd_soc_ssm4567 snd_soc_rt5640 snd_hda_intel snd_soc_rl6231 uvcvideo snd_soc_core snd_usb_audio videobuf2_vmalloc snd_hda_codec videobuf2_memops snd_compress iwlwifi snd_usbmidi_lib videobuf2_v4l2 snd_rawmidi snd_hda_core sg pcspkr snd_hwdep videobuf2_core snd_seq i2c_i801 snd_seq_device videodev lpc_ich snd_pcm joydev cfg80211 cdc_acm mei_me rfkill snd_timer mei shpchp snd soundcore regmap_i2c snd_soc_sst_acpi i2c_designware_platform snd_soc_sst_match dw_dmac i2c_designware_core
*    acpi_pad ip_tables ext4 mbcache jbd2 hid_multitouch sd_mod i915 drm_kms_helper syscopyarea ahci sysfillrect libahci sysimgblt fb_sys_fops igb crc32c_intel libata drm ptp pps_core dca i2c_algo_bit sdhci_acpi sdhci mmc_core video i2c_hid i2c_core
*   CPU: 3 PID: 1283 Comm: AudioThread Not tainted 4.14.15-sr300 #1
*   Hardware name: To be filled by O.E.M.
*   task: ffff909751a15d00 task.stack: ffffa9bb00e84000
*   RIP: 0010:v4l_enum_fmt+0xd7b/0x1450 [videodev]
*   RSP: 0018:ffffa9bb00e87c90 EFLAGS: 00010246
*   RAX: 000000000000001e RBX: ffffa9bb00e87d88 RCX: 0000000000000000
*   RDX: 0000000000000000 RSI: ffff909800b96978 RDI: ffff909800b96978
*   RBP: 0000000000000000 R08: 0000000000000000 R09: 00000000000003cd
*   R10: 0000000000000003 R11: 0000000000000000 R12: ffffffffc0bc8900
*   R13: ffff90973f3a5200 R14: 00000000c0405602 R15: ffffffffc0a571e0
*   FS:  00007fa09affd700(0000) GS:ffff909800b80000(0000) knlGS:0000000000000000
*   CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
*   CR2: 00007fa0ee3a8000 CR3: 0000000077b32005 CR4: 00000000001606e0
*   Call Trace:
*    __video_do_ioctl+0x275/0x2f0 [videodev]
*    ? path_openat+0x604/0x12c0
*    ? video_ioctl2+0x20/0x20 [videodev]
*    video_usercopy+0x23a/0x560 [videodev]
*    ? do_filp_open+0xa5/0x100
*    v4l2_ioctl+0x95/0xb0 [videodev]
*    do_vfs_ioctl+0xa6/0x5f0
*    SyS_ioctl+0x74/0x80
*    do_syscall_64+0x61/0x1a0
*    entry_SYSCALL64_slow_path+0x25/0x25
*   RIP: 0033:0x7fa0e5ef0107
*   RSP: 002b:00007fa09affc2a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
*   RAX: ffffffffffffffda RBX: 00007fa09affc5f0 RCX: 00007fa0e5ef0107
*   RDX: 00007fa09affc360 RSI: 00000000c0405602 RDI: 0000000000000044
*   RBP: 00000000c02c564a R08: 00007fa078002060 R09: 00007fa078001e10
*   R10: 0000028000000000 R11: 0000000000000246 R12: 0000000000000044
*   R13: 00007fa09affc3b0 R14: 00007fa09affc3b0 R15: 00000000c0405602
*   Code: 32 36 34 0f 84 71 06 00 00 3d 4d 32 36 34 48 c7 c6 a7 9a a4 c0 0f 84 6d f4 ff ff 89 c6 48 c7 c7 b9 a5 a4 c0 31 c0 e8 42 f0 0a f0 <0f> ff 80 7b 0c 00 0f 85 fa f2 ff ff 8b 43 2c 48 c7 c1 05 99 a4 
*   ---[ end trace e17fdddbd3764b09 ]---
*   Unknown pixelformat 0x00000000
*   ------------[ cut here ]------------
*   WARNING: CPU: 3 PID: 1283 at drivers/media/v4l2-core/v4l2-ioctl.c:1294 v4l_enum_fmt+0xd7b/0x1450 [videodev]
*   Modules linked in: intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp vfat fat kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc iTCO_wdt iTCO_vendor_support arc4 snd_hda_codec_realtek iwldvm snd_hda_codec_hdmi snd_hda_codec_generic aesni_intel mac80211 crypto_simd glue_helper cryptd snd_soc_ssm4567 snd_soc_rt5640 snd_hda_intel snd_soc_rl6231 uvcvideo snd_soc_core snd_usb_audio videobuf2_vmalloc snd_hda_codec videobuf2_memops snd_compress iwlwifi snd_usbmidi_lib videobuf2_v4l2 snd_rawmidi snd_hda_core sg pcspkr snd_hwdep videobuf2_core snd_seq i2c_i801 snd_seq_device videodev lpc_ich snd_pcm joydev cfg80211 cdc_acm mei_me rfkill snd_timer mei shpchp snd soundcore regmap_i2c snd_soc_sst_acpi i2c_designware_platform snd_soc_sst_match dw_dmac i2c_designware_core
*    acpi_pad ip_tables ext4 mbcache jbd2 hid_multitouch sd_mod i915 drm_kms_helper syscopyarea ahci sysfillrect libahci sysimgblt fb_sys_fops igb crc32c_intel libata drm ptp pps_core dca i2c_algo_bit sdhci_acpi sdhci mmc_core video i2c_hid i2c_core
*   CPU: 3 PID: 1283 Comm: AudioThread Tainted: G        W       4.14.15-sr300 #1
*   Hardware name: To be filled by O.E.M.
*   task: ffff909751a15d00 task.stack: ffffa9bb00e84000
*   RIP: 0010:v4l_enum_fmt+0xd7b/0x1450 [videodev]
*   RSP: 0018:ffffa9bb00e87c90 EFLAGS: 00010246
*   RAX: 000000000000001e RBX: ffffa9bb00e87d88 RCX: 0000000000000006
*   RDX: 0000000000000000 RSI: 0000000000000092 RDI: ffff909800b96970
*   RBP: 0000000000000000 R08: 0000000000000000 R09: 00000000000003f3
*   R10: 0000000000000003 R11: 0000000000000000 R12: ffffffffc0bc8900
*   R13: ffff90973f3a5200 R14: 00000000c0405602 R15: ffffffffc0a571e0
*   FS:  00007fa09affd700(0000) GS:ffff909800b80000(0000) knlGS:0000000000000000
[... cut ...]
*   R10: 0000028000000000 R11: 0000000000000246 R12: 0000000000000044
*   R13: 00007fa09affc3b0 R14: 00007fa09affc3b0 R15: 00000000c0405602
*   Code: 32 36 34 0f 84 71 06 00 00 3d 4d 32 36 34 48 c7 c6 a7 9a a4 c0 0f 84 6d f4 ff ff 89 c6 48 c7 c7 b9 a5 a4 c0 31 c0 e8 42 f0 0a f0 <0f> ff 80 7b 0c 00 0f 85 fa f2 ff ff 8b 43 2c 48 c7 c1 05 99 a4 
*   ---[ end trace e17fdddbd3764b0a ]---
```
There are two warnings generated by `v4l2` and these warnings come each time as `librealsense` accesses camera.

`librealsense` is at `2.9.1`.

rcny, 2018-02-07:
I have a suspicion that `pixeformat`'s absence is directly linked to missing declarations of `PAIR` and `INZC`, but as you said before, these are purely utility formats. I have no clue what to do next, except going for the dirty hacks, such as suppressing `v4l2` warnings if `pixelformat` is empty.

ev-mp, 2018-02-07:
These two formats cause the kernel warning you've attached. 
This, however, can be changed only by modifying SR300 firmware.

In case you absolutely cannot allow for the kernel messages to appear then you can apply the patch for these two formats - it will suppress the warnings. 

Note that these warnings are generated only once per enumeration, so unless you have to replug the device frequently it should not clutter dmesg console.

rcny, 2018-02-07:
My `librealsense` application accesses camera on-demand and I see these warnings as application starts each time. `dmesg` dumps its contents to the log aggregation infrastructure, so to keep these warnings flow isn't the viable option for me.

What patch are you talking about? Sorry, I haven't found the patch with support for these formats anywhere in the public.

ev-mp, 2018-02-07:
While there are no official patches for `INZC` and `RELI` (see above), you may find references in community work, such as https://github.com/teknotus/depthview/blob/kernelpatchfmt/0001-Add-video-formats-for-Intel-real-sense-F200-camera.patch

rcny, 2018-02-07:
Sorry, but I don't understand what kind of references I should find there. I've looked through that patch via provided link and haven't found anything concerning support of `INZC`/`PAIR`. Once again, I understand that these proprietary formats don't provide any necessary functionality for end user or developer, but why then they aren't hidden for enumeration? And it looks like they are supported in Windows.

Simply put, I totally acknowledge what you are talking to me, but still I don't see any answer to this except suppressing kernel warnings by myself. 

Honestly, I don't need any out-of-box kernel warnings even if device is functional. And suppressing output of kernel module looks like a really dirty solution for the device with official Linux support. Is there anything you can do with this?

ev-mp, 2018-02-07:
Hi @rcny,
The above link is an example of registering proprietary formats with uvc kernel driver - it can be used as a reference to generate kernel patch that will register the two additional formats with v4l and remove the dmesg mesages. This way you don't have to modify the kernel flows by suppressing messages.

As SR300 [is being gradually retired](http://qdms.intel.com/dm/i.aspx/604E4A15-B7DD-4833-9264-5C6951CF0D54/PCN115898-00.pdf)  requesting FW modificatons would not be feasible.
Having a kernel patch, on the other hand, would be an effective and efficient measure to address this issue.

rcny, 2018-02-08:
Hello!

I've added `INZC` and `PAIR` support:

- `uvc_driver.c`:

```
	{
		.name		= "IR with Confidence (INZC)",
		.guid		= UVC_GUID_FORMAT_INZC,
		.fcc		= V4L2_PIX_FMT_INZC,
	},
	{
		.name		= "Custom IR (PAIR)",
		.guid		= UVC_GUID_FORMAT_PAIR,
		.fcc		= V4L2_PIX_FMT_PAIR,
	},
```

- `uvcvideo.h`:

```
#define UVC_GUID_FORMAT_INZC \
    { 'I',  'N',  'Z',  'C', 0x02, 0xb6, 0x0f, 0x48, \
     0x97, 0x8c, 0xe4, 0xe8, 0x8a, 0xe8, 0x9b, 0x89}
#define UVC_GUID_FORMAT_PAIR \
    { 'P',  'A',  'I',  'R', 0x36, 0x85, 0x41, 0x48, \
     0xb6, 0xbf, 0x8f, 0xc6, 0xff, 0xb0, 0x83, 0xa8}
```

- `videodev2.h`:

```
#define V4L2_PIX_FMT_INZC     v4l2_fourcc('I', 'N', 'Z', 'C') /* Both the IR image and a confidence image */
#define V4L2_PIX_FMT_PAIR     v4l2_fourcc('P', 'A', 'I', 'R') /* Appears to switch between the IR Image and another unknown image */

```

- `v4l2-ioctl.c`:

```
	case V4L2_PIX_FMT_INZC:		descr = "IR with Confidence"; break;
	case V4L2_PIX_FMT_PAIR:		descr = "Custom IR"; break;
```

Now everything is detected correctly:

```
[root@XXXXXXX ~]# v4l2-ctl --list-formats-ext -d /dev/video1
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'Z16 '
	Name        : 16-bit Depth
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'INZI'
	Name        : Planar 10:16 Greyscale Depth
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 3
	Type        : Video Capture
	Pixel Format: 'INZC'
	Name        : IR with Confidence
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 4
	Type        : Video Capture
	Pixel Format: 'Y10 '
	Name        : 10-bit Greyscale
		Size: Discrete 640x480
			Interval: Discrete 0.005s (200.000 fps)
			Interval: Discrete 0.008s (120.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 5
	Type        : Video Capture
	Pixel Format: 'PAIR'
	Name        : Custom IR
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)

```

However, there are some errors left. `uvcvideo` continues to report with some errors:

```
[   58.233754] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 2: -32 (exp. 1).
[   58.239498] uvcvideo: Non-zero status (-71) in video completion handler.
```

These errors persist even on vanilla 4.14 kernel cloned from [kernel.org](https://kernel.org). Is there any clarification for these errors?

RealSense-Customer-Engineering, 2018-03-13:
[Realsense Customer Engineering Team Comment] 
Hi @rcny

To run the realsense-viewer or rs-capture app in librealsense 2.10.1, not see below error in Ubuntu 16.04.3 + Kernel 4.14.15 with the patches with INZC and PAIR format applied. Is the issue "UVC control" still in your setup?

    [58.233754] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 2: -32 (exp. 1).

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
Hi @rcny,

Still see any issue for further check?

closed 2018-06-22 by RealSense-Customer-Engineering