realsenseai/librealsense #1140: How to get left and right IR images from D435

anson2004110, 2018-02-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0 |                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
we can use intel realsense SDK 2.0 sample to capture Depth and RGB images, but we cannot find the API to get the raw left and right IR images from D435.
So any suggestion to solve it？

zivsha, 2018-02-07:
Hi @anson2004110 ,
Please have a look at the [API-How-To : Start Streaming Left and Right Infrared Imagers](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers)

anson2004110, 2018-02-11:
Thanks for your reply, we found the API as followed, but we don't know how to capture left and right IR images at the same time. @zivsha

    //Contruct a pipeline which abstracts the device
	rs2::pipeline pipe;

	//Create a configuration for configuring the pipeline with a non default profile
	rs2::config cfg;

	//Add desired streams to configuration
	cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);

	//Instruct pipeline to start streaming with the requested configuration
	pipe.start(cfg);

	// Camera warmup - dropping several first frames to let auto-exposure stabilize
	rs2::frameset frames;
	for (int i = 0; i < 30; i++)
	{
		//Wait for all configured streams to produce a frame
		frames = pipe.wait_for_frames();
	}

	//Get each frame
	rs2::frame ir_frame = frames.first(RS2_STREAM_INFRARED);
	rs2::frame depth_frame = frames.get_depth_frame();

	// Creating OpenCV matrix from IR image
	Mat ir(Size(640, 480), CV_8UC1, (void*)ir_frame.get_data(), Mat::AUTO_STEP);

	// Apply Histogram Equalization
	equalizeHist(ir, ir);
	applyColorMap(ir, ir, COLORMAP_JET);

	// Display the image in GUI
	namedWindow("Display Image", WINDOW_AUTOSIZE);
	imshow("Display Image", ir);
	waitKey(0);

zivsha, 2018-02-11:
Hi,
In cases where there are multiple streams of the same type in one sensor, like the case where the depth sensor has IR left and IR Right, you need to specify the indices numbers of the streams you want, so to get Depth and 2 Infrareds you should:
```cpp
c.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
c.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
c.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
```

zivsha, 2018-02-11:
To get the frames, you can use `frameset::first<>` anymore. You can either iterate over the frameset, or use the new `get_infrared(index)` method

anson2004110, 2018-02-11:
@zivsha
thanks，do you know how to convert rs2::video_frame ir_frame to Opencv Mat？

dorodnic, 2018-02-11:
Please see [opencv RealSense samples](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv)

anson2004110, 2018-02-12:
Thank you all, then the useful code was shown as followed, note that we also add function get_infrared_frame into the rs_frame.hpp (see https://github.com/IntelRealSense/librealsense/pull/1122/commits/c391aa3bdbfe3609e3854652eb4b14e134ab4721)

int main(int argc, char * argv[])
{
	int width = 1280;
	int height = 720;
	int fps = 30;
	rs2::config config;
	config.enable_stream(RS2_STREAM_INFRARED, 1, width, height, RS2_FORMAT_Y8, fps);
	config.enable_stream(RS2_STREAM_INFRARED, 2, width, height, RS2_FORMAT_Y8, fps);

	// start pipeline
	rs2::pipeline pipeline;
	rs2::pipeline_profile pipeline_profile = pipeline.start(config);

	while (1) // Application still alive?
	{
		// wait for frames and get frameset
		rs2::frameset frameset = pipeline.wait_for_frames();

		// get single infrared frame from frameset
		//rs2::video_frame ir_frame = frameset.get_infrared_frame();

		// get left and right infrared frames from frameset
		rs2::video_frame ir_frame_left = frameset.get_infrared_frame(1);
		rs2::video_frame ir_frame_right = frameset.get_infrared_frame(2);

		cv::Mat dMat_left = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)ir_frame_left.get_data());
		cv::Mat dMat_right = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)ir_frame_right.get_data());

		cv::imshow("img_l", dMat_left);
		cv::imshow("img_r", dMat_right);
		char c = cv::waitKey(1);
		if (c == 's')
		{

		}
		else if (c == 'q')
			break;
	}

	return EXIT_SUCCESS;
}

closed 2018-02-12 by anson2004110

alexcherpi, 2019-03-11:
Hi, 
I know this post is closed but I still have a question about the IR image of the D435 or D435i.
Is it possible to get the IR image of the left image without the dot patern? 
Do you think we could use this IR image to observe thermal leakage of a window for example (compared to an IR Flir camera)?

agrunnet, 2019-03-11:
The only way to make the IR pattern disappear for the d435 is to turn off the projector. 

In your second question, no this won’t detect heat. That is a different wavelength range, generally closer to 10000nm.  The IR we use is 850nm and 940nm but which is invisible to humans but won’t detect heat. It is also commonly called “near IR”.

necroen, 2019-03-13:
You could use some plastic sheet stick on the IR projector.

npd26, 2019-07-04:
how to get left and right image using cv_bridge in ROS?

chutsu, 2019-07-11:
> Hi,
> In cases where there are multiple streams of the same type in one sensor, like the case where the depth sensor has IR left and IR Right, you need to specify the indices numbers of the streams you want, so to get Depth and 2 Infrareds you should:
> 
> ```c++
> cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
> cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
> cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
> ```

Why are the infrared imagers index-ed from 1? Also the [Wiki](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers) is wrong.