realsenseai/librealsense #1145: Librealsense + OpenCV build, VTK issue

cwule, 2018-02-07:
I am trying to build the OpenCV examples. 

My OpenCV (3.3.1) with VTK (tried both 7.0 & 7.1.1) built without errors, and I can run a program including ``#include <opencv2/opencv.hpp>`` and eg. the ``imshow()`` method without problems.

However, trying to build all the Realsense OpenCV examples (rs-dnn, grabcut, imshow, latency) led to the following error:

``
Error	LNK1104	cannot open file 'vtkRenderingOpenGL.lib'
``

I found that this might be an issue when building VTK with OpenGL2 vs. OpenGL. Unfortunately, trying to build vtk with these options did not resolve the issue yet.

freemanlo, 2018-02-08:
Hi @cwule 
I don't see the build issue of RealSense OpenCV examples. My build setup as follows:
Win10 14393 / VS2015 / librealsense 2.9.1 / OpenCV 3.4

cwule, 2018-02-08:
Thanks a lot for the comment. I got it to work with the OpenCV 3.4 prebuilt binaries.

closed 2018-02-08 by cwule