realsenseai/librealsense #1146: Cannot open file 'realsense.lib'

praveenlovely, 2018-02-07:
I am trying to get the Python Wrapper working using the steps described in the readme file. Here is what I did.
1. Built the libraries using CMAKE-GUI and opened the solution "Realsense Python Wrappers"
tried to build the project pyrealsense2 and I am getting the following error:

"LNK1104 Cannot Open File "realsense2.lib""

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { / / SR300 / / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) /   | 
| Kernel Version (Linux Only)    |       05.07.05.00                            | 
| Platform                                 | PC/  |
| SDK Version                            | SDK 2.0 build 2.9.1 }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

freemanlo, 2018-02-08:
Hi @praveenlovely 
what's your Win10 version? Use the cmake configuration as the attached picture, there is no problem to generate "realsense2.lib". Please try to build the all librealsense2 solution instead of building python wrappers solution only. Then, "realsense2.lib" should generate normally.
[image]

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
Hi @praveenlovely

Is the issue resolved? Still need any further help?

closed 2018-03-09 by RealSense-Customer-Engineering