realsenseai/librealsense #1147: Migrating from Kinect & SDK integration w/ Microsoft?

valgaze, 2018-02-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n/a | 
| Firmware Version                   | n/a | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  n/a                                         | 
| Platform                                 | PC |
| SDK Version                            |  n/a                          | 

Hi RealSense community!

Like many others my team was using the Kinect V2 sensor (with its ecosystem of Visual Gesture Builder & Kinect Studio) to quickly train gestures.

[Since the Kinect is going extinct]([discontinued](https://www.fastcodesign.com/90147868/exclusive-microsoft-has-stopped-manufacturing-the-kinect)) and with it all of its amazing gesture training/tagging tools like [Visual Gesture Builder](https://channel9.msdn.com/Blogs/k4wdev/Custom-Gestures-End-to-End-with-Kinect-and-Visual-Gesture-Builder), we'll need to migrate to a new system.

I'm especially curious what this community has found works best for training in custom gestures.

- Are there any rough equivalents to the Kinect Visual Gesture Builder for RealSense? 
	
	ex. 

	- Record clips of actors attempting a gesture then tag/train it

	- Emit events when those gestures occur with a payload like gesture name, system confidence of the gesture, & bodyId

Also on the main [Microsoft Kinect site](https://developer.microsoft.com/en-us/windows/kinect), current users are directed to Intel's RealSense cameras & the text hints that Microsoft is working w/ Intel to "provide an option" for developers to transition away:

"*Microsoft is working with Intel to provide an option for developers looking to transition from the Kinect for Windows platform. Microsoft will continue to provide support for the Kinect for Windows SDK via our online forums, premiere and paid technical support. As developers transition from Kinect hardware, Microsoft encourages developers to look into Intel’s RealSense depth cameras.*"

- Are there any details about what this might entail?

- Specifically (and most important) what are the plans for software tools like Visual Gesture Builder to work w/ RealSense cameras under this arrangement?

jlgarcia75, 2018-02-14:
Hello @valgaze,

RealSense focuses on providing depth across numerous platforms and wrappers.  For Kinect users, we do not have the middleware Kinect has.  We are evaluating options with third parties and will announce more as we make the decision to adopt middleware for RealSense D400 series.

Regards,
Jesus G.
Intel Customer Support

RealSense-Customer-Engineering, 2018-02-28:
[Realsense Customer Engineering Team Comment] 
Hello @valgaze,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

valgaze, 2018-03-01:
Hey Jesus--

Hi @ jlgarcia75 --

Hi @jlgarcia75 --

Nope all is resolved for now-- very interested to learn which 3rd-party gesture support the Realsense team adopts!

closed 2018-03-01 by valgaze