realsenseai/librealsense #1148: Using device 0, an Intel RealSense USB2

zenving, 2018-02-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   |05.08.15.00| 
| Operating System & Version |  ubuntu16 | 
| Kernel Version (Linux Only)    |  4.13.0-32-generic                                      | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.9.1. }                          | 

### Issue Description

First : The name in Intel RealSense Viewer is " Intel RealSense USB2" not "D415", and I make sure I use usb3.0 cable and my pc is support usb3.0 too.
Second : I apt-get the sdk just step and step with the tutortual, and use the example code, but the I can't use c++ api, it's tell me can't get the frames, just like "terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 5000
" ,what's more  :"

 There are 2 connected RealSense devices.
Using device 0, an Intel RealSense USB2
    Serial number:746112061198
    Firmware version: 05.08.15.00"

 why there are 2 connected RealSense devices ? I just connect one.

Please help me, thanx!!

icarpis, 2018-02-08:
Hello @zenving, could you please post the output of `lsusb`, `lsusb -t` and `rs-enumerate-devices` ?

zenving, 2018-02-08:
@icarpis  Thanks a lot. It's my fault, I think the real reason is that the realsense can't be recognized as a USB3.0 device.Maybe it's my computer's problem? But my computer have USB3.0 port  so I change another computer, it's the same fault, can't be recognized as a USB3.0 device, what should I do ? Please help me, thank you!!

And there are some output of the lsusb ....

`Bus 002 Device 004: ID f055:9800  
Bus 002 Device 003: ID 046a:0011 Cherry GmbH G83 (RS 6000) Keyboard
Bus 002 Device 006: ID 8086:0ad6 Intel Corp. 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

`/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 2: Dev 6, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 2: Dev 6, If 1, Class=Video, Driver=uvcvideo, 480M
        |__ Port 2: Dev 6, If 2, Class=Video, Driver=uvcvideo, 480M
        |__ Port 2: Dev 6, If 3, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 7: Dev 4, If 1, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 7: Dev 4, If 2, Class=CDC Data, Driver=cdc_acm, 12M
        |__ Port 7: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
`
`Device info: 
    Name                          : 	Intel RealSense USB2
    Serial Number                 : 	746112061198
    Firmware Version              : 	05.08.15.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD6
`

icarpis, 2018-02-08:
It seems that you connected the camera to `port 1` under `Bus 2` with a speed of `480 Mbps` which means that it is a **`USB2`** interface.
Please connect the camera to one of the ports under `Bus 6` or `Bus 4`. According to the output you provided these two Buses supporting **`USB3`** interface.

zenving, 2018-02-08:
Maybe I know the REAL reason, it's my type-c cable, I just have the camera, not with the attached cable, so the cable I am using is not USB3.0, it' s a USB2.0 cable, I am sad. Tomorrow I will get a new cable to try again, I think it's will work. Thanks a lot !!!!!

jb455, 2018-02-08:
If it's the cable you got with the camera, try plugging the USBC connector into the camera the other way around. I had this problem with both a 415 and 435 and both the cables they came with, turning the plug over fixed it immediately despite the fact that USBC is supposed to work either way around!

zenving, 2018-02-09:
I change my cable, it works. THANKS A LOT ！！！ @icarpis

zenving, 2018-02-09:
@icarpis I think you should post  post the output of lsusb, lsusb -t and rs-enumerate-devices too．．．

closed 2018-02-09 by zenving