realsenseai/librealsense #1149: error: ‘class rs2::sensor’ has no member named ‘get_motion_intrinsics’

kimkildae, 2018-02-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / **D400** } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu **14**/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
hello ~~
I want install realsense-2.0.1 but i have some problem

/catkin_ws/src/realsense-2.0.1/realsense_ros_camera/src/realsense_camera_node.cpp:993:58: error: ‘class rs2::sensor’ has no member named ‘get_motion_intrinsics’
             auto imuIntrinsics = _sensors[stream_index]->get_motion_intrinsics(stream_index.first);
                                                          ^
make[2]: *** [realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/src/realsense_camera_node.cpp.o] Error 1
make[1]: *** [realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

icarpis, 2018-02-08:
Hello @kimkildae, the [ROS wrapper 2.0.1](https://github.com/intel-ros/realsense/tree/2.0.2) compiles only with [RealSense SDK 2.9.1](https://github.com/IntelRealSense/librealsense/tree/v2.9.1).
Please install the SDK following the [installation instruction](https://github.com/IntelRealSense/librealsense/blob/v2.9.1/doc/distribution_linux.md) and update if it solved the issue.

kimkildae, 2018-02-08:
Thank you
I already install RealSense SDK 2.9.1 following instruction
(https://github.com/IntelRealSense/librealsense/blob/v2.9.1/doc/distribution_linux.md)

but, i have this problem

icarpis, 2018-02-11:
Sorry my bad, the [ROS wrapper 2.0.1](https://github.com/intel-ros/realsense/releases/tag/2.0.1) compiles with [RealSense SDK 2.8.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.8.1).
Please [remove](https://github.com/IntelRealSense/librealsense/blob/v2.8.1/doc/distribution_linux.md#uninstalling-the-packages) the SDK 2.9.1 Debian Package and install the [2.8.1](https://github.com/intel-ros/realsense/releases/tag/2.0.1) from [sources](https://github.com/IntelRealSense/librealsense/blob/v2.8.1/doc/installation.md).

kimkildae, 2018-02-12:
Only i can install debian package because my cmake vesion is 2.8.12.2.
when upgrade cmake, my other packages do not run...
so i try to install ros wrapper 2.0.2 there is some issue 

**CMake Error at realsense-2.0.2/realsense_ros_camera/CMakeLists.txt:21 (message):

   OpenMP is missing!

-- Configuring incomplete, errors occurred!**

gcc version 5.4.1 20160904 (Ubuntu 5.4.1-2ubuntu1~14.04)

icarpis, 2018-02-12:
Regarding the compilation issue please try to compile the ROS wrapper without OpenMP by the following line `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_OPENMP=false`

kimkildae, 2018-02-12:
Thank you
I try to catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_OPENMP=false

but there are some problem 

/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/librealsense-file.a(rate.cpp.o): unrecognized relocation (0x2a) in section `.text.startup'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status

icarpis, 2018-02-12:
@kimkildae, could you please post the whole output ?

kimkildae, 2018-02-13:
Thank you for replying

```
Base path: /home/eti/catkin_ws
Source space: /home/eti/catkin_ws/src
Build space: /home/eti/catkin_ws/build
Devel space: /home/eti/catkin_ws/devel
Install space: /home/eti/catkin_ws/install
####
#### Running command: "cmake /home/eti/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/eti/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/eti/catkin_ws/install -G Unix Makefiles" in "/home/eti/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/eti/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/eti/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/eti/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 12 packages in topological order:
-- ~~  - geometry (metapackage)
-- ~~  - medicart_msgs
-- ~~  - eigen_conversions
-- ~~  - kdl_conversions
-- ~~  - hokuyo_node
-- ~~  - tf
-- ~~  - realsense_ros_camera
-- ~~  - rosaria
-- ~~  - bag2bi
-- ~~  - navigation
-- ~~  - rosaria_client
-- ~~  - tf_conversions
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'geometry'
-- ==> add_subdirectory(geometry/geometry)
-- +++ processing catkin package: 'medicart_msgs'
-- ==> add_subdirectory(medicart_msgs)
-- +++ processing catkin package: 'eigen_conversions'
-- ==> add_subdirectory(geometry/eigen_conversions)
-- +++ processing catkin package: 'kdl_conversions'
-- ==> add_subdirectory(geometry/kdl_conversions)
-- +++ processing catkin package: 'hokuyo_node'
-- ==> add_subdirectory(hokuyo_node)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
-- +++ processing catkin package: 'tf'
-- ==> add_subdirectory(geometry/tf)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
-- tf: 1 messages, 1 services
-- +++ processing catkin package: 'realsense_ros_camera'
-- ==> add_subdirectory(realsense-2.0.2/realsense_ros_camera)
-- Using these message generators: gencpp;genlisp;genpy
-- Create Debug Build.
-- realsense_ros_camera: 2 messages, 0 services
-- +++ processing catkin package: 'rosaria'
-- ==> add_subdirectory(rosaria)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
-- rosaria: 1 messages, 0 services
[... 90 lines cut ...]
[ 65%] Built target odom_node
[ 68%] Built target RosAria
[ 70%] Built target scan_node
[ 71%] Built target scan_odom_node
[ 73%] Built target Fingerprint
[ 75%] [ 76%] Built target GlobalLocalizer
Built target GlobalLocationSimpleViewer
[ 78%] Built target GlobalLocationViewer
[ 79%] Built target enable_motors
[ 81%] Built target go_three_second
[ 82%] Built target interface
[ 84%] [ 85%] Built target spin_clockwise
[ 87%] Built target print_state
Built target spin_counterclockwise
[ 89%] Built target teleop
[ 92%] Built target tf_conversions
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/librealsense-file.a(rate.cpp.o): unrecognized relocation (0x2a) in section `.text.startup'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/eti/catkin_ws/devel/lib/librealsense_ros_camera.so] Error 1
make[1]: *** [realsense-2.0.2/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

```

RealSense-Customer-Engineering, 2018-07-08:
[Realsense Customer Engineering Team Comment]
Hi @kimkildae,

I will close this first. If you still have any issue, you can still re-open or create another ticket.

closed 2018-07-08 by RealSense-Customer-Engineering