realsenseai/librealsense #1153: Unable to build for x64

jb455, 2018-02-08:
Hi,
I'm trying to build realsense2 for x64 using Visual Studio 2017 but I keep getting the error `realsense-file.lib(serialization.obj) : fatal error LNK1112: module machine type 'x86' conflicts with target machine type 'x64'`. I generated the solution using CMake 3.10.0 in Windows 10 with OpenMP turned off (though I've been unable to build it for x64 with normal settings too). Building for Win32 works without any problems.

I've deleted the "Additional Options" line from each Property Pages>Linker>Command Line which were initially causing problems (the `/machine:X86` flag overriding the set Target Machine), checked the Target Machine is `MachineX64` for all projects (including realsense-file), tried cleaning, rebuilding etc, but I still keep getting that error.
Are there any other changes I need to make to get this to build?

Thanks,
James

zivsha, 2018-02-08:
Hi James,
There shouldn't be any issue building the library for x64 platform, I am using VS 2017 (with the "Load Folder" feature, also by using CMake directly- both GUI and terminal) and I don't see this issue.

Can you describe what exactly you are doing before getting this error?
Please note that when running CMake you should choose the "Visual Studio 15 2017 **Win64**" generator:
[image]

If you can, please delete the CMake cache, and try again, this time paste the CMake "Configure" output here as well.

jb455, 2018-02-08:
Ah, I'd missed that step. Works now, thanks for your swift reply!

closed 2018-02-08 by jb455