realsenseai/librealsense #1156: offset when try to align to depth issue

travisWangXiaowei, 2018-02-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |SR300 | 
| Firmware Version                   | (3.21.0.0| 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  2.9.0                          | 

I'm trying to detect some mark points in color frame and  deproject them.  The problem is that two aligned frames(color frame and depth frame) are not actually spatially aligned.  Which means pixel[i,j] in color frame is not at the same position in depth frame.
[image]
[image]

as you can see, the line in two frames are at different positions.

If I get a pixel in color frame, how can I project this pixel to the depth frame?
Thank you very much.
here is my code 
`rs2::frameset frameset;
		frameset = pipe1.wait_for_frames(); // Wait for next set of frames from the camera
		auto processed = align.proccess(frameset);
		rs2::depth_frame aligned_depth_frame = frameset.get_depth_frame();
		rs2::video_frame other_frame = frameset.get_color_frame();

		auto s_color_inst = other_frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
		auto s_depth_inst = aligned_depth_frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
		auto s_depth2color_extrincs = aligned_depth_frame.get_profile().get_extrinsics_to(other_frame.get_profile());

		if (!aligned_depth_frame || !other_frame)
		{
			continue;
		}
		Mat rightColorImage = frame_to_mat(other_frame);
		Mat colorized = frame_to_mat(color_map(aligned_depth_frame));
		Mat res;
		_gammaCorrection(rightColorImage, res, 1 / 2.2f);
		//219 373
		//202 452
		cv::Point u(219, 373);
		cv::Point v(202, 452);
		cv::line(rightColorImage, u, v, Scalar(255, 255, 255), 3);
		cv::line(colorized, u, v, Scalar(255, 255, 255), 3);
		imshow("depth", colorized);
		imshow("color", rightColorImage);
		if (waitKey(100) == 27) {
			break;
		}`
see also
#1051

zivsha, 2018-02-09:
Hi @travisWangXiaowei , please see my answer in #1051, let me know if you have any questions

travisWangXiaowei, 2018-02-10:
Hi, @zivsha
so rs::align object cannot guarantee that depth and color frame all spatially aligned? I need to deproject every pixel in depth frame in order to find the target pixel?

zivsha, 2018-02-11:
> so rs::align object cannot guarantee that depth and color frame all spatially aligned?

It does guarantee that, the result from `align::process` is a frame set of spatially aligned images.

> I need to deproject every pixel in depth frame in order to find the target pixel?
If you use align, it already does that, so you can access `depth[i][j]` to get the depth of color_pixel[i][j]` and vice versa

jb455, 2018-02-12:
@travisWangXiaowei in lines 4 & 5 of your code you should use **processed**.get_depth_frame() (and same for colour) - it should be aligned then.

zivsha, 2018-02-12:
@jb455 nice catch, @travisWangXiaowei is this the real issue here?

RealSense-Customer-Engineering, 2018-03-02:
[Realsense Customer Engineering Team Comment] 
hi @travisWangXiaowei,

would like to know if you still need help on this topic?
thanks.

closed 2018-03-09 by RealSense-Customer-Engineering