realsenseai/librealsense #1159: Configuration file load

AleBasso80, 2018-02-09:
Hello,
I saved in the RealSenseViewer the current settings to file, then, when I try to load the file I've just saved, I get this error: Unknown in rs2_load_json(dev:0x7f9d500008c0, json_content:0x2de4000, content_size:3976):
Operation failed with error code=-6.

In the same way I'm not able to open it in my C++ code.

Any suggestion?

Thank you

zivsha, 2018-02-09:
Can you post the file here?

AleBasso80, 2018-02-13:
Hello,
this is a config json file saved by the Realsense viewer that after does not work:

{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "0",
    "aux-param-colorcorrection11": "0",
    "aux-param-colorcorrection12": "0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "0",
    "aux-param-colorcorrection6": "0",
    "aux-param-colorcorrection7": "0",
    "aux-param-colorcorrection8": "0",
    "aux-param-colorcorrection9": "0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "3",
    "param-censususize": "9",
    "param-censusvsize": "3",
    "param-colorcorrection1": "0.298828",
    "param-colorcorrection10": "0",
    "param-colorcorrection11": "0",
    "param-colorcorrection12": "0",
    "param-colorcorrection2": "0.293945",
    "param-colorcorrection3": "0.293945",
    "param-colorcorrection4": "0.114258",
    "param-colorcorrection5": "0",
    "param-colorcorrection6": "0",
    "param-colorcorrection7": "0",
    "param-colorcorrection8": "0",
    "param-colorcorrection9": "0",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "1",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "850",
    "param-lambdacensus": "19",
    "param-leftrightthreshold": "32",
    "param-maxscorethreshb": "1949",
    "param-medianthreshold": "276",
    "param-minscorethresha": "68",
    "param-neighborthresh": "4",
    "param-raumine": "7",
    "param-rauminn": "1",
    "param-rauminnssum": "6",
    "param-raumins": "3",
    "param-rauminw": "6",
    "param-rauminwesum": "14",
    "param-regioncolorthresholdb": "0.0107632",
    "param-regioncolorthresholdg": "0.645793",
    "param-regioncolorthresholdr": "0.0166341",
    "param-regionshrinku": "4",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "1",
    "param-robbinsmonroincrement": "4",
    "param-rsmdiffthreshold": "5.0625",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.446429",
    "param-scanlineedgetaub": "4",
    "param-scanlineedgetaug": "542",
    "param-scanlineedgetaur": "982",
    "param-scanlinep1": "479",
    "param-scanlinep1onediscon": "4",
    "param-scanlinep1twodiscon": "25",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "69",
    "param-scanlinep2twodiscon": "225",
    "param-secondpeakdelta": "6",
    "param-texturecountthresh": "1",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-ir-format": "Y8",
    "stream-width": "1280"
}
`

AleBasso80, 2018-02-13:
Sorry, 
I closed the issue by mistake

zivsha, 2018-02-13:
Which version of the SDK are you using? When I save this to file and load it, it works for me on `v2.10.0`

AleBasso80, 2018-02-13:
It does not work for me even with your version: same error.

dorodnic, 2018-02-14:
Hi @AleBasso80 
What OS and what firmware version are you using?

AleBasso80, 2018-02-14:
Sorry, I forgot the firmware version: 2.10.0

dorodnic, 2018-02-14:
Could you please try running `rs-enumerate-devices` and attach its output?

AleBasso80, 2018-02-14:
Device info: 
    Name                          : 	Intel RealSense 435
    Serial Number                 : 	733512070771
    Firmware Version              : 	05.08.15.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 2	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 25Hz	Y16
    Infrared 2	  1280x800	@ 25Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x800	@ 15Hz	Y8
    Infrared 2	  1280x720	@ 30Hz	Y8
    Infrared 1	  1280x720	@ 30Hz	Y8
    Infrared 1	  1280x720	@ 15Hz	Y8
    Infrared 2	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 2	  1280x720	@ 6Hz	Y8
    Infrared 2	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 30Hz	Y8
    Infrared 1	  848x480	@ 30Hz	Y8
    Infrared 2	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 6Hz	Y8
    Infrared 2	  848x480	@ 6Hz	Y8
    Infrared 2	  640x480	@ 90Hz	Y8
    Infrared 1	  640x480	@ 90Hz	Y8
    Infrared 1	  640x480	@ 60Hz	Y8
    Infrared 2	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 2	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 15Hz	Y8
    Infrared 1	  640x480	@ 6Hz	Y8
    Infrared 2	  640x480	@ 6Hz	Y8
    Infrared 1	  640x400	@ 25Hz	Y16
    Infrared 2	  640x400	@ 25Hz	Y16
    Infrared 1	  640x400	@ 15Hz	Y16
    Infrared 2	  640x400	@ 15Hz	Y16
    Infrared 1	  640x360	@ 90Hz	Y8
    Infrared 2	  640x360	@ 90Hz	Y8
    Infrared 2	  640x360	@ 60Hz	Y8
    Infrared 1	  640x360	@ 60Hz	Y8
    Infrared 1	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 15Hz	Y8
    Infrared 1	  640x360	@ 15Hz	Y8
    Infrared 2	  640x360	@ 6Hz	Y8
    Infrared 1	  640x360	@ 6Hz	Y8
    Infrared 2	  480x270	@ 90Hz	Y8
    Infrared 1	  480x270	@ 90Hz	Y8
    Infrared 1	  480x270	@ 60Hz	Y8
    Infrared 2	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 6Hz	Y8
    Infrared 2	  480x270	@ 6Hz	Y8
    Infrared 1	  424x240	@ 90Hz	Y8
[... 207 lines cut ...]
    Color	  320x240	@ 30Hz	YUYV
    Color	  320x240	@ 6Hz	Y16
    Color	  320x240	@ 6Hz	BGRA8
    Color	  320x240	@ 6Hz	RGBA8
    Color	  320x240	@ 6Hz	BGR8
    Color	  320x240	@ 6Hz	RGB8
    Color	  320x240	@ 6Hz	YUYV
    Color	  320x180	@ 60Hz	Y16
    Color	  320x180	@ 60Hz	BGRA8
    Color	  320x180	@ 60Hz	RGBA8
    Color	  320x180	@ 60Hz	BGR8
    Color	  320x180	@ 60Hz	RGB8
    Color	  320x180	@ 60Hz	YUYV
    Color	  320x180	@ 30Hz	Y16
    Color	  320x180	@ 30Hz	BGRA8
    Color	  320x180	@ 30Hz	RGBA8
    Color	  320x180	@ 30Hz	BGR8
    Color	  320x180	@ 30Hz	RGB8
    Color	  320x180	@ 30Hz	YUYV
    Color	  320x180	@ 6Hz	Y16
    Color	  320x180	@ 6Hz	BGRA8
    Color	  320x180	@ 6Hz	RGBA8
    Color	  320x180	@ 6Hz	BGR8
    Color	  320x180	@ 6Hz	RGB8
    Color	  320x180	@ 6Hz	YUYV

dorodnic, 2018-02-25:
Hi @AleBasso80 
I would like to make sure the issue still reproduces with latest release. Also, are you able to load this JSON using the RealSense Viewer? We tried it on several systems and could not reproduce the problem. What C++ code you are using to load the JSON? 
Is it the same camera you used to generate the file?

AleBasso80, 2018-02-26:
Hi @dorodnic 
I'm using the last release. 
I made two different tests:
1) I only use the RealSense Viewer: I save the current configuration as a json file with the embedded function (the button besides the presets list) and then I just try to reload it. I get an unknown error in rs2_load_json.
2) in c++ I use this code that I found in another issue:
auto profile = pipe.start();
std::ifstream t("test.json");
std::string str((std::istreambuf_iterator<char>(t)),
                std::istreambuf_iterator<char>());
rs400::advanced_mode dev = profile.get_device();
dev.load_json(str);
I get a parsing error, which I guess is the same I get in the first case.

Thank you

Nicogene, 2018-03-26:
Hi,

I'm encountering the same issue.

```bash
$ rs-enumerate-devices
 Device info: 
    Name                          :     Intel RealSense 435
    Serial Number                 :     728612070403
    Firmware Version              :     05.08.15.00
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/video4linux/video1
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B07

```
```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:        16.04
Codename:       xenial
```

SDK version: 2.10.2-0

Nicogene, 2018-06-08:
Hi,

this is my actual setup:

```
$ rs-enumerate-devices

 Device info: 
    Name                          :     Intel RealSense D415
    Serial Number                 :     739112060128
    Firmware Version              :     05.08.15.00
    Recommended Firmware Version  :     05.09.09.02
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video1
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD3

```

`librealsense 2.12.0-0~realsense0.73`

and after saving a .json from the `realsense-viewer`, the load of the same file fails.

here's the file:

```json
{
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.461914",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "33000",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "400",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-ir-format": "Y8",
    "stream-width": "1280"
}
```

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

i've checked your JSON file, somehow there are two line of "controls-color-white-balance-auto",
please remove the 2nd one then it's working good here.

Nicogene, 2018-06-11:
Hi,
Thank for the reply!
The json that I posted has been generated from the `realsense-viewer`, so probably the issue in the generation phase instead of the parsing one?

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

yup, I've consider about this too, so I compared with the one I generate, it have no issue.
but this remind me that I am using Realsense viewer from LibRealsense v2.12.0 and FW 5.9.11.
I see your FW version is v5.8.15, not sure your Realsense viewer version?
and BTW, if you have no concern about FW version, I would suggest to update your FW with latest.
thanks.

Nicogene, 2018-06-11:
Ok I will update the firmware and I will let you know thanks !

RealSense-Customer-Engineering, 2018-06-21:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

would like to know if your test with new FW is good?

Nicogene, 2018-06-21:
Hi,

I have updated the firmware(05.09.11.00), the save and the load of the json works, but after loading the json when I try to enable the color stream it complaints that a configuration is not suppported.

The problem is that now after updating the firmware the depth  stream stopped working, starting the `realsense-viewer` I get this error:
```
Invalid Value in rs2_get_option_range(options:0x7f4140005340, option:StereoOpCodes did not match! Sent 21 but received -6
```
and this is what I see:

[image]

I tried the camera just before updating the firmware and it worked fine.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

can you please try "Intel.Realsense.CustomRW.exe –g" from https://downloadcenter.intel.com/download/27415/?v=t.
thanks.

Nicogene, 2018-06-26:
After running `Intel.Realsense.CustomRW.exe –g` it works again thanks!

> I have updated the firmware(05.09.11.00), the save and the load of the json works, but after loading the json when I try to enable the color stream it complaints that a configuration is not suppported.

On the other hand, this problem persist

Nicogene, 2018-06-27:
Hi @RealSense-Customer-Engineering 

For the camera D415 broken and then fixed this piece of code throws an exception:

```c++
    rs2::frameset data;
    try
    {
        data = m_pipeline.wait_for_frames();

    }
    catch (const rs2::error& e)
    {
        yError() << "realsense2Driver: failed to get frames:"<< "(" << e.what() << ")";
        return false;
    }
```
where `m_pipeline` is a  `rs2::pipeline`.
With other D415/435(firmware not updated) this piece of code works smoothly, the error printed is:
```
Frame didn't arrived within 5000
```
On the pc where this issue occurs I have  `librealsense v2.10.3`
Do you have any idea which could be the problem?
Before the firmware update everything was fine.

RealSense-Customer-Engineering, 2018-06-28:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

from previous experience, this was caused by the cable quality or not connected to USB3 port,
can you please confirm you connect to right port and see if Realsense Viewer works?

Nicogene, 2018-06-28:
Hi @RealSense-Customer-Engineering,

The strange thing is that with the `realsense-viewer` the camera works without problem, instead that piece of code throws an execepion.
Is it possible that inside the viewer `pipeline.wait_for_frame()` is not called?

RealSense-Customer-Engineering, 2018-07-03:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

sorry, you are right, should compare with rs-capture which used pipeline.wait_for_frame().
can you please provide log with running rs-capture?
https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs

and i noticed you was using libRealsense v2.12.0 and v2.10.3, do you have issue on both versions? or only v2.10.3?
thanks.

Nicogene, 2018-07-03:
Hi @RealSense-Customer-Engineering 

I tried `rs-capture` and the piece of code containing `pipe.wait_for_frame()` and in both cases sometimes it works and sometimes I get the timeout exception(`Frame didn't arrived within 5000`).
(I tried with `librealsense2` v2.10.3 and v2.12.0)
Probably you are right there is a problem with the usb cable.
I noticed that when the problem happen the device seems to be not opened properly because the infrared stays turned off.
When the infrared light up it works smoothly.

RealSense-Customer-Engineering, 2018-07-12:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

if it's cable issue, suggest you to find a USB3 certified cable then it should have no problem.

Nicogene, 2018-07-12:
Hi @RealSense-Customer-Engineering,

Thanks for support!

RealSense-Customer-Engineering, 2018-07-19:
[Realsense Customer Engineering Team Comment]
hi @Nicogene,

Great, let me close this ticket here and please feel free to open new ticket if any question.

closed 2018-07-19 by RealSense-Customer-Engineering