realsenseai/librealsense #1160: Not able to update the sdk 2.0 to newest version  build 2.10.0

nrasulnrasul, 2018-02-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16/| 
| Kernel Version (Linux Only)    |  4.13.0-31 |                                       | 
| Platform                                 | PC|
| SDK Version                            |  2.1| 

### Issue Description
Hi,

I am not able to update the sdk 2.0 to newest version **Intel® RealSense™ SDK 2.0 (build 2.10.0)**.

I got an error  with ` sudo apt-get update` . It has some problem with signature verification, check the error below.

```
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
W: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/Release.gpg  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
```

Thank you.

dorodnic, 2018-02-11:
Hi @nrasulnrasul 
We have detected a problem with 2.10 (also reported [here](https://github.com/IntelRealSense/librealsense/issues/1165)) during Linux validation over the weekend and decided to postpone publishing the package for 2.10. 
You can still build it from source, but most likely both problems will be resolved later today.

dorodnic, 2018-02-11:
Should be resolved now. 

Please note:
1. We had to change signature public key, please re-run:
```shell
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
```
2. `udev-rules` package is very sensitive to kernel version, please review some of the recent issues if you encounter any problems

nrasulnrasul, 2018-02-12:
Thanks @dorodnic , Now I was able to update to 2.10.0

closed 2018-02-12 by nrasulnrasul