realsenseai/librealsense #1164: D415 loses IR stream on restart...

drhalftone, 2018-02-11:
I got my D415, and there are no issues with regards to grabbing IR and depth; however, I did have an issue with trying to write to a memory card on the same USB 3 hub as the sensor.  After a while, the camera stops returning video in the 100 msec time I expected it.  I suspect that the bus is busy writing video to disk, but its not this specific situation that is the problem.  What is the problem is that I decided that if this situation happens where the camera stops sending video, I’d like to be able to reset the connection and restart the video.  So I wrote a version of my code that grabs video, and then on the 50th frame, just decides to close the connection, deleting all pointers to RS objects, and then tries to reconnect from scratch.  On the restart, the particular video stream of IR video at full resolution and 30 fps is no longer listed as an available stream.  All the other resolutions and frame rates are still there, just not the one I was using before I triggered the reset.

dorodnic, 2018-02-11:
Hi @drhalftone 
Could you please specify what operating system you are using, and attach the output of `rs-enumerate-device -m` before and after the reset?

jlgarcia75, 2018-02-15:
Hello @drhalftone,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

drhalftone, 2018-02-16:
Yes, I will post a sample application to demonstrate the issue shortly

drhalftone, 2018-02-20:
> It sounds like the camera is going out of advanced mode.
> I've noticed that the IR streams on the D435 are only available when the device is in advanced mode.
>
> You can programatically check if the device is in advanced mode.

I have the D415.  Is there a function call for advanced mode that I’m missing in my code?

ioreed, 2018-02-20:
Found the C++ version while looking for our own uses: https://github.com/IntelRealSense/librealsense/blob/9437bc24c0f4ce16193f1045ceea703a8151b96d/src/ds5/advanced_mode/advanced_mode.cpp

drhalftone, 2018-02-20:
This seems like way overkill.  In my original application, I had no problem grabbing IR over depth.  I just couldn’t reset the camera to start over.  In the test application I posted, I have no problem grabbing IR over depth for those first 15 frames.  But it stops on the 16th.  

Do you really think this is necessary to get basic functionality out of the camera?

ioreed, 2018-02-20:
Just a user, not from Intel. As best we can tell, the code is a cpp implementation of advanced features like those in the RealSense viewer.  We are implementing some of these in our code. The response is an answer to your original question, "Is there a function call for advanced mode that I’m missing in my code?" The idea is to use the example as a starting point and implement what you need. There are some very useful features in there for noise reduction, etc. 

We found that many lock-ups and other problems were solved by using USB 3.1 (not USB 3.0). There is a good Linux compatible card on Amazon at https://www.amazon.com/gp/product/B018YSB2I8/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1. We upgraded a five yr old Dell running Ubuntu 16.04 LTS with this and can run our D415 on a 3m USB cable without loss.

agrunnet, 2018-02-25:
@drhalftone regarding the issue of streaming past 15 frames, this is expected behavior if you don't **release frames**. I confess I have not looked at your code so do not know if that is your specific issue, but it sounded familiar enough that I thought I would add that recommendation.

drhalftone, 2018-02-25:
Interesting.  Which frame am I supposed to release.  I have the following line of code:

     rs2_frame *frames = rs2_pipeline_wait_for_frames(pipeline, 15000, &err);

followed by:

     rs2_frame *frame = rs2_extract_frame(frames, n, &err);

I later release "frame" with:

     rs2_release_frame(frame);

but not "frames"  How do I release "frames"

dorodnic, 2018-02-26:
Hi @drhalftone 
You should release `frames` in the same way. Alternatively, you can call `rs2_keep_frame` if you want to keep more then 16 frames at a time (at some minimal performance expense)

drhalftone, 2018-04-16:
> [Realsense Customer Engineering Team Comment]
> @drhalftone <https://github.com/drhalftone>
> Do you still have issue on this topic? Can I close it?

yes close it.

closed 2018-04-16 by RealSense-Customer-Engineering