realsenseai/librealsense #1165: python.cpp fails to build in release 2.10

bitzombie1, 2018-02-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | newest version | 
| Operating System & Version |  Ubuntu 16.04.3 LTS  | 
| Kernel Version (Linux Only)    | 4.13.0-32-generic                                 | 
| Platform                                 | UpBoard  |
| SDK Version                            |  2.10                      | 

### Issue Description
Hello I am trying to compile SDK ver 2.10 on Ubuntu and it gets to 90% and fails with this error: 

`Scanning dependencies of target pyrealsense2
[ 90%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/python.cpp.o
/home/charles/librealsense-2.10.0/wrappers/python/python.cpp: In lambda function:
/home/charles/librealsense-2.10.0/wrappers/python/python.cpp:86:147: internal compiler error: Segmentation fault
 D_RAW_ARRAY_GETTER(T, member, valueT, SIZE) [](const T& self) { return reinterpret_cast<const std::array<valueT, SIZE>&>(self.member); }
                                                                                                                                        ^
/home/charles/librealsense-2.10.0/wrappers/python/python.cpp:92:67: note: in expansion of macro ‘BIND_RAW_ARRAY_GETTER’
 #define BIND_RAW_ARRAY_PROPERTY(T, member, valueT, SIZE) #member, BIND_RAW_ARRAY_GETTER(T, member, valueT, SIZE), BIND_RAW_ARRAY_SETTER(
                                                                   ^
/home/charles/librealsense-2.10.0/wrappers/python/python.cpp:146:23: note: in expansion of macro ‘BIND_RAW_ARRAY_PROPERTY’
         .def_property(BIND_RAW_ARRAY_PROPERTY(rs2_extrinsics, rotation, float, 9))
                       ^

Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
wrappers/python/CMakeFiles/pyrealsense2.dir/build.make:62: recipe for target 'wrappers/python/CMakeFiles/pyrealsense2.dir/python.cpp.o' failed`

I'm using gcc version 5.4.0 and I had no previous problem compiling SDK 2.9.1 . I looked inside python.cpp at the line numbers 86, 92, and 146 but couldn't find any obvious reason why it would fail like this. Any input would be appreciated.

zivsha, 2018-02-11:
We've crashed GCC, nice :)
Other than names changed and no return value declaration for the lambda this is the same as `BIND_RAW_ARRAY(class, name, type, size) #name, [](const class &c) -> const std::array<type, size>& { return reinterpret_cast<const std::array<type, size>&>(c.name); }`

Can you please try to add the return value for the lambda (  `-> const std::array<type, size>&`) and see if it still fails? I will try to get a Linux machine with GCC 5.4 and test it as well.

dorodnic, 2018-02-11:
Hi @bitzombie1 
While this change passed [our CI](https://travis-ci.org/IntelRealSense/librealsense/jobs/339595213) this is serious enough to issue a hotfix. Will update soon.

zivsha, 2018-02-11:
Confirm this happens with `gcc 5.4` on my machine as well. and that adding the return type fixes this.
Will issue a patch with the following diff:
```diff
diff --git a/wrappers/python/python.cpp b/wrappers/python/python.cpp
index 5acf137..812af3c 100644
--- a/wrappers/python/python.cpp
+++ b/wrappers/python/python.cpp
@@ -83,10 +83,10 @@ std::string array_to_string(const T(&arr)[N])
     return oss.str();
 }
 
-#define BIND_RAW_ARRAY_GETTER(T, member, valueT, SIZE) [](const T& self) { return reinterpret_cast<const std::array<valueT, SIZE>&>(self.member); }
+#define BIND_RAW_ARRAY_GETTER(T, member, valueT, SIZE) [](const T& self) -> const std::array<valueT, SIZE>& { return reinterpret_cast<const std::array<valueT, SIZE>&>(self.member); }
 #define BIND_RAW_ARRAY_SETTER(T, member, valueT, SIZE) [](T& self, const std::array<valueT, SIZE>& src) { copy_raw_array(self.member, src); }
 
-#define BIND_RAW_2D_ARRAY_GETTER(T, member, valueT, NROWS, NCOLS) [](const T& self) { return reinterpret_cast<const std::array<std::array<valueT, NCOLS>, NROWS>&>(self.member); }
+#define BIND_RAW_2D_ARRAY_GETTER(T, member, valueT, NROWS, NCOLS) [](const T& self) -> const std::array<std::array<valueT, NCOLS>, NROWS>& { return reinterpret_cast<const std::array<std::array<valueT, NCOLS>, NROWS>&>(self.member); }
 #define BIND_RAW_2D_ARRAY_SETTER(T, member, valueT, NROWS, NCOLS) [](T& self, const std::array<std::array<valueT, NCOLS>, NROWS>& src) { copy_raw_2d_array(self.member, src); }
 
 #define BIND_RAW_ARRAY_PROPERTY(T, member, valueT, SIZE) #member, BIND_RAW_ARRAY_GETTER(T, member, valueT, SIZE), BIND_RAW_ARRAY_SETTER(T, member, valueT, SIZE)
```

bitzombie1, 2018-02-11:
I will try adding the code changes and recompiling.

bitzombie1, 2018-02-11:
Yay! We have a build. That fixed it. Thank you so much.

closed 2018-02-11 by dorodnic