realsenseai/librealsense #1168: openNI2 support for RealSense SDK2.0

cseJuha, 2018-02-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0.0 }                          | 

### Issue Description

Is there posibility to get realsense.dll and LibrealsenseWrapper.dll work with openNI2?

jlgarcia75, 2018-02-14:
Hello @cseJuha,

We are currently developing wrappers for the D400 cameras based on use case prioritization and technology demand. The OpenNI2 project has been dormant on GitHub for at least 4 years and is no longer maintained. Please let us know what are your specific requirements and use cases for using RealSense with OpenNI2.

Regards,
Jesus G.
Intel Customer Support

cseJuha, 2018-02-14:
Thanks @jlgarcia75 

Im looking for just getting D400 work with openNI2 viewer at this point. Im more than happy to assist on this one. You can PM me on skype cseJuha. Im sure this is something that lot of devs would like to see happen because it would make old kinect apps work with D400 on small changes.

jlgarcia75, 2018-02-14:
Hello @cseJuha,

The engineering team has been made aware of your request.

Regards,
Jesus G.
Intel Customer Support

DominionSoftware, 2018-02-14:
I’m also interested in OpenNI- I’ve ordered one of your cameras and was hoping for some standard support so I don5 have to special case each camera

cseJuha, 2018-02-14:
Asus and Orbbec has openNI2 support already on their cameras and i think many devs has move to one of these only because of this support.

DominionSoftware, 2018-02-14:
I have forked the openNI2 for my own purposes to create a CMake version of it so I can fix a couple of things. If I get that going I can make it available. Right now the OpenNI2 project doesn't use CMake which is becoming the more useful environment.

jlgarcia75, 2018-02-21:
Hello @cseJuha and @DominionSoftware, 

We have other Kinect users asking for Unity and Unreal, can these be suitable for your usages?

Also, what more do you need to move ahead to implement OpenNI support? Our wrappers are open-source, and the contribution guidelines are very simple: https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md

You can create a PR at your convenience.

Regards,
Jesus G.
Intel Customer Support

cseJuha, 2018-02-21:
Hi @jlgarcia75 Im more than happy to help. Do you have some kind of instant messaging in availability or do we only communicate tru github?

closed 2018-02-27 by jlgarcia75

DominionSoftware, 2018-02-28:
> After some research i found that orbbec has openni source available on github. could this help you guys to modify your driver to fit to openni2? https://github.com/orbbec/astra/tree/master/src/plugins/openni_sensor
>
> Unfortunately making drivers is not really my core thing so i need some help with this one.

A couple things:

The OpenNI2 code base is not based on CMake. This makes it less appealing as they give you a Visual Studio 2010 solution, for instance,  which is going to be stale soon if not already.

I have successfully moved the basics of OpenNI2 to CMake under Windows VS 2015. I did not do any of the wrapper stuff.

If anyone is interested in helping with this effort on other platforms I think it could help OpenNI move forward and make writing drivers simpler.

Let me know and I’ll push my changes up to the fork. Specifically it needs to be built and tested under Linux. And then people can put the wrappers for Java etc back in.

Thanks.

Rick Frank
Dominion Software, Inc.
825 Beacon Street
Newton, MA 02459
Medical, Scientific, and Industrial Software

janbajana, 2018-03-31:
It would be so nice if OpenNI has better support for newer cameras like real-sense and other RGB-D. Kinect and similar are too big. Microsoft by self is referencing to real-sense instead of deprecated Kinect. Realsence is capable to do the nice volumetric 3D mapping.

arunrajagopalan, 2018-05-31:
Hi @jlgarcia75 

Wonder if you have any updates on OpenNI. I got serveral d400 cameras, and openNI could make it lot easier to use them in my app

Thanks
Arun

Onsager, 2018-06-07:
I am also be interested in OpenNI / OpenNI2 support.  However I would also be happy with another good quality option (other than OpenCV).

johanromuald, 2018-06-08:
Hi,

I am also interested in OpenNI2 support. Any progress in this ??

wongfei, 2018-07-03:
Any one who interested in OpenNI, could you describe the use-cases? Now I'm working on driver and already got it working (and possible going open source soon). But OpenNI API is rich and lot of features can't be tested using default NiViewer..

Onsager, 2018-07-03:
@wongfei - Thanks for the note!  I am only interested in smooth skeletal tracking (hoping to avoid having to depend/distribute 3rd party libraries).  I am not concerned about their comprehensive API.

johanromuald, 2018-07-04:
Thanks. I am interested in getting smooth depth and IR Streams to develop my own application.

drzraf, 2018-07-11:
OpenNI is a defacto standard device abstraction layer and most successful development bricks have taken care to use it so they app can be made compatible with kinect, structure, astra, ... rather bound to a specific device.

The life of OpenNI seems the one of each company getting in the field ([PrimeSense > Apple](https://github.com/OpenNI/OpenNI2), then [Structure](https://github.com/occipital/OpenNI2) and currently [orbbec](https://github.com/orbbec/OpenNI2/tree/orbbec-dev).

I can't believe an Intel-driven ecosystem could actually _develop_ without devices being supported [there](https://github.com/orbbec/OpenNI2/pulls)... which end up here: https://github.com/IntelRealSense/OpenNI2

BingBingAR, 2018-07-17:
I'm also interesting in smooth skeletal tracking with realsense d435. @wongfei , can you help me?

wongfei, 2018-07-21:
As for now I can only provide some not official experimental binaries for Win8+. Package contains OpenNI2/Nite2 samples with RealSense2 driver and supports Color/Depth/IR streaming and user tracking.

https://drive.google.com/open?id=1udb59IunTXW9xUKlO21RYj71BYrdQOEx

ShikiGami, 2018-10-04:
I'm also interested. I'm making software that works with Orbbec Astra, Xtion 2 and Realsense D-series cameras, so it would be very interesting if we could use the same codebase using OpenNI2 for Realsense.