realsenseai/librealsense #1172: Power consumption

Alarion24, 2018-02-13:
----------------------------------------------------------------------------------------------------

| INFO                        |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.0                          | 

### Issue Description

I have issue the same problem to #685. SR300 works good on PC but not on laptop.
Device often disconnects on start streaming. Sometimes all is ok.

Unfortunately powered usb hub ([UH700](https://www.tp-link.com/us/products/details/cat-5522_UH700.html)) doesnt solve the problem :(
Do i need to try different powered usb hub?

Also i want to ask the question which one is drawing more power SR300 or D415/D435?

Alarion24, 2018-02-14:
Additional information that the device has a greater chance of depth streaming in RealSense Viewer if you first turn on the "RGB Camera" and only then turn on the "Coded-Light Depth Sensor".

jlgarcia75, 2018-02-15:
Hello @Alarion24,

We have seen that the SR300 is sensitive to the power delivery from the USB ports on PCs. Most users can get it work using a powered USB hub. Other users can get it to work when they use a powered USB hub without connecting it to power. Please refer to the datasheets for the SR300, https://www.intel.com/content/www/us/en/support/articles/000023535/emerging-technologies/intel-realsense-technology.html, and the D400 series, https://www.intel.com/content/www/us/en/support/articles/000026827/emerging-technologies/intel-realsense-technology.html, to compare the power draws. The D400 series cameras draw less power at higher resolutions.

Regards,
Jesus G.
Intel Customer Support

Alarion24, 2018-02-16:
It works if the USB hub is not connected to a power.
Thank you very much!

closed 2018-02-21 by Alarion24

tispratik, 2020-03-11:
Try a 4 Amp one (like https://www.bestbuy.com/site/insignia-4-port-usb-3-0-hub-black/5631038.p?skuId=5631038 ) instead of the UH700 which is a 1.5Amps one.