realsenseai/librealsense #1173: Issues running Unity wrapper (High CPU usage/No texture write)

Zeraphil, 2018-02-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win (10)   | 
| Platform                                 | PC  |
| SDK Version                            |  2.1 }                          | 
| Unity version 2017.3.1.f |

### Issue Description
Using the Unity wrapper, attempted to run the RealSenseTextures scene. Only the alignment scene runs, while the other streams do not write to the raw image texture, or are not rendering correctly. This is the same in both multithread and unity thread device mode. 

Additionally, running the scene with depth and color streams enabled jacks up the CPU usage to 100% *all 6 cores, Xeon E5-1660*. Disabling everything except the depth stream resolves this. CPU usage remains high whether alignment script is running or is removed from the scene.

zivsha, 2018-02-13:
Hi @Zeraphil ,
Please make sure that all binding of the stream textures exists.
For each `RealSenseStreamTexture` make sure there is an image bound to it, and that `Fetch Frames From Device` is checked:

[image]

Only `Alignment` should not fetch frames from the device.

Also, make sure each image under `Images` has a matching material and shader:
[image]

The scene we provide should have these all configured, maybe try to revert it the restore it.

Regarding the CPU usage- this is a known issue, which happens due to the usage of OpenMP in the native dll. You can build the native dll with OpenMP disable by specifying it in the CMake flags (see issue #1105), which should lower the usage, but might hit performance.

Zeraphil, 2018-02-13:
Hi Zivsha, thank you for your prompt response. Yes, the scene is as is from the repo. Only alignment works as is. That said, remove the alignment scripts/objects from the prefab and scene resolves it, I wonder if it's directly related to OpenMP. 

Alternatively setting the material texture target directly also resolves it but it's better to write to the raw image texture like you are doing. Will try your suggestion about OpenMP to confirm whether these two issues are related. Thank you!

zivsha, 2018-02-13:
There shouldn't be a connection between you not seeing the texture, and OpenMP. Sorry if my answer suggested that, it was not my intention.

Zeraphil, 2018-02-13:
Ok, no problem. Well regardless, note that for my problem, removing the alignment stream resolved the issue for the rest of the streams.

closed 2018-02-27 by jlgarcia75

elektrikmoonkey, 2018-04-05:
Hello 
I believe this problem is far from solved.
Here testing on a 6700hq + Nvidia 1060
As soon as the Colored camera starts. Cpu goes to 100%. 
Also a huge memory leak. from 4gb goes to 16gb just with color camera. 

Is it someone there checking this ?