realsenseai/librealsense #1174: Input/output error

dheera, 2018-02-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.0-112                                        | 
| Platform                                 | PC  |
| SDK Version                            |  latest from apt-get                          | 

### Issue Description
Upon starting, lots of input/output errors of various sorts. No ability to get any depth data from camera whatsoever.

[image]

[image]

[image]

dorodnic, 2018-02-14:
Hi @dheera 
Note that the camera was identified as a USB2 device. Please try connecting it to a USB3 port or via a different cable. If you have already tried this or the problem persists, please provide dmesg logs.

dheera, 2018-02-14:
Hi @dorodnic ,
Thanks for the reply. I tried two copies of a D435 on multiple USB 3.1 ports on 2 different PCs, all running Ubuntu 16.04 with the exact same issues. It detects the device as a USB 2.0 device only.
[image]
Any thoughts? Thanks!

jlgarcia75, 2018-02-15:
Hello @dheera,

Please provide the output from: lsusb -t

Regards,
Jesus G.
Intel Customer Support

dheera, 2018-02-17:
It seems to have magically resolved itself after plugging the RealSense into a 3rd PC and the plugging it back into the 1st and 2nd PC.

closed 2018-02-27 by jlgarcia75