realsenseai/librealsense #1176: RealSense SDK2.0 and hand tracking/gesture recognition algorithms

VeraCurly, 2018-02-14:
Are there any hand tracking and hand gesture recognition algorithms in RealSense SDK2.0? Does the D400 camera work with previous version of SDK? What camera and SDK version are better if I'm going to work with hand gesture recognition?

jlgarcia75, 2018-02-14:
Hello @VeraCurly,

There are currently no hand tracking and hand gesture recognition algorithms in RealSense SDK2.0. They are being evaluated for a future a release. The D400 cameras work _only_ with this SDK. 

The combination of SR300 camera and RealSense SDK for Windows Release 3 are capable of doing hand gesture recognition. However, both of these products are discontinued.

Regards,
Jesus
Intel Customer Support

VeraCurly, 2018-02-15:
Thank you for the fast answer!

closed 2018-02-15 by jlgarcia75

zrenegade, 2018-05-11:
Is there any update on this? Has the hand tracking support been added to the SDK2.0 yet?

jlgarcia75, 2018-05-11:
Hello @zrenegade,

All the hand tracking info for SDK 2.0 can be found here: https://github.com/IntelRealSense/hand_tracking_samples. Note that SDK 2.0 does not have hand-tracking support built-in. These are samples that show how to do it using the SDK.

Regards,
Jesus
Intel Customer Support

jlgarcia75, 2018-09-24:
Hello @liarokapisv,

You are right. The hand tracking samples are extremely out of date. Notice they were put up over a year ago and have not been updated since.

I recommend you take a look at Nuitrack.com, which has created an SDK with skeletal and tracking algorithms for the D400 series cameras.

Regards,
Jesus G.
Intel Customer Support

simpple28, 2019-05-06:
Can I package an Unreal project with a Windows version?

TheBricktop, 2019-11-27:
Nuitrack currently does only palm tracking and there is no finger/gesture tracking, hopefully somebody will pickup the realsense handtracking and port it to librealsense 2.

vnt1537, 2020-07-20:
> I am going to proceed with a project that recognizes hands with real license d435.
> Nulitrack says Unreal only supports Android.
> Did you package successfully with Windows version?

Hello @simpple28 , Did you come up with any hand recognizer for d435?

L-xn, 2021-04-13:
So if intel realsense D415 is used for hand tracking, should I choose opencv or Nulitrack? Seek advice! Thanks!

L-xn, 2021-04-13:
> Nuitrack当前仅执行手掌跟踪，并且没有手指/手势跟踪，希望有人能够使用realsense手动跟踪并将其移植到librealsense 2。

那么我认为Cubemos Skeleton Tracking SDK是不可行的方案。那么我要想用D415做手部追踪该怎么做啊。希望得到前辈的指点！感谢！

jlgarcia75, 2021-04-13:
Hello L-xn, you can use either one. It depends on how comfortable you are developing your own solutions and if the Nuitrack SDK provides what you are looking for. The Nuitrack SDK must be purchased. I recommend contacting Nuitrack to determine if it will fit your needs.

L-xn, 2021-04-15:
> 您好@ L-xn，您可以使用任何一个。这取决于您开发自己的解决方案的舒适程度以及Nuitrack SDK是否提供您想要的东西。必须购买Nuitrack SDK。我建议联系Nuitrack以确定它是否适合您的需求。

If I use opencv, is there any information about this? Thank you!