realsenseai/librealsense #1178: CreateNamedMutex returned Mutex_TotalFailure

esseswann, 2018-02-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300                                                     | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |  Win10  | 
| Platform                                 | PC |
| SDK Version                            |  2.10.0                         | 

### Issue Description
I'm having a strange behavior: when I try to use `pipeline.start()` or just assign a device `rs2::device = devices[some_valid_device_index]` I receive the following error:
 `CreateNamedMutex returned Mutex_TotalFailure`
The Viewer works properly and the same code seems to run well on another machine. Any suggestions?

dorodnic, 2018-02-24:
Hi @esseswann 
In general, admin permissions are not required to run the code. If you were able to discover what on your setup was causing this please share.

esseswann, 2018-02-24:
Hey, @dorodnic. The problem might be connected to `librealsense` being used inside an ActiveX control for (sigh) IE8.

dorodnic, 2018-02-25:
Thanks @esseswann, that makes sense. It is safe to comment out the named mutex class if you are certain you are the only application accessing the camera. It is used to synchronize between processes using the device at system level.

closed 2018-02-25 by dorodnic

GrandFoin, 2019-09-26:
I have the same exception.

Camera Model | SenseRS
Firmware Version | ?
Operating System & Version | Win10
Platform | PC
SDK Version | 2.10.0

When measure example is compiled in a DLL on windows 
the following code throw the exception
[14:23:24]CreateNamedMutex returned Mutex_TotalFailure

DLL is called as Native code from a C# UWP application.
Line that throws exception is 
measure->rs_measure.cpp->auto profile = pipe.start(cfg)
Line that throws exception is 
rs_pipeline.hpp->...start(...)->auto p = std::shared_ptr<rs2_pipeline_profile>(
                rs2_pipeline_start_with_config(_pipeline.get(), config.get().get(), &e),
                rs2_delete_pipeline_profile);

How can I avoid that exception?