realsenseai/librealsense #1179: Error in RealSense Viewer after plugging in D435

c-sja, 2018-02-15:
Hi there

I got my D435 around a month ago. It used to work fine on my (Windows) computer when using it via the Viewer.

Hadn't touched it in awhile and upgraded to SDK 2.0 latest version (2.10.0) to test out the new functionality. Upon plugging the device in, the Viewer always gives the following error:

`Backend in rs2_create_device(info_list:000001B8F1C39EE0, index:0):
_source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: "No such interface supported"`

I have tried using our other D435 on this computer, same error. I tried the same SDK version installed on another computer and it works fine, so it's something on my computer only. I also tried installing the old version (2.9.1) on my computer and it also had the same error now, so it must have been something that changed on my computer recently. Changing USB ports did not help.

The main things I remember changing here include installing the most recent Cuda version. Could this affect it at all?

Any help is appreciated. Thanks.

Jiloc, 2018-02-15:
Probably related to #931? Same error but when the camera is unplugged while streaming and then re-plugged in

c-sja, 2018-02-21:
I have tried uninstalling all versions of CUDA and confirm this is still happening.

c-sja, 2018-02-21:
Okay I finally got it working! I went into Device Manager and randomly tried to uninstall both drivers (under Imaging Devices) - the depth and also RGB modules. Plugged it in again and it finally worked!

Now I can't see them in the Device Manager though, as a side note.

jlgarcia75, 2018-02-27:
Hello @c-sja,

Can you send pictures of your Device Manager with nodes Imaging Devices, Cameras, and Universal Serial Bus Devices expanded?

Regards,
Jesus G.
Intel Customer Support

closed 2018-03-02 by jlgarcia75

sergioamr, 2018-07-06:
I found my answer here:
https://github.com/IntelRealSense/librealsense/issues/1930
It is related to privacy settings in 1803... Microsoft keeps breaking stuff with their release cycles.

yangliu995, 2019-08-02:
@c-sja Hi, I have received the same error, but I tried to uninstall and reconnect my device, this error is still happening, may I see your screenshot in Device Manager? Thanks a lot.

andreas-eg, 2019-09-18:
I got this on my mac after cmake the sdk's from source then using Xcode to build it.

Backend in rs2_open_multiple(sensor:0x7fd1025a9410, profiles:0x7fd10762b970, count:1):
fail to start streaming. Last Error: Operation timed out

Normal video feed is working.

the-faboratory, 2020-05-11:
Hello all, I received similar errors during the update process (rs-fw-update)
```
11/05 15:55:34,736 ERROR [18868] (handle-winusb.h:55) CreateFile failed, error: 5
11/05 15:55:34,738 ERROR [18868] (context.cpp:427) Exception thrown from user callback handler

```
and in the Viewer after clicking "Install" when prompted with a new update on SDK 2.33.1.1360.
`UNKNOWN in rs2_create_device(info_list:00000202B287FFA0, index:0): Failed to create FW update device, device id: [A bunch of device-specific details]`

The solution posted by c-sja solved my solution. I uninstalled the drivers (make sure to also mark "delete driver" if that is an extra option given by windows!), and when I re-inserted the D425, the command-line update utility worked. Now the device shows up in the "Cameras" section of device manager, and "rs-fw-update -l" shows the proper firmware version number. Thanks, c-sja!