realsenseai/librealsense #1180: Reading and processing the ros*.bag file slowly without frame drop

nawalgupta, 2018-02-15:
**This is a question and not a issue.**

### Issue Description
I want to read and process all the frames using pipeline api (wait_for_frames or poll_for_frames). But it seems that some internal threads are dropping the frames. 

It seems that we can use playback_device to slowly read the frames. I am looking for some example where I can get composite frame sets like pipeline in an easy way without droping the frames or reading the file slowly.

zivsha, 2018-02-15:
Hi @nawalgupta ,
Playback device mimics the real device and should not be dropping frames that are raised from its playback sensors. If you register to each of the playback device's sensors's frame callback then you should get all frames without drops. The pipeline is unaware of the device that is being used and does the same for live and playback device in terms of synchronization (time) and aggregation (to frame set) of frames. 

For an example of how to register for sensor callback please see the rs-sensor-control example and just pass a `playback_device` instance as the `rs2::device` instance (they are polymorphic).
You can then do whatever you wish with the frames, for example synchronize and aggregate the frames using a processing_block (just like the pipeline does). Since the aggregation part is not a public API, we don't have an example that shows how to collect the frames from all sensors and group them into a frameset, you could always look at the pipeline source code but what's the point of doing the same thing that the pipeline does?

We have plans to allow further control over the pipeline (like deciding which processing it should use) but for now, nothing like that is available other than using the device and sensor APIs.

closed 2018-02-27 by jlgarcia75

apoorva2398, 2018-08-07:
See https://communities.intel.com/message/553535#553535.