realsenseai/librealsense #1181: D435: USB device is inserted in USB3.0 but recognized as USB2.0.

unagiHuman, 2018-02-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  latest                       | 

### Issue Description

Realsense D435 often misidentifies as USB 2.0 though it is inserted in USB 3.0.
If I pull out the USB terminal several times and insert it and repeat and repeat, what is the cause of this phenomenon?

jlgarcia75, 2018-02-15:
Hello @unagiHuman,

Please provide more information so that we can propertly investigate this issue. Do you mean that the D435 is sometimes identified as USB2.0 only when you pull out the USB terminal several times and insert it repeatedly?

Does this happen when you connect it once and leave it in?

Please be more specific on exactly how you create this error.

If you can, please send a close up picture of the USB port on the PC to which you are connecting the camera and a screenshot of the screen where the camera is identified as USB 2.0.

Regards,
Jesus G.
Intel Customer Support

unagiHuman, 2018-02-16:
OK, I will tell you the conditions under which the problem will occur in detail.

- No error occurs when Realsense is operating properly, but it occurs after reconnecting the USB terminal
- Frequently reproduce after reconnecting the USB terminal. However, it does not mean that the error is reliably reproduced.
- When I start RealSenseViewer after rebooting my PC with realsense USB terminal connected to PC, Realsense can not connect or an indication that USB 2.0 is recognized appears

- Image of the USB 3.0 terminal on the PC side
[image]
- RealSenseViewer image when a problem occurs
"Intel Realsense USB 2" is displayed.
In this state the application can not receive Depth
[image]

jlgarcia75, 2018-02-16:
Thank you, @unagiHuman. We will investigate this issue.

Regards,
Jesus G.
Intel Customer Support

GitwellAnyohub, 2018-02-16:
Hey @unagiHuman, the thing that @jlgarcia75 was describing (when you pull out the cable and insert it again many times) has happened to me with the D415 camera. I was able to fix it (get the computer to recognize that it was a D415 and not just an intel USB 2.0 device) by using another cable. @jlgarcia75  do you have any idea why that happened?

dheera, 2018-02-17:
I had this issue as well.
Restarted PC. Issue persisted.
Tried installing it on another PC. Worked fine.
Plugged it back into first PC. Issue magically resolved itself.

vingsony, 2018-02-19:
Can you make sure that the cable you use is USB3.0 ? Please make sure that.

zenving, 2018-02-19:
I think the issue [https://github.com/IntelRealSense/librealsense/issues/1148](url) will be helpful.

unagiHuman, 2018-02-19:
@vingsony @zenving 
I am using the USB cable that was in the box containing Realsensed 435 I bought. it should be a USB 3.0 cable.

dheera, 2018-02-19:
I can verify the cable I am using is USB 3.0. The problem happens sometimes and not other times, using the exact same computer, cable, and D435 device with repeated connect/disconnection. The probability of success is different across different computers. The probability of it succeeding appears very far from IID (I find if it fails once, it is more likely to fail the next reconnection as well).

WolleTD, 2018-03-28:
Probably related, so I think its worth contributing: I got one D435 on an Intel NUC running Ubuntu. When the RealSense is connected while booting, according to `dmesg`, it exclusively communicates with the USB 2.0 hub and fails enumerating.

I am certain that `usb 2-2` is the RealSense while `usb 2` is a USB 2.0 host controller:
```
[    2.028034] usb 2-2: new full-speed USB device number 2 using xhci_hcd
⋮
[    2.148042] usb 2-2: device descriptor read/64, error -71
⋮
[    2.376043] usb 2-2: device descriptor read/64, error -71
[    2.604032] usb 2-2: new full-speed USB device number 3 using xhci_hcd
[    2.724041] usb 2-2: device descriptor read/64, error -71
⋮
[    2.965963] usb 2-2: device descriptor read/64, error -71
⋮
[    3.191993] usb 2-2: new full-speed USB device number 4 using xhci_hcd
[    3.192086] usb 2-2: Device not responding to setup address.
⋮
[    3.400173] usb 2-2: Device not responding to setup address.
[    3.608025] usb 2-2: device not accepting address 4, error -71
[    3.728026] usb 2-2: new full-speed USB device number 5 using xhci_hcd
[    3.728173] usb 2-2: Device not responding to setup address.
[    3.936179] usb 2-2: Device not responding to setup address.
[    4.144023] usb 2-2: device not accepting address 5, error -71
[    4.144042] usb usb2-port2: unable to enumerate USB device
```

Upon reconnection it physically, USB3-Controller `usb 3` is taking care of it:
```
[   46.936210] usb 3-2: new SuperSpeed USB device number 2 using xhci_hcd
[   46.956487] usb 3-2: New USB device found, idVendor=8086, idProduct=0b07
[   46.956489] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   46.956490] usb 3-2: Product: Intel(R) RealSense(TM) 430
[   46.956492] usb 3-2: Manufacturer: Intel(R) RealSense(TM) 430
```

RealSense-Customer-Engineering, 2018-03-29:
[Realsense Customer Engineering Team Comment] 
You can refer to the similar  issue #1391 (enumerated as USB2 rather than USB3) to see if it can help in your case.
https://github.com/IntelRealSense/librealsense/issues/1391
in that issue, the D4xx is enumerated as USB2 high speed device (not full speed)

rohitsinghbhadana, 2018-06-30:
Definitely a faulty cable issue.  reversing & reinserting  the typeC connector to my d415 solves the issue. This is with the included cable. I hope the issue is resolved with future cables, else whats the point of including a typeC connector.

arkulinskis, 2018-07-07:
Reversing the cable lets my d435 go to usb 2.0 or be not detected. I have not been able to get it to usb 3.0 . I even got a powered usb 3.1 usb hub but still no luck. Tried 3.0 and 3.1 ports. Played with bios settings. Bought a 10gbps 3.1 cable. Nothing works. Any help will be appreciated.

arkulinskis, 2018-07-13:
Getting a reversible (small port on both ends) usb-c 3.1 doesn't work either. I will try again and post my progress. Has the issue of USB3.0 been resolved with actionable steps?

Redunca, 2018-08-31:
What cable gives the best results ?

ioreed, 2018-09-12:
We found that Anker cables work really well long term. Amazon has them. USB3 also works best over the shortest distance possible.

Anker cable on Amazon: https://www.amazon.com/gp/product/B01HF0YGCK/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1

smirkingman, 2018-12-04:
I am experiencing the same problems with Windows 10 Pro 1709 16299.431, D435 #825312070943 firmware 05.10.13.00, using the USB 3.0 cable supplied with the camera on a recently-purchased Dell Latitude 5580.

Incorrectly detecting the 435 as USB2.0 is random, about 2/3 of the time, unplug - replug - reboot, I've tried repeatedly and there there's no systematic pattern, it works after a fiddling for a while.

However, I have a bit more luck with the RealSense viewer as compared to my own application; is there some sort of 'reset' that the RealSense viewer is doing that I am unaware of?

Better still, is there a way to programmatically force the camera to be re-recognised?

P.S. I've never experienced this on a separate rig running Windows 10 on an Intel NUC.

mpcuter, 2019-01-22:
I experience the same issue as @smirkingman. A workaround I use is it use camera parameter compatible with USB2. However, this restricts the FOV of the sensor and is becoming an increasingly frustrating issue. I see the device recognized as USB2 on coworkers windows machine through realsense viewer, my ubuntu 16.04 Dell laptop and odroids through the realsense sdk.

Does anyone know any reliable remedies for this issue?

Nandhiny-Duraikannu, 2019-02-06:
@smirkingman @mpcuter i have the same issue. were you able to find a solution?

smirkingman, 2019-02-09:
I found that the dropback from Usb3 to Usb2 was due to cables made of
Chineseum (poor quality). Try with a short, high-quality Usb3 cable.

neocubist, 2019-05-30:
Just adding a comment that could help. I was using a win10 computer with a usb3 port, but a cable that was usb-c to usb2. The realsense viewer kept identifying the connection as a usb2 and the rgb stream option was not available (only depth). Upon switching to a usb3-usb-c cable, my D435i works as expected.

robertofazio, 2019-06-10:
If you get into 'no devices found' maybe you could be find useful a similar approach 

https://github.com/robertofazio/usb_reset_intel_realsense_435/blob/master/USB_RESET.bat

it's harcoded but it works in my case

a-m-p-m, 2020-02-09:
I have a similar if not the same problem as @unagiHuman. 

**Some background**: 
I have two D435 cameras and two laptops running Windows 10. 
One laptop has a USB 3 port, the other one has 2 USB-C ports. 
I am using the cables provided with the D435 cameras as well as a bunch of other cables and adapters. 
I know that one of the cameras and some of the cables/adapters are definitely working because I checked them on a different Windows computer. The camera was recognized as a USB3 device and worked fine there. 

**The problem**: 
On both laptops, both D435 show up in RealSense Viewer as connected through a USB 2.1 port. 

**The weird part**: 
After checking one of the cameras on a different Windows computer (worked fine there), 
the camera was recognized as a USB3 device on both laptops and worked fine with them, 
and, also, the second camera (which I did not connect to that other Windows computer) 
was properly recognized as a USB3 device on both laptops and worked fine with them too. 
However, after turning the laptops off and back on, the situation returned to square one: 
both cameras are shown in RealSense Viewer as connected via USB 2.1 on both laptops. 

I did not repeat connecting the camera(s) to that other Windows machine because I no longer had access to it. 

**What can be checked for the cameras or laptops to fix this problem?**

smirkingman, 2020-02-10:
I had these problems, all due to cheap cables in chineseum. Change cables
until you get consistent usb3. If your pc supports thunderbolt, use that,
works better still

arkulinskis, 2020-02-10:
> I had these problems, all due to cheap cables in chineseum. Change cables until you get consistent usb3. If your pc supports thunderbolt, use that, works better still
> […](#)

You used it twice now in this thread, what is chineseum?

a-m-p-m, 2020-02-11:
@smirkingman : 
I am using the cables which worked fine with D435 & D415 on several other computers. 
I don't think the cables are the problem. 

@unagiHuman : 
What was the cause of your problem with D435?

a-m-p-m, 2020-02-12:
@RealSense-Customer-Engineering @RealSenseCustomerSupport 
@jlgarcia75 @dorodnic :

Unfortunately, the issue persists. 
Any suggestions, please?

tispratik, 2020-03-26:
We faced similar issue, the out of the box cable with the camera was not good. We separately bought 3Amps cable from Newnex which helped solve the issue. The problem is most USB cables are designed to carry 0.9Amps of current. Realsense needs the 3Amps one.

arkulinskis, 2020-03-26:
USB power delivery might be an issue. You said 3Amp cable is required but usb ports only give out 500mAmps max, often defaulting to 100mAmps. I am not sure if that is a path of questions to check out? How do I give max juice to the usb port? because i don't think the camera's software is requesting it by default.

tispratik, 2020-03-26:
USB 3.0 is 900 mAmps. But unfortunately regular 0.9A cables don't do the job well. Realsense when streaming with depth needs USB 3.0 port.