realsenseai/librealsense #1186: RuntimeError: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!

kmonachopoulos, 2018-02-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300} | 
| Firmware Version                   | (??) | 
| Operating System & Version |   {Debian GNU/Linux}  | 
| Kernel Version (Linux Only)    |  (4.4.74-g10816f6)                                         | 
| Platform                                 | HikeyBoard 960   |
| SDK Version                            |  { ?? }                          | 

### Issue Description
I managed to build and flash the debian kernel from scratch, enabling also the usb module during configuration. I build librealsense from source with python support and managed to work on the RGB space from the camera. Configuring with : 

`config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
`

However, when I configure to stream the depth image : 

`config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
`

I get a runtime error : 

```
Traceback (most recent call last):
  File "minimum-example.py", line 13, in <module>
    pipeline.start(config)
RuntimeError: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device

```
I tried several methods to patch the kernels in order to support INZI format but with no luck. I also tried this approach : 

`https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
`

since I use ARMv8 architecture but it didn't work either. Of course following the instructions on X86 it works ok, but on Hikey I get errors. Any suggestion?

dorodnic, 2018-02-16:
Hi @kmonachopoulos 
If you are able to build the kernel from source and your main goal is to get depth streaming (and not some of more advanced controls / metadata support), the core patches are relatively easy to apply by hand. 
You will need to add couple lines of code to `/drivers/media/usb/uvc/uvc_driver.c`, `/drivers/media/usb/uvc/uvcvideo.h` and `/include/uapi/linux/videodev2.h` to make V4L2 recognise `INZI`. You can use [the patch for Ubuntu](https://github.com/IntelRealSense/librealsense/blob/development/scripts/realsense-camera-formats_ubuntu-xenial-hwe-zesty.patch) as reference. 
Once you make the changes and successfully rebuild the kernel, it would be relatively easy to generate new `.patch` that you are welcomed to share here for future use. 
Alternatively, you can build the library with cmake flag `-DFORCE_LIBUVC=true` to circumvent the V4L2 sub-system. It will most likely work, but the result is less stable (hence we don't recommend it if you can patch uvcvideo instead)

kmonachopoulos, 2018-02-25:
Hello @dorodnic 

I managed to patch and rebuild the kernel and everything works fine. I used the provided patch to add additional supported camera formats. 

Steps to build for Hikey960: 

1) Download Kernel source files for Hikey960.
2) Build the initial Kernel.
3) Patch the kernel to support realsense INZI data type when streaming depth from rs300.
4) Configure the kernel and enable USB Video Class (UVC) driver. (menuconfig)
5)  Build the Kernel and generate the image file.
6) Configure Hikey to fastboot from external devices.
7) Install the image file to Hikey.

Thank you.

closed 2018-02-25 by kmonachopoulos

KeCh96, 2018-05-28:
I run into the same problem. My OS is Debain, Linux kernel is 4.11.0 . I only want to get depth frames from SR300. 
I try to follow your guide, but I cannot found `/drivers/media/usb/uvc/uvc_driver.c` or `/drivers/media/usb/uvc/uvcvideo.h` or `/include/uapi/linux/videodev2.h` in my computer. Could you please tell me where are these three files? @dorodnic