realsenseai/librealsense #1187: Unity Wrapper - Crashes when set to Unity Thread

gjmsoftware, 2018-02-15:
### Issue Overview
I have successfully installed the Unity wrapper and it works properly when the **RealSenseDevice**'s "Process Mode" is set to **Multithread**. However, the example scene (and any of my test scenes) crash when trying to have the "Process Mode" set to **Unity Thread**. 

----------------------------------------------------------------------------------------------------
### Issue Description
When set to **Unity Thread** the scene will either crash while loading or (more common) will start and collect a few frames and then crash. When this happens, I get an error in the Console:
_"Exception: Object does not support "librealsense::depth_sensor" interface!"_

Furthermore, the Colorized Depth image is not correct, see the image below:
[image]

I have attached my most recent [error.log](https://github.com/IntelRealSense/librealsense/files/1728763/error.log) (generated from Unity, which appears to be mostly a memory dump).

As I've mentioned, the wrapper works correctly when the "Process Mode" is set to **Multithread**.

Any assistance/advice is greatly appreciated - thank you!

----------------------------------------------------------------------------------------------------
### System Info

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | Unity -  5.5.1f1 |
| SDK Version                            |  V2.10.0                         |

zivsha, 2018-02-16:
Thank you for the information, looks like it's coming from the Colorized Depth (or Alignment), as a temporary work around you can disable these objects. Raw streams look ok with unitythread mode

gjmsoftware, 2018-02-16:
Good suggestion @zivsha and I'll keep you updated if I'm able to solve the issue/find any more information that could help in debugging. 
Unfortunately, my project requires the Colorized Depth, so I will rework my code to work in the Multithread mode for the time being.

c-sja, 2018-02-21:
Confirmed this is also happening for me, crashes on Unity Thread mode every time unless I disable the Colorized Depth and Alignment objects. Multithread works fine.

I would note that I'm also getting the "Exception: Object does not support "librealsense::depth_sensor" interface!" error even in Multithread mode though, but it doesn't appear to break anything?

zivsha, 2018-02-21:
@c-sja thanks for the information, I've noticed that exception as well and my PR will fix that as well.

zivsha, 2018-02-22:
Hi guys, please update if the merged change works for you

c-sja, 2018-02-23:
Hi @zivsha thanks for the changes.

I switched to development branch for your changes. In Unity it's complaining "Assets/RealSenseSDK2.0/Scripts/PointCloudGenerator.cs(19,13): error CS0246: The type or namespace name `PointCloud' could not be found. Are you missing an assembly reference?"

Did I need to update the SDK as well or something? I have version 2.10.0 from 11 days ago, which appears to be the most recent.

unagiHuman, 2018-02-23:
@c-sja 
Perhaps development branch PointCloud add has not yet been reflected in LibrealsenseWrapper.dll.
Copy  csharp wrapper source code from development branch to unity your project.　
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/csharp/Intel.RealSense

c-sja, 2018-02-23:
@unagiHuman That worked thanks!

closed 2018-02-25 by dorodnic