realsenseai/librealsense #1189: More efficient way of aligning images

dheera, 2018-02-17:
The current method of aligning images is highly CPU-intensive, involving repeated function calls to rs2_deproject_pixel_to_point, rs2_transform_point_to_point, and rs2_project_point_to_pixel on every point in the image. At 30fps this can consume as much as 50% of a core for just depth-RGB alone, let alone IR images.

Can we have a hardware-accelerated function that can align an entire image in one go instead of millions of repeated function calls per second? Better yet, can this be done in firmware?

dorodnic, 2018-02-17:
Hi @dheera 

The D415 device offers "RGB from Left Imager" format. This mode is extracting the colour component from Left Infrared sensor, and using the same hardware rectification pipeline as depth. Resulting image is perfectly aligned to depth. The downside is that it lacks the Full-HD resolution of the side RGB sensor, as well as some of the controls. 

As far as I know the D400 ASIC does not offer the functionality of stream alignment in hardware. 
We are looking into ways to reduce CPU utilisation on various platforms.

dheera, 2018-02-19:
Thanks @dorodnic . Although my time is limited I'm playing with some potential possibilities:
- Beyond >2 meters or so, a simple crop+translation is "good enough" since the sensors are close to each other and the depth image resolution is not very high.
- For <2 meters, some approximations can be made.
- The Y axis mapping can be hard coded based on camera extrinsics/intrinsics and does not depend on the depth value. Only the X axis needs a computation.
- I would like to put the X axis computation into the loop itself and eliminate the function calls if possible.
Feedback appreciated or if someone is already working on this let me know.

dorodnic, 2018-02-25:
We are looking into optimizing this flow, but focusing on more efficient code generation (using SIMD / GPU). These are all good points, but we don't want to rely on some of the assumptions you are making (there are RealSense cameras with larger baseline or with RGB sensor not being on the same plane with stereo cameras) 
That said, you are welcomed to share your results and it will likely help others in the community.

TheMikeyR, 2018-03-01:
@dorodnic I've tried aligning with left infrared camera instead of rgb color camera, and it works pretty well. But the color is pretty bad in the infrared camera module compared to the color camera and kinda defies the purpose of using the infrared to align with, or is there any smart way to use the alignment of infrared->depth to make a lookup in the rgb camera?

RealSense-Customer-Engineering, 2018-05-10:
[Realsense Customer Engineering Team Comment]
hi @TheMikeyR, we are working on improving the left imager picture on D415, but it's it's what it is for now.
hi @ @dheera, GPU and CPU investigations are still ongoing, but no breakthrus yet, will keep you posted.

dorodnic, 2018-07-26:
We implemented some significant improvements to align in last release. This is best we can do with current hardware. If you have any ideas on how this can be further accelerated, please open a follow-up ticket.

closed 2018-07-26 by dorodnic

hpf9017, 2019-08-15:
> Hi @dheera
>
> The D415 device offers "RGB from Left Imager" format. This mode is extracting the colour component from Left Infrared sensor, and using the same hardware rectification pipeline as depth. [...]

hi @dorodnic,
could I align the depth to left IR image with d435i so that the alignment is perfect ?
and how to achieve that since I find it deals only with the rgb color image in the code example rs-align and rs-align-advanced?

agrunnet, 2019-08-15:
Inherently the Left IR is always perfectly aligned with the depth. No calibration needed. This is because this is one of the image pair used for the depth calculation.

moeller0, 2019-09-09:
Silly question, in the viewer IR cameras are identified by number, which of the two is the left IR camera?

hpf9017, 2019-10-01:
@agrunnet I got it. thank you very much.
By the way, do you know how to evaluate the effect of the color to depth alignment?
In other words, how can I know that I need to calibrate the extrinsics to get better alignment.

agrunnet, 2019-10-01:
I usually point at a White wall and then place a colored object between wall and camera, such as my hand. If I see part of my hand on the wall or some white on my hand then I know they are misaligned. Clearly there are ways to do this more rigorously.